Koenkk/zigbee2mqtt #18606: PC321-Z-TY 3-phase Clamp Power Meter jumping values

morph027, 2023-08-13:
### What happened?

Just got a PC321-Z-TY 3-phase Clamp Power Meter wired up and connected via brand new z2m installation using a Sonoff USB Stick. Z2M part is working correct, i guess. By looking at https://github.com/Koenkk/zigbee2mqtt/issues/17153, i see weird values too sometimes. Saw them during installation and just flipped the clamp to get proper values. Changing the clamp to another phase shows the wrong values there. Once i got it working, i noticed the behaviour again.

[image]
[image]
[image]
[image]

Still not sure what happens there. Not really an Z2M problem, or?

### Zigbee2MQTT version

1.32.2 commit: 1ec1e57

### Adapter firmware version

zStack3x0

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

dhde, 2023-08-14:
Same here for me. I see the values like 39.xxx W if the Value that I expect is a negative value. So if the PV generate more energy than in the house is consumed.

morph027, 2023-08-14:
On this phases, i do not use PV or export to grid, this is attached to a different meter just for the flat within our house. On vacation right now, but will fiddle around with the clamps a bit more. Consumption on the other 2 phases is just 60W which can't be right (regarding the oven or microwave or vacuum)

morph027, 2023-08-22:
Jip, I've also paired the device unclamped, will try it later as I'm in vacation right now, thanks for the hint 👍

morph027, 2023-08-27:
Did a reset yesterday and it looks good so far. Will keep an eye on it and if everything works, i can create a docs PR to update device page.

Is there a way to reset the overall stats? I've removed from z2m and paired again, but all consumption values still have the old (wrong) values.

morph027, 2023-08-28:
Nargh, this morning the strange readings appeared again 😞

closed 2023-10-05 by github-actions[bot]

morph027, 2024-01-21:
Just bought this one and it seems to work ;)

https://zigbee.blakadder.com/Zemismart_SPM02-D2TZ.html