Koenkk/zigbee2mqtt #18612: [New device support]:

chewth91, 2023-08-14:
### Link

https://www.aliexpress.com/item/1005005909540714.html?gps-id=pcStoreNewArrivals&scm=1007.23409.271287.0&scm_id=1007.23409.271287.0&scm-url=1007.23409.271287.0&pvid=5175abbe-b739-46df-8ec7-c2c2a8f4c801&_t=gps-id:pcStoreNewArrivals,scm-url:1007.23409.271287.0,pvid:5175abbe-b739-46df-8ec7-c2c2a8f4c801,tpp_buckets:668%232846%238114%231999&pdp_npi=4%40dis%21MYR%21107.42%2175.21%21%21%2122.84%21%21%402101c59816919999368335919e0b96%2112000034805127430%21rec%21MY%21%21A&spm=a2g0o.store_pc_home.smartNewArrivals_2002856433204.1005005909540714

### Database entry

{"id":67,"type":"Router","ieeeAddr":"0xa4c138f81e577a0c","nwkAddr":64592,"manufId":4417,"manufName":"_TZE200_2aaelwxk","powerSource":"DC Source","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,57346,61184,60928,57344,1024],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"zclVersion":3}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00257cc7da","zoneState":0}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":32770}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1691999105436,"defaultSendRequestWhen":"immediate"}

### Comments

Hi, just got this device from aliexpress recently and found that it is not supported in Z2M. 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [...tuya.tz.datapoints.key, 'alarm', 'alarm_time', 'alarm_volume', 'type', 'volume', 'ringtone', 'duration','indicator',
    'motion_detection_distance', 'motion_detection_sensitivity', 'small_motion_detection_distance',
    'small_motion_detection_sensitivity', 'static_detection_distance', 'static_detection_sensitivity','motion_false_detection','fading_time',
    'breathe_false_detection','working_mode']
}

const definition = {
    fingerprint: tuya.fingerprint('TS0225', ['_TZE200_2aaelwxk']),
    model: 'ZG-205Z/A',
    vendor: 'TuYa',
    description: '24Ghz Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
    exposes: [
        e.enum('presence_state', ea.STATE, ['none', 'presence']).withDescription('Presence'),
        e.enum('human_motion_state', ea.STATE, ['none', 'large', 'small', 'breathe']).withDescription('Human Motion state'),
        e.illuminance_lux(),        
        e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(28800).withValueStep(1).withUnit('s')
            .withDescription('Presence keep time'),
        e.numeric('motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.01).withUnit('m')
            .withDescription('Motion detection distance'),
        e.numeric('motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
            .withDescription('Motion detection sensitivity'),
        e.numeric('small_motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
            .withDescription('Small motion detection distance'),
        e.numeric('small_motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
            .withDescription('Small motion detection sensitivity'),
        e.numeric('static_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
            .withDescription('Static detection distance'),
        e.numeric('static_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
            .withDescription('Static detection sensitivity'),
        e.numeric('motion_false_detection', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
            .withDescription('Motion false detection'),            
        e.enum('working_mode', ea.STATE_SET, ['off', 'arm', 'alarm']).withDescription('Alarm mode'),
        e.enum('alarm_volume', ea.STATE_SET, ['mute', 'low', 'medium', 'high']).withDescription('Alarm volume'),
        e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(60).withValueStep(1).withUnit('m').withDescription('Alarm time'),
        e.binary('indicator', ea.STATE_SET, 'ON', 'OFF').withDescription('LED Indicator'),
        e.binary('breathe_false_detection', ea.STATE_SET, 'ON', 'OFF').withDescription('Breathe False Detection'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'presence': tuya.enum(1)})],
            [2, 'motion_detection_sensitivity', tuya.valueConverter.raw],
            [3, 'mov_minimum_distance', tuya.valueConverter.raw],
            [4, 'motion_detection_distance', tuya.valueConverter.divideBy100],
            [101, 'human_motion_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'large': tuya.enum(1), 'small': tuya.enum(2), 'breathe': tuya.enum(3)})],
            [102, 'fading_time', tuya.valueConverter.raw],
            [103, 'motion_false_detection', tuya.valueConverter.raw],
            [104, 'small_motion_detection_distance', tuya.valueConverter.divideBy100],
            [105, 'small_motion_detection_sensitivity', tuya.valueConverter.raw],
            [106, 'illuminance_lux', tuya.valueConverter.raw],
            [107, 'indicator', tuya.valueConverter.onOff],
            [108, 'static_detection_distance', tuya.valueConverter.divideBy100],
            [109, 'static_detection_sensitivity', tuya.valueConverter.raw],
            [110, 'micro_minimum_distance', tuya.valueConverter.raw],
            [111, 'motionless_minimum_distance', tuya.valueConverter.raw],
            [112, 'reset_setting', tuya.valueConverter.raw],
            [113, 'breathe_false_detection', tuya.valueConverter.raw],
            [114, 'time', tuya.valueConverter.raw],      
            [115, 'alarm_time', tuya.valueConverter.raw],            
            [116, 'alarm_volume', tuya.valueConverterBasic.lookup({'low': tuya.enum(0), 'medium': tuya.enum(1), 'high': tuya.enum(2), 'mute': tuya.enum(3)})],
            [117, 'working_mode', tuya.valueConverterBasic.lookup({'arm': tuya.enum(0), 'off': tuya.enum(1), 'alarm': tuya.enum(2),  'doorbell': tuya.enum(3)})],
            [118, 'auto1', tuya.valueConverter.raw],      
            [119, 'auto2', tuya.valueConverter.raw],      
            [120, 'auto3', tuya.valueConverter.raw],                  
        ],
    },
};

module.exports = definition;
```

Koenkk, 2023-08-14:
Does this ext converter work? I yes, could you make a pull request?

chewth91, 2023-08-18:
> Does this ext converter work? I yes, could you make a pull request?

@Koenkk it's working fine for and haven't seen any issues. I am not familiar with GitHub and am not sure about how to make a pull request.

kkossev, 2023-08-18:
I can confirm that Tuya DPs shown above are the correct ones for TS0225 _TZE200_2aaelwxk

@chewth91 I suppose that you have Tuya Zigbee GW, are there "Move Self Test", "Small Move Self Test" and "Breathe Self Test" functions in the Smart Life app?   ( these could be potentially DP 118, 119, and 120).

chewth91, 2023-08-18:
> I can confirm that Tuya DPs shown above are the correct ones for TS0225 _TZE200_2aaelwxk
>
> @chewth91 I suppose that you have Tuya Zigbee GW, are there "Move Self Test", "Small Move Self Test" and [...]

Hi, i did not see self test in Smartlife. 

[image]

[image]

[image]

[image]

Koenkk, 2023-08-18:
All `TS0225` will be detected as supported now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

leroyloren, 2023-08-18:
I have **_TZE200_2aaelwxk** with this converter it detects but doesn't show **motion_state** and **illuminance_lux**, the rest seems to work.

I am attaching debug logs etc.

[image]

`{"id":41,"type":"Router","ieeeAddr":"0xa4c1388baef32d03","nwkAddr":34496,"manufId":4417,"manufName":"_TZE200_2aaelwxk","powerSource":"DC Source","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,57346,61184,60928,57344,1024],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0021b9518e","zoneState":0}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":33545}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1692383344135,"defaultSendRequestWhen":"immediate"}`

[image]

[image]

https://www.aliexpress.com/item/1005005904142527.html

[image]

kkossev, 2023-08-18:
@chewth91 thank you for the screenshots - obviously, this model does not have exactly the same functions as the other TS0225 radar :

[image]

chewth91, 2023-08-19:
> I have **_TZE200_2aaelwxk** with this converter it detects but doesn't show **motion_state** and **illuminance_lux**, the rest seems to work.
>
> I am attaching debug logs etc.
>
> [image] [...]

This happened to me before as well. Have you entered all the parameters?

Koenkk, 2023-08-19:
I've adapter the converter for the `_TZE200_2aaelwxk`, can you test if everything works? 

https://gist.github.com/Koenkk/581a5f396d1ca9ac0ea9477b1017af36

closed 2023-08-19 by Koenkk

alexvaltchev, 2023-08-28:
@Koenkk when will this sensor be fully operational and working in regular Z2M ? I tried this external converter for **"TS0225 by _TZE200_2aaelwxk"** version, and it shows, but none of the sliders seem to be working. Am I doing anything wrong? OR is there better way to work in ZHA perhaps? Have anyone made this sensor working? Thanks

ZHA:
[image]

Z2M:
[image]
[image]
[image]

Is this Zigbee2MQTT fully functional now?

Does anyone know what each of those settings mean and how to set them for best results?
Thanks

Koenkk, 2023-08-29:
@13717033460 since you added support for this, can you check the issue above?

alexvaltchev, 2023-08-29:
I contacted the AliExpress seller that sent me the sensor and they have emailed me their converter. It is different. Can anybody check both and let me know which one is better and should be used?

Here is the link:[]( https://doc.szalarm.com/zg-205Z/z2m.html)

I also asked what each setting do, and here is their link with information: 
LINK: [](https://doc.szalarm.com/zg-205Z/en/)

alexvaltchev, 2023-08-29:
@13717033460 so the AliExpress seller converter is the latest and better?

13717033460, 2023-08-29:
sorry,I made a mistake. This is it. Please check the attachment.

13717033460, 2023-08-29:
> Sorry, am I blind? I dont see attachment. Do you mean the converter that is posted by @chewth91 is the latest and best? OR the seller from AliExpress is newer and better? I feel stupid :( I am so sorry to ask again...

{
        fingerprint: tuya.fingerprint('TS0225', ['_TZE200_2aaelwxk']),
        model: 'ZG-205Z-A',
        vendor: 'TuYa',
        description: 'Mini 24Ghz human presence sensor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tzDatapoints],
        exposes: [
            e.presence(),
            e.enum('motion_state', ea.STATE, ['none', 'large', 'small', 'static']).withDescription('Motion state'),
            e.illuminance_lux(),
            e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(1).withUnit('s')
                .withDescription('Presence keep time'),
            e.numeric('large_motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.01).withUnit('m')
                .withDescription('Large motion detection distance'),
            e.numeric('large_motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
                .withDescription('Large motion detection sensitivity'),
            e.numeric('small_motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
                .withDescription('Small motion detection distance'),
            e.numeric('small_motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
                .withDescription('Small motion detection sensitivity'),
            e.numeric('static_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
                .withDescription('Static detection distance'),
            e.numeric('static_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
                .withDescription('Static detection sensitivity'),
            e.binary('indicator', ea.STATE_SET, 'ON', 'OFF').withDescription('LED indicator mode'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'presence', tuya.valueConverter.trueFalse1],
                [106, 'illuminance_lux', tuya.valueConverter.raw],
                [101, 'motion_state', tuya.valueConverterBasic.lookup({
                    'none': tuya.enum(0), 'large': tuya.enum(1), 'small': tuya.enum(2), 'static': tuya.enum(3),
                })],
                [102, 'fading_time', tuya.valueConverter.raw],
                [4, 'large_motion_detection_distance', tuya.valueConverter.divideBy100],
                [2, 'large_motion_detection_sensitivity', tuya.valueConverter.raw],
                [104, 'small_motion_detection_distance', tuya.valueConverter.divideBy100],
                [105, 'small_motion_detection_sensitivity', tuya.valueConverter.raw],
                [108, 'static_detection_distance', tuya.valueConverter.divideBy100],
                [109, 'static_detection_sensitivity', tuya.valueConverter.raw],
                [107, 'indicator', tuya.valueConverter.onOff],
            ],
        },
    },

alexvaltchev, 2023-08-29:
Thank you. Will try it now :)
What is new/different than the AliExpress version, do you know? Does it work better or more accurate? Thanks

13717033460, 2023-08-29:
OK,
In the process of using the product, you can contact me if you have any other questions, we are the manufacturer.

发自我的iPhone

alexvaltchev, 2023-08-29:
Great, I just want the latest external converter that support TS0225_TZE200_2aaelwxk version. I dont see attached file or link that I can download it. Thanks

13717033460, 2023-08-29:
Yes, this one is 5.8G, but the algorithm is much better than the normal 5.8G

发自我的iPhone

  

  
Hi @13717033460 ,
 
The first question is about what mmWave module is actually inside TS0225_TZE200_2aaelwxk
 
 
This is not a 24Ghz module as advertised, but a 5.8 Ghz mmWave module - can you confirm that?
 
Following the AliExpress seller link in the first post by @chewth91 , we see it has changed - there is no "24 GHz" anymore on the product description. Other AE sellers removed the ''24GHz" from the description, but the wrong info can still be seen in some of the pictures.
 
Can you comment on this situation?

13717033460, 2023-08-29:
Sorry!   AE seller advertising is wrong, long ago notified them to change

发自我的iPhone

kkossev, 2023-08-29:
Thank you for the answer, @13717033460 , fair enough. I appreciate your honesty.
The AE sellers bad practices are another topic, I don't want to hijack this technical thread.

alexvaltchev, 2023-08-29:
I checked on AliExpress most sellers say 24G:

https://www.aliexpress.com/item/1005005919353153.html
https://www.aliexpress.com/item/1005005904142527.html
https://www.aliexpress.com/item/1005005914116739.html

kkossev, 2023-08-29:
> I checked on AliExpress most sellers say 24G:

Open a dispute and ask for a full refund.

13717033460, 2023-08-29:
Sorry, it has a lot to do with our early technical publicity mistakes, because we did produce 24G products in the early days, and later because of the wrong 24G alarm, the experience is not very good. After 24G matures, 24G radar will be used again

发自我的iPhone

kkossev, 2023-08-29:
> Is there any real 24G? - https://www.aliexpress.com/item/1005005947206956.html
> What about this?

This one is real 24Ghz, based on LD2410C mmWave sensor. 
 But it is terribly chatty as the previous TS0601 models (sends the unnecessary distance measurement every single second - non-stop until a presence is sensed).

kkossev, 2023-08-29:
@13717033460 one very good thing about your new products is that they are not as chatty / not spammy as the rest of the Tuya-based 5.8Ghz radars. This is what made me so enthusiastic initially... until recently.

alexvaltchev, 2023-08-29:
Sorry to hijack the topic here, we figured it out these square ones are 5.8G. Does anyone have experience finding real 24G?

How about THIS:

Moes - https://www.aliexpress.com/item/1005005932571226.html
[image]

13717033460, 2023-08-29:
At the beginning of last month, we notified most of the sellers and changed the parameter to 5.8G。

Tomorrow, we'll find out which customer this is, and if you need to return it, I can coordinate.

发自我的iPhone

alexvaltchev, 2023-08-29:
@13717033460 what is the detection angel in degree? I put it 2 meters above and here are my settings, and is still not detecting.

[image]

13717033460, 2023-08-29:
At a very close distance ，Close to 90 degrees，After setting the parameters, wait 10 seconds。

Please check http://doc.szalarm.com/zg-205Z

发自我的iPhone

13717033460, 2023-08-29:
The triggering of the device from "NONE" to "Presence" state must first detect large movements in order to trigger radar small movements and static detection;

发自我的iPhone

alexvaltchev, 2023-08-29:
Yes, am aware of it. Thank you. I already read everything there. I have been reading all materials about this sensor, but so far in my case does not work good. Is even worst than the older version on my opinion.

[image]
[image]

It shows HOME from time to time than AWAY again.

13717033460, 2023-08-29:
You mean it says Home all the time?

发自我的iPhone

alexvaltchev, 2023-08-29:
@13717033460 
No, the opposite. IT shows presence FALSE, and rarely goes to TRUE for a little time and go back to FALSE. NO presence. The room is 5 meters long and 2.7 meters tall. I put the sensor on 2.1 meters high. I angeled it toward me a little down and still loses me. IS not detecting me.

alexvaltchev, 2023-08-29:
IF sensitivity is 9 is not detecting. IF sensitivity is 10 is detecting all the time (even when nobody is there)

13717033460, 2023-08-29:
How far away are you from him? This side mount product, you can not install so high，and fading time set to 60S

发自我的iPhone

alexvaltchev, 2023-08-29:
@13717033460 I am 1.7 meters from the device. Is mounted 2.1 meters from the ground. Fading time is 10 seconds, I want to react fast not to wait too long. I sit on the computer, I am static but it shows there is nobody, no presence detected.

13717033460, 2023-08-29:
You are still, fading must be 40s chapped on, and it is recommended to mount parallel to the person, because this product is side mount, there is no bracket

发自我的iPhone

13717033460, 2023-08-29:
If you are stationary and need to detect breathing action, you will be facing it, within 3m. But most people make tiny movements when they're not asleep,

发自我的iPhone

alexvaltchev, 2023-08-29:
@13717033460  so if more than 3 meters it will not detect me if sleeping? In order to detect static need to be less than 3 meters?

Why do I need 40 seconds fade? I want the light to turn off suoer fast l, after 5 seconds after I left the room.

alexvaltchev, 2023-08-29:
There is stand for the sensor
https://a.aliexpress.com/_mP2Ol6e

13717033460, 2023-08-29:
Yes, this is a problem with either 24G or 60G radar. If you are not completely stationary, you can set it for 10 seconds and move the radar as close to the detection area as possible. You can test what you're doing when you're reading a book, when you're working

发自我的iPhone

13717033460, 2023-08-29:
Sorry,I can't open the link. I'm on my phone. Give me the screenshot

发自我的iPhone

 

 https://a.aliexpress.com/_mP2Ol6e

alexvaltchev, 2023-08-29:
[image]
[image]
[image]
[image]

13717033460, 2023-08-29:
This is pir

发自我的iPhone

13717033460, 2023-08-29:
发自我的iPhone

alexvaltchev, 2023-08-29:
[image]
[image]

@13717033460  this is another senaor holder to angel the sensor into the right direction.

kkossev, 2023-08-29:
@alexvaltchev that's a very good idea (using the Aqara sensor holder) - the precise aiming to the direction where the stationary/breath sensing is expected is of critical importance for achieving the best results.  Precise aiming also helps a lot to avoid interference from ceiling of floor-standing fans, air conditioners, etc.

alexvaltchev, 2023-08-29:
Yes I already purchased. Waiting on it. I can send you a link for the second one when I get home if you want.

kkossev, 2023-08-29:
Yes, please.

magicmellon, 2023-10-17:
Appologies for joining a stale chat - I just wanted to confirm what this device _actually_ is - I have the Zigbee Model: TS0225, Zigbee Manufacturer: _TZE200_2aaelwxk, is this a 5.8GHz or 28GHz device? If 5.8, is it worth updating the Zigbee2MQTT description to reflect that?

hepspl, 2023-12-01:
> use https://gist.github.com/Koenkk/79a20ea7e5eaf35749e157d0c31a1f9b for TS0225 and _TZE200_hl0ss9oa /sorry

It crashes the daemon :( 

> (node:8920) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'datapoints' of undefined

Hammerhand17, 2023-12-02:
> One more question. Bought several of these devices, which are now detected as soon as I add them in z2mqtt:
> 
> ![image](https://private-user-images.githubusercontent.com/20372496/286981510-faf78092-f3d1-4f8e-a7aa-be3b77a7bff9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE1MTg4MjYsIm5iZiI6MTcwMTUxODUyNiwicGF0aCI6Ii8yMDM3MjQ5Ni8yODY5ODE1MTAtZmFmNzgwOTItZjNkMS00ZjhlLWE3YWEtYmUzYjc3YTdiZmY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjAyVDEyMDIwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3OGJhZjdlNzk3ODRkYmNiZmJhMWMzY2E0YzlkMjY0N2ZjZWRhNWMwNWIwYzE3OGIxNmRmZjdmNjE1ZDhiNGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lVfDAKXjoCmqan8YXOhk3GUuF0f-HfwAoGWGl6VwNTg)
> 
> Besides, illuminance value is not detected from Home Assistant:
> 
> ![image](https://private-user-images.githubusercontent.com/20372496/286981732-44a3cda9-0d11-4d37-8b9f-dca5f37f3e64.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE1MTg4MjYsIm5iZiI6MTcwMTUxODUyNiwicGF0aCI6Ii8yMDM3MjQ5Ni8yODY5ODE3MzItNDRhM2NkYTktMGQxMS00ZDM3LThiOWYtZGNhNWYzN2YzZTY0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjAyVDEyMDIwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3OTIwNGYwOTQyYmIzYjg3ZWM3NDdhYmNlOGJiMDI2NTExZTEzMDhmNDhiOGFhNGI3ZGI0OTNkYjViMGY5ODcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.HpRxUIbnjPN97t1aipna1kf3T4EXBwbH6PLRl-sYhdc)
> 
> Any clue?

Just in case anyone needs it; just have to go to sensor page and enable this entity.

HNKNTA, 2023-12-10:
I know this ticket is closed, so I'd appreciate any recommendations on what could I do with the following:
I've ordered a few devices on aliexpress recently, like the ones mentioned on many sources as 5.8GHz device, like this
[image]
https://www.amazon.com/Presence-Millimeter-Detection-Zigbee2mqtt-Assistant%C2%A1%C2%AD/dp/B0CFHL81P8?th=1

It works currently as _TZE200_2aaelwxk via zigbee2mqtt, described as 5.8Ghz Human presence sensor.

And looking at this issue, I started a dispute on aliexpress, as advised above (and on some other forum) where guys prove this is 5.8GHz device by taking pictures of the disassembled device.

So, I've disassembled mine, as what I see: it's actually 24GHz! 
Revision V2.1, 24000MHz quartz:
[image]
Antenna:
[image]

Should I open a new issue for this? So far it works as is, but the illuminance values are crazy high, not sure if it's possible to calibrate it.

GitTheFOuttaHere, 2023-12-18:
> Should I open a new issue for this? So far it works as is, but the illuminance values are crazy high, not sure if it's possible to calibrate it.

What value do you get & do you have another luminance sensor to compare the values at the same location?

A tuya.valueConverter.divideByxx (xx being the value you set) can be added to an external converter.

So if you have one luminance sensor showing 48 & you have the one with high values showing 480, 4800, 48000 or the likes, you can then use the above to divide by 10, 100, 1000 respectively.

HNKNTA, 2023-12-20:
> What value do you get & do you have another luminance sensor to compare the values at the same location?
>
> A tuya.valueConverter.divideByxx (xx being the value you set) can be added to an external [...]

Yeah, I got like 10x values right now in one room, my other luminance sensors also not perfect though. The other room shows 56 vs 2134 on tuya sensor 🫤

I guess you are referring to [this](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#network-map), I will try to look into it on weekends, and see what I can do with it, thanks for the help!

GitTheFOuttaHere, 2023-12-22:
> Yeah, I got like 10x values right now in one room, my other luminance sensors also not perfect though. The other room shows 56 vs 2134 on tuya sensor 🫤
>
> I guess you are referring to [...]

I have the exact same sensor with same id code. It is automatically recognized in z2mqtt & no external converter is required.

Mine went up to over 3000lm almost every time there was motion. Found it was the led indicator on the front panel causing it.

Turned off, problem solved.

GitTheFOuttaHere, 2023-12-22:
And as for the external converter, there is some in this thread.

HNKNTA, 2023-12-22:
@GitTheFOuttaHere I have the LED turned off, but the values are still high.
[image]

And it's really higher than it should be, according to the description on the wiki and my other sensors:
https://en.wikipedia.org/wiki/Lux
[image]

Do you have normally accurate values for your device?

GitTheFOuttaHere, 2023-12-24:
Depends what "normally accurate" entails. My sensor (specifically TS0225_TZE200_2aaelwxk) would only go higher than 2000lm if I shine a bright led torch directly at it. I only have one of these and have mounted it under a cupboard. I have it there so it does not detect movement from the ceiling fan. And won't pickup my animals most times because there is a bench below it as well. It's also a good way to know how much actual sunlight there is. The values aren't accurate, but it works enough for me to use the readings I'm given.

A led torch I have which has a 1000lm output (on high) shining directly at it, shows over 3000lm.

The photo sensitive component isn't exposed, it's hidden under the plastic case. So not sure how accurate you will need it to be.

If your need for lumens value is critical, you may need a dedicated lux sensor.

Having said that, the lumens value for mine seems to be on par with the wiki. Had mostly overcast days and the value didn't go much higher than 1000lm.

The other sensors I have have different outputs, not in lumens. Their reading goes from 0 to 50 only.

I've ordered another one of these and I'll post the results for you.

[10 more replies]