Koenkk/zigbee2mqtt #18617: Homeassistant reports " Invalid modes mode: emergency_heating"

RedNo7, 2023-08-14:
### What happened?

In Home-assistant, calling for 'native boost' with the Hive SLR2c creates an error in the HA logs, but works fine with the device itself.

- I am following a script (almost identical) to the Z2M devices page [here](https://www.zigbee2mqtt.io/devices/SLR2b.html#how-to-start-edit-native-boost-heat-endpoint).
- The only difference is that my device is the SLR2c and that page is the SLR2b.  The SLR2b and SLR2c devices expose the same things but the SLR3b page does not give the 'native boost` example. 

**note**:
I originally raised the problem in home-assistant [here](https://github.com/home-assistant/core/issues/98382), but it was quickly closed with the comment:
> Open a Zigbee2MQTT issue for this. The MQTT config must have publish a valid modes configuration. This is not in issue with MQTT or Home Assistant.
I have left a reply stating that Z2M specifically mentions what I did and am waiting for a reply.

### What did you expect to happen?

I expected:
- my hive device to `boost` (it did)
- my HA to not have an error in it's logs 

The HA log error was:
```yaml
Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:818 
Integration: MQTT (documentation, issues) 
First occurred: 11:56:03 (10 occurrences) 
Last logged: 11:56:10

Invalid modes mode: emergency_heating
```

### How to reproduce it (minimal and precise)

I used this HA script:
```yaml
script: 
  hive_heating_set_mode_boost:
    alias: "Hive heating: ➜ Boost"
    sequence:
      - service: mqtt.publish
        data:
          topic: zigbee2mqtt/Hive/heat/set
          payload: |-
            {
              "system_mode":"emergency_heating",
              "temperature_setpoint_hold":"1",
              "temperature_setpoint_hold_duration":"30",
              "occupied_heating_setpoint":"25"
            }
    icon: mdi:radiator
    mode: queued
```

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20221226

### Adapter

Sonoff dongle-P

### Debug log

There are no errors in Z2M, only in HA.

```log
Debug 2023-08-14 18:08:06Received MQTT message on 'zigbee2mqtt/Hive/heat/set' with data '{ "system_mode":"emergency_heating", "temperature_setpoint_hold":"1", "temperature_setpoint_hold_duration":"30.0", "occupied_heating_setpoint":"24.3" }'
Debug 2023-08-14 18:08:06Publishing 'set' 'system_mode' to 'Hive'
[... 5 lines cut ...]
Info 2023-08-14 18:08:07MQTT publish: topic 'zigbee2mqtt/Hive', payload '{"linkquality":51,"local_temperature_heat":23.82,"local_temperature_water":21,"occupied_heating_setpoint":"16.0","occupied_heating_setpoint_heat":18,"occupied_heating_setpoint_water":22,"running_state_heat":"idle","running_state_water":"idle","system_mode":"off","system_mode_heat":"emergency_heating","system_mode_water":"off","temperature_setpoint_hold_duration_heat":30,"temperature_setpoint_hold_duration_water":0,"temperature_setpoint_hold_heat":true,"temperature_setpoint_hold_water":true}'
Debug 2023-08-14 18:08:07Publishing 'set' 'occupied_heating_setpoint' to 'Hive'
Debug 2023-08-14 18:08:07Received Zigbee message from 'Hive', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":2450}' from endpoint 5 with groupID 0
Info 2023-08-14 18:08:07MQTT publish: topic 'zigbee2mqtt/Hive', payload '{"linkquality":51,"local_temperature_heat":23.82,"local_temperature_water":21,"occupied_heating_setpoint":"16.0","occupied_heating_setpoint_heat":24.5,"occupied_heating_setpoint_water":22,"running_state_heat":"idle","running_state_water":"idle","system_mode":"off","system_mode_heat":"emergency_heating","system_mode_water":"off","temperature_setpoint_hold_duration_heat":30,"temperature_setpoint_hold_duration_water":0,"temperature_setpoint_hold_heat":true,"temperature_setpoint_hold_water":true}'
Info 2023-08-14 18:08:07MQTT publish: topic 'zigbee2mqtt/Hive', payload '{"linkquality":51,"local_temperature_heat":23.82,"local_temperature_water":21,"occupied_heating_setpoint":"16.0","occupied_heating_setpoint_heat":"24.3","occupied_heating_setpoint_water":22,"running_state_heat":"idle","running_state_water":"idle","system_mode":"off","system_mode_heat":"emergency_heating","system_mode_water":"off","temperature_setpoint_hold_duration_heat":30,"temperature_setpoint_hold_duration_water":0,"temperature_setpoint_hold_heat":true,"temperature_setpoint_hold_water":true}'
Debug 2023-08-14 18:08:07Received Zigbee message from 'Hive', type 'attributeReport', cluster 'hvacThermostat', data '{"runningState":1}' from endpoint 5 with groupID 0
Info 2023-08-14 18:08:07MQTT publish: topic 'zigbee2mqtt/Hive', payload '{"linkquality":51,"local_temperature_heat":23.82,"local_temperature_water":21,"occupied_heating_setpoint":"16.0","occupied_heating_setpoint_heat":"24.3","occupied_heating_setpoint_water":22,"running_state_heat":"heat","running_state_water":"idle","system_mode":"off","system_mode_heat":"emergency_heating","system_mode_water":"off","temperature_setpoint_hold_duration_heat":30,"temperature_setpoint_hold_duration_water":0,"temperature_setpoint_hold_heat":true,"temperature_setpoint_hold_water":true}'
Debug 2023-08-14 18:08:24Received Zigbee message from 'Bedroom 2 motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":14910}' from endpoint 2 with groupID 0
Info 2023-08-14 18:08:24MQTT publish: topic 'zigbee2mqtt/Bedroom 2 motion', payload '{"battery":69,"illuminance":14910,"illuminance_lux":31,"led_indication":null,"linkquality":36,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":19.51,"update":{"installed_version":1107323831,"latest_version":1107323831,"state":"idle"},"update_available":false}'
Debug 2023-08-14 18:08:30Received MQTT message on 'zigbee2mqtt/Hive/heat/set' with data '{ "system_mode":"heat", "temperature_setpoint_hold":"1", "occupied_heating_setpoint":"18.0" }'
Debug 2023-08-14 18:08:30Publishing 'set' 'system_mode' to 'Hive'
Debug 2023-08-14 18:08:30Received Zigbee message from 'Hive', type 'attributeReport', cluster 'hvacThermostat', data '{"systemMode":4}' from endpoint 5 with groupID 0
Info 2023-08-14 18:08:30MQTT publish: topic 'zigbee2mqtt/Hive', payload '{"linkquality":58,"local_temperature_heat":23.82,"local_temperature_water":21,"occupied_heating_setpoint":"16.0","occupied_heating_setpoint_heat":"24.3","occupied_heating_setpoint_water":22,"running_state_heat":"heat","running_state_water":"idle","system_mode":"off","system_mode_heat":"heat","system_mode_water":"off","temperature_setpoint_hold_duration_heat":30,"temperature_setpoint_hold_duration_water":0,"temperature_setpoint_hold_heat":true,"temperature_setpoint_hold_water":true}'
Debug 2023-08-14 18:08:30Publishing 'set' 'temperature_setpoint_hold' to 'Hive'
Debug 2023-08-14 18:08:30Received Zigbee message from 'Hive', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1800}' from endpoint 5 with groupID 0
Info 2023-08-14 18:08:30MQTT publish: topic 'zigbee2mqtt/Hive', payload '{"linkquality":51,"local_temperature_heat":23.82,"local_temperature_water":21,"occupied_heating_setpoint":"16.0","occupied_heating_setpoint_heat":18,"occupied_heating_setpoint_water":22,"running_state_heat":"heat","running_state_water":"idle","system_mode":"off","system_mode_heat":"heat","system_mode_water":"off","temperature_setpoint_hold_duration_heat":30,"temperature_setpoint_hold_duration_water":0,"temperature_setpoint_hold_heat":true,"temperature_setpoint_hold_water":true}'
Debug 2023-08-14 18:08:30Publishing 'set' 'occupied_heating_setpoint' to 'Hive'
Debug 2023-08-14 18:08:30Received Zigbee message from 'Hive', type 'attributeReport', cluster 'hvacThermostat', data '{"runningState":0}' from endpoint 5 with groupID 0
Info 2023-08-14 18:08:30MQTT publish: topic 'zigbee2mqtt/Hive', payload '{"linkquality":51,"local_temperature_heat":23.82,"local_temperature_water":21,"occupied_heating_setpoint":"16.0","occupied_heating_setpoint_heat":18,"occupied_heating_setpoint_water":22,"running_state_heat":"idle","running_state_water":"idle","system_mode":"off","system_mode_heat":"heat","system_mode_water":"off","temperature_setpoint_hold_duration_heat":30,"temperature_setpoint_hold_duration_water":0,"temperature_setpoint_hold_heat":true,"temperature_setpoint_hold_water":true}'
Info 2023-08-14 18:08:30MQTT publish: topic 'zigbee2mqtt/Hive', payload '{"linkquality":51,"local_temperature_heat":23.82,"local_temperature_water":21,"occupied_heating_setpoint":"16.0","occupied_heating_setpoint_heat": [... cut ...]
Debug 2023-08-14 18:08:38Received Zigbee message from 'Front motion', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":35056}' from endpoint 2 with groupID 0
Info 2023-08-14 18:08:38MQTT publish: topic 'zigbee2mqtt/Front motion', payload '{"battery":100,"illuminance":35056,"illuminance_lux":3203,"led_indication":null,"linkquality":43,"motion_sensitivity":"high","occupancy":false,"occupancy_timeout":0,"temperature":22.9,"update":{"installed_version":1107323831,"latest_version":1107323831,"state":"idle"},"update_available":false}'
```

RedNo7, 2023-08-16:
OK, I think I have found the cause of the problem.

Under Hive > devices > esposes, I see this:
[image]

1. Both system_modes should have an option for `emergency_heating`
2. Pressing any of the buttons (pics below) does not result in the right thing happening on the device (although when I run my scripts per the SLR2b page, it works fine:
[image]
[image]

Is it possible that the **SLR2c** device is incorrectly configured and needs to be an identical copy of the  **SLR2b** and **SLR2** devices (they are all the same, do the same thing, just newer versions over time).  If so, this is a simple fix, but I do not have the expertise to work it correctly.

RedNo7, 2023-08-16:
@robertalexa I hope you don't mind me tagging you directly but, having searched for Hive and SLR to try and find out who/how the SLR2s were initially set up with Z2M, I can see you were the guy and am hoping you can help me here.

To the best of my knowledge, the SLR2, SLR2b and SLR2c are identical in functionality.

robertalexa, 2023-08-16:
Hi Red.

Currently away from home so cant provide easy links to things.

The misbehaviour is knows. Emergency heating is not a zigbee standard on how boosting should be performed. Hive are deviating from tve standard here.

When it comes to home assistant, they are respecting the standards and thus this mode is invalid. There is no real workaround either. In fact, boosting should be done via.... I dont remember the name.... Profile, state, preset, sth like that.

Ps. Go to zigbee2mqtt converter repository and look gor my pr on the hive remotes. I am providing details there around this non standard mode implementation.

I know the above soesn4 directly resolve your problem, but it is known and expected misbehaviour between zigbee2mqtt which implements the devices properties and home assistant which implements much closer to zigbee standards

RedNo7, 2023-08-16:
Thanks - I really appreciate your reply; I will definitely go and look at your PR 👍 

If I understand you correctly:
- All 3 units - SLR2, SLR2b, SLR2C - all behave the same way
- There is nothing to fix as this is a Hive unique deviation from signee standards

Is it OK to use the commands per the SLR2b page (example below) and just ignore the error in HA logs?
```yaml
{
  "system_mode":"emergency_heating",
  "temperature_setpoint_hold":"1",
  "temperature_setpoint_hold_duration":"30",
  "occupied_heating_setpoint":"25"
}
```
...or is it better to not use `emergency_heating` and use automations within HA to simulate it instead?

RedNo7, 2023-08-17:
@robertalexa No need to apologise!  ...I appreciated the quick reply!

10hrs - wow - where from and to?

OK, so I have also simulated boost within HA (I don't use Nodered).  I also don't use the thermostat (other than to provide the current temp reading to HA), but it is tethered to the receiver.

Something inside me says that using `emergency_heating` would be more reliable should my HA (running HA Container in docker on a Pi) go down for any reason mid-boost.  ...so I was actively looking at how best to do it.

I am happy to email the yaml for HA control and `emergency_heating` (when it's done) if you are interested?  let me know 👍

closed 2023-08-17 by RedNo7