Koenkk/zigbee2mqtt #18618: Philips Hue bricked by OTA

wbsantos, 2023-08-14:
### What happened?

I have bought 5 Philips Hue bulbs A21 high lumen (model [9290024717](https://www.zigbee2mqtt.io/devices/9290024717.html#philips-9290024717)).

They all paired with zigbee2mqtt through a sonOff dongle. 
And I got it paired with Hue app through Bluetooth.

As they were all new, I was prompted with OTA update in zigbee2mqtt, and I was not careful enough to update them one by one. I just started all of the updates at the same time.

It took just about 6 hours to finish, but three out of the 5 bulbs are completely dead now.

The two the updated correctly are shown in zigbee2mqtt with the new firmware version. The ones that are bricked are offline in zigbee2mqtt, and they are shown with the old firmware version.

The Hue app can't connect with those three through bluetooth.

I have tried a couple of things to reset the three bulbs:
* Turn on the lights for 8 seconds and then off for 2 seconds (repeated 5 times in a row)
* Pressing the buttons "on" and "off/hue" of a hue dimmer very close to the bulbs
* Getting the bulbs close to the coordinator and ran a touchlink scan in zigbee2mqtt
* Reset the bulbs using an Alexa with zigbee, informing the serial number of the bulbs

What I haven't tried yet:
* Reset the bulbs with a Hue bridge. I don't have one and they are kind of expensive just for a test
* Running Hue Thief script (I will try this one, but I waiting a new zigbee coordinator I ordered that is compatible with this solution). But I have little hope that it will work

I can't blame zigbee2mqtt because I have no idea of what happened. And even if it is some bug here, this still a free and amazing software. I just want to report this problem so other people may be more careful than I when updating with OTA.
And if anyone has any tip of how I can get my faulty bulbs to work again, it will be much appreciated (this things are really expensive).

### What did you expect to happen?

I expected the bulbs to be updated successfully. 

### How to reproduce it (minimal and precise)

Update up to 5 devices firmware at the same time.

To be honest I have no idea of what caused the problem. It might have been zigbee2mqtt, my sonoff coordinator, the firmware, the bulbs themselves...

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee-Zigbee 3.0 USB Dongle Plus

### Debug log

The debug log wasn't on during the update.

uli-rpi, 2023-08-14:
Use an echo speaker with an integrated ZigBee hub. These are currently, for example, the large Echo Show or the Echo Plus of the second generation.

If you meet these requirements, open the Alexa app and start the search for a new Philips Hue lamp in the device menu, which of course cannot be found. In the following menu you then tap on "Get help with connecting" and there you will find the action "Reset Philips Hue lamp".

In the following display you have to enter the six-digit serial number, which can be found directly on the bulb, in the Alexa app and can then reset the lamp. It should then be found immediately via the search, whether via the Philips Hue Bridge or another ZigBee hub.

wbsantos, 2023-08-14:
> Use an echo speaker with an integrated ZigBee hub. These are currently, for example, the large Echo Show or the Echo Plus of the second generation.
>
> If you meet these requirements, open the Alexa [...]

I have tried it already, I have an Echo 4th generation (https://www.amazon.com/Echo-4th-Gen/dp/B07XKF5RM3).
Nothing happens, not even a blink of the bulb.

By the way, the bulb doesn't turn on when plugged, and I haven't changed its "power on" settings from the "turn on" by default.

WFLM, 2023-08-17:
I've just faced exactly the same situation with [Hue white and color ambiance E26/E27 1600lm](https://www.zigbee2mqtt.io/devices/929002471601.html#philips-929002471601) and want to find a solution.
I'd bought 2 lamps and decided to update them. One of them got 1.104.2 without any issues, the second one was bricked after the update and it doesn't give signs of life. 

The adapter, Adapter's FW, and Z2M versions are the same.

I've already tried Touchlink, a Direct Bluetooth connection from the smartphone.

maartenhbe, 2023-08-17:
Had the same issue with one of my bulbs. I bought a dimmer and was able to reset the bulb using the dimmer so there seems to be a way out (at least it was for me). 

The strange thing is that whenever I added the once-bricked bulb to my network, others bulbs would randomly go offline after a couple of hours while it was a stable network for months before that. Currently not using that bulb anymore and all stable but unable to force re-apply the OTA update (which might help and maybe is a concrete feature to add?).

deviantintegral, 2023-09-11:
I also ran into this with a [Philips 9290024683](https://www.zigbee2mqtt.io/devices/9290024683.html) Hue white ambiance A19 1100lm with Bluetooth. One bulb updated fine, the other has stopped working completely. The dimmer switch didn't work to reset. Hue support suggested this reset procedure (which didn't work):

> Reset with wall switch (this will reset all lights controlled by that wall switch):
> 
>   - Turn the switch off, wait 5 seconds, turn it on, wait 5 seconds, turn it off again.
>   - Repeat this process at least 5 times.
>   - The light will blink if the process was done correctly.
>   - Repeat the process for all the lights, if needed.

I suppose it's possible there's some hardware issue that breaks updates, especially given I've never had trouble on any previous bulb.

wernerdaehn, 2023-12-29:
Same here. When turning the light on via the 230V power switch, they remain dark. Unresponsive to dimmer switch reset, bluetooth, and hue bridge. Happened to two out of six.

DaleJP, 2024-02-21:
The same thing happened to me. I banged it on the table a couple of times, and it started working. No lie. 

Bizarre.

wernerdaehn, 2024-02-21:
Bricked one more. This time I updated via bluetooth and then the lights got the signal to turn off. I wonder if it's that - getting a command during the update.

matiasnl23, 2024-05-28:
I'm reading this while my Hue LightStrip is updating. Thankfully, I started with one light instead of updating all of them at once. It's my first time updating it with zigbee2mqtt. Previously, I've always used the Hue Bridge. I'm hoping the update goes smoothly. I'm not sure if I'll proceed with updating the other lights...

wbsantos, 2024-05-28:
> I'm reading this while my Hue LightStrip is updating. Thankfully, I started with one light instead of updating all of them at once. It's my first time updating it with zigbee2mqtt. Previously, I've [...]

What I usually do nowadays is to pair the light to the smartphone app using Bluetooth, I then update the firmware through Bluetooth instead of through ZigBee.

matiasnl23, 2024-05-28:
> What I usually do nowadays is to pair the light to the smartphone app using Bluetooth, I then update the firmware through Bluetooth instead of through ZigBee.

Thanks for the recommendation. I'll do that. I hope it takes less time than through the Zigbee network.

wbsantos, 2024-05-28:
> > What I usually do nowadays is to pair the light to the smartphone app using Bluetooth, I then update the firmware through Bluetooth instead of through ZigBee.
> 
> Thanks for the recommendation. I'll do that. I hope it takes less time than through the Zigbee network.

It does. Through Bluetooth takes a couple of minutes (5 at maximum). Through ZigBee I found that takes about an hour. It might be a problem with the protocol or with my mesh network.

calos9999, 2024-06-18:
Hi I was upgrading my bulb to the latest firmware version.
Everything seemed to go smoothly, the bulb rebooted as it usually does, but since then it can't connect to bluetooth
It tries but can't
Any idea?

WFLM, 2024-06-19:
Hi @calos9999 
Have you tried to reset the bulb via Touchlink?

calos9999, 2024-06-19:
> Hi @calos9999 
> Have you tried to reset the bulb via Touchlink?

No, touchlink is part of “Hue Essential” app?
I have to unpair the bulb from my phone?

WFLM, 2024-06-19:
> > Hi @calos9999 
> > Have you tried to reset the bulb via Touchlink?
> 
> No, touchlink is part of “Hue Essential” app?
> I have to unpair the bulb from my phone?

No, you can find it in Zigbee2MQTT. 
You need to power on the lamp and place it a few centimeters from your Zigbee adapter. Then, use Touchlink in Z2M to find and reset the lamp.

calos9999, 2024-06-19:
> No, you can find it in Zigbee2MQTT.
> You need to power on the lamp and place it a few centimeters from your Zigbee adapter. Then, use Touchlink in Z2M to find and reset the lamp.

I need to setup a server for that? Lol

wernerdaehn, 2024-08-11:
> Bricked one more. This time I updated via bluetooth and then the lights got the signal to turn off. I wonder if it's that - getting a command during the update.

I felt brave and updated some more Hue lights. No issues. So either something changed or my suspicion that any command during an update bricks the hue light is correct.

Before I updated approx 5 devices and 3 got bricked.
Now I updated 20 devices without a single issue. I disable all automations and make sure nobody is using the lights and then click on update in the ZigBee2MQTT --> OTA screen one after the other.
Each device takes approx 30 minutes to update and if I update two simultaneously, the time estimate doubles. Most of the time the send is interrupted when updating multiple in parallel and a new update picks up where the send was interrupted, e.g. with 20%. Hence updating one after the other.

My suggestions:

1. Even if the user updates multiple, queue the requests and update a single one at any time. This way I can queue to update all 100 lights and don't have to watch the updates myself.
2. During the update do not let any commands pass to the lights.
3. Document this issue with hue lights!

doff-1, 2024-11-30:
> What I usually do nowadays is to pair the light to the smartphone app using Bluetooth, I then update the firmware through Bluetooth instead of through ZigBee.

Does this affect how the light is integrated in home assistant? does it disappear and need to be re-added?
Or do you just connect via bluetooth, and update the firmware without Home assistant and the zigbee network being affected at all?

WFLM, 2024-11-30:
> Does this affect how the light is integrated in home assistant? does it disappear and need to be re-added?
> 
> Or do you just connect via bluetooth, and update the firmware without Home assistant and the zigbee network being affected at all?

You can use it in parallel without any issues. I set up the light alarm clock via Bluetooth, update it the same way, and also use it via Zigbee (Z2M) without any problems. The only issue is that if you change the lamp status via Bluetooth, it won’t be reflected in Z2M immediately.

deviantintegral, 2024-12-01:
I have done many firmware updates in the past year (say 3 updates times 20 devices) without losing any of them. Given that this issue hasn't blown up with hundreds of replies, it's entirely possible that there was an actual hardware fault in the one bulb of mine that died, and that's what we're all facing.

Updating over bluetooth is much faster, but note you won't always get the same firmware as what's in the OTA index.

Frans-Willem, 2025-01-06:
Any updates on this? I've updated 11 bulbs last week, 4 of which were bricked.
I've tried the 2-sec-off-8-sec-on, tried connecting to them over Bluetooth, and tried resetting them with a Hue remote, but nothing gets them to turn on or communicate anymore.

doff-1, 2025-01-08:
How did you update them?
I ended up updating mine one by one, making sure the device didn't get any other commands while updating. Of my 65 hue devices I thought I bricked one, but I got it back again. And it was after that I made sure to not send the light any commands while updating.

amondit, 2025-01-10:
[_zigbee2mqtt_logs(1).txt](https://github.com/user-attachments/files/18377769/_zigbee2mqtt_logs.1.txt)
Hello,

I've just migrated my zigbee devices from ZHA to Z2M (to a Sonoff Dongle-e with firmware 8.0.2), and noticed that my original Philips Hue bulbs (LCT001, 1st version, pre-bluetooth, probably 12 years old) showed an OTA update available (ZHA did not show such update available).

I've upgraded one but then the bulb is not working anymore. It still lights up when powered on, correctly resets using touchlink (using a hue dimmer), connects to the network but interview fails at some point (`Interview failed because can not get active endpoints`) so I can't get the bulb to stick to the network (gets kicked out by z2m it seems, it disappears from devices list).

I figured it may be bad luck, something happened during ota process, or maybe just bulb ROM was too old and got corrupted, and will use the bulb as a dumb bulb from now on).

Then I tried to update the 2nd bulb and the same thing happened (same end result, bulb powers on but I can't get it to the network properly). Tried to power off/on again and that doesn't seem to change anything.

Attached logs, but basically what happens is:
Checking OTA
```
[2025-01-09 15:53:33] info: 	z2m: Checking if update available for 'PHILIPS_E27_Color_1'
[2025-01-09 15:53:34] info: 	z2m: Update available for 'PHILIPS_E27_Color_1'
[2025-01-09 15:53:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PHILIPS_E27_Color_1', payload '{"color_mode":"xy","linkquality":136,"power_on_behavior":null,"state":"ON","update":{"installed_version":1107322837,"latest_version":1124103171,"state":"available"},"update_available":null}'
[2025-01-09 15:53:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"PHILIPS_E27_Color_1","updateAvailable":true},"status":"ok","transaction":"49vji-4"}'
```
OTA started
```
[2025-01-10 00:03:30] info: 	z2m: Updating 'PHILIPS_E27_Color_1' to latest firmware
[2025-01-10 00:03:37] info: 	z2m: Update of 'PHILIPS_E27_Color_1' at 0.00%
[2025-01-10 00:03:37] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PHILIPS_E27_Color_1', payload '{"color_mode":"xy","linkquality":216,"power_on_behavior":null,"state":"OFF","update":{"installed_version":1107322837,"latest_version":1124103171,"progress":0,"state":"updating"},"update_available":null}'
[2025-01-10 00:04:08] info: 	z2m: Update of 'PHILIPS_E27_Color_1' at 2.35%, ≈ 26 minutes remaining
...
```
OTA completes, reconfigures devices correctly but then fails ping 10 minutes later (availability check)
```
...
[2025-01-10 00:26:20] info: 	z2m: Finished update of 'PHILIPS_E27_Color_1'
[2025-01-10 00:26:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PHILIPS_E27_Color_1', payload '{"color_mode":"xy","linkquality":212,"power_on_behavior":null,"state":"OFF","update":{"installed_version":1124103171,"latest_version":1124103171,"state":"idle"},"update_available":null}'
[2025-01-10 00:26:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"PHILIPS_E27_Color_1","ieee_address":"0x0017880100d655f3"},"type":"device_announce"}'
[2025-01-10 00:26:25] info: 	z2m: Device 'PHILIPS_E27_Color_1' was updated from '{"dateCode":"20181129","softwareBuildID":"5.127.1.26581"}' to 'undefined'
[2025-01-10 00:26:25] info: 	z2m: Configuring 'PHILIPS_E27_Color_1'
[2025-01-10 00:26:25] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20181129","software_build_id":"5.127.1.26581"},"id":"PHILIPS_E27_Color_1","to":null},"status":"ok","transaction":"k8l9y-3"}'
[2025-01-10 00:26:30] error: 	z2m: Failed to read state of 'PHILIPS_E27_Color_1' after reconnect (ZCL command 0x0017880100d655f3/11 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '18763'.))
[2025-01-10 00:26:34] info: 	z2m: Successfully configured 'PHILIPS_E27_Color_1'
[2025-01-10 00:36:26] warning: 	z2m: Failed to ping 'PHILIPS_E27_Color_1' (attempt 1/2, ZCL command 0x0017880100d655f3/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '18763'.))
```
And now, if I manually trigger a 'reconfigure' action on the bulb, it completes, but if I try to trigger again the device interview, it fails to get active endpoints:
```
[2025-01-10 06:40:55] info: 	z2m: Configuring 'PHILIPS_E27_Color_1'
[2025-01-10 06:40:59] info: 	z2m: Successfully configured 'PHILIPS_E27_Color_1'
[2025-01-10 06:40:59] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"PHILIPS_E27_Color_1"},"status":"ok","transaction":"i0d2e-1"}'
...
[2025-01-10 07:02:27] error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'PHILIPS_E27_Color_1' (0x0017880100d655f3) failed: Error: Interview failed because can not get active endpoints ('0x0017880100d655f3')'
```

I hope my devices are recoverable (if not then at least I'll use them as dumb lights, I have a couple places where I don't need brightness or color to be modified), but at least I'd hope for others to not have their bulbs have the same fate than mine.
It's really strange that such old bulbs were offered an OTA update after so long (last firmware was in 2018!), it probably was a compatibility configuration mistake?

closed 2025-03-20 by github-actions[bot]