Koenkk/zigbee2mqtt #18619: ELKO 4523430 (ESH Plus Super TR RF PH) sends update to z2m every second.

glennher, 2023-08-14:
### What happened?

I have a network with about 12 thermostates, and with with all of them updating the system every second they flood the adapter and after a while make it crash. When this happends the only solution so far is to unplug the usb adapter and replug it again and restart zigbee2mqtt.

Reference to same issue that is closed without any solutions:
#16142

### What did you expect to happen?

- Be able to run a stable network with minimum report interval that does not overload the adapter.
- Be able to adjust minimum report interval.

### How to reproduce it (minimal and precise)

Connect a device and look at the logs, try to adjust minimum report interval and see if reporting changes.

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

CC2652R_coordinator_20230507

### Adapter

Electrolama zzh CC2652

arnerek, 2023-09-14:
Anyone found the reason for this?

glennher, 2023-09-14:
I suspect the root cause might be related to the thermostat firmware rather than the adapter. I have thermostats with firmware versions 1.2.0R and 1.2.4R that exhibit this flooding behavior. However, a new thermostat with firmware version 1.2.5R does not seem to have this issue. I've reached out to Elko support to inquire about firmware updates for the thermostats but have not received any response yet.

arnerek, 2023-09-15:
That’s interesting. I have 1.2.5R on two devices and experience the
flooding unless I comment the subscription code.

tor. 14. sep. 2023 kl. 19:37 skrev Glenn Hermansen ***@***.***

arnerek, 2023-10-24:
I checked in the dev console: Could this shed some light?

I think the attributes are incorrectly converted to a number?

[image]

glennher, 2023-10-24:
I think you are on to something.

In the reporting section of the device, the 1025 attribute cannot be controlled or changed intervals on (you get an error that it does not exist/is undefined). However if you change the 1025 attribute to the corresponding "elkoLoad" instead, you are able to control the timers again, and then you can try to set "min rep interval" to 60 seconds and apply.
It will now report only once per minute.

arnerek, 2023-10-24:
I finally managed to progress here:
1) To get the reporting every second to disappear, set reportableChange to 1 in https://github.com/Koenkk/zigbee-herdsman-converters/blob/06ff10bd56b50e402ecd51740df9d5cfbc5ffc99/src/devices/elko.ts#L97
2) The Elko attributes are not recognized in the reporting section, e.g. elkoLoad is mapped directly to 1025. If I remove the manufacturerCode: manufacturerCode_1.default.ELKO from https://github.com/Koenkk/zigbee-herdsman/blob/4a929684f58eb1e57b998f9cae0da8420dafce32/src/zcl/definition/cluster.ts#L2040 - all is good! I don't know if this has side-effects for other devices, but at least it is working for me

closed 2023-10-25 by Koenkk