Koenkk/zigbee2mqtt #18620: [New device support]: Schneider Electric Wiser 40/300-Series Module AC Fan Controller

lucasteligioridis, 2023-08-15:
### Link

https://www.clipsal.com/products/detail?CatNo=41ECSFWMZ&itemno=41ECSFWMZ-VW

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x8cf681fffe7dd8c8","nwkAddr":10837,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"CHFAN/SWITCH/1","epList":[7,242],"endpoints":{"7":{"profId":260,"epId":7,"devId":776,"inClusterList":[0,3,4,5,6,514,2821,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"CHFAN/SWITCH/1","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.002.000 D"}},"hvacFanCtrl":{"attributes":{"fanMode":1}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":514,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8b6ab","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8b6ab","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":514,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.002.000 D","zclVersion":3,"interviewCompleted":true,"meta":{"configured":246586501},"lastSeen":1692071424053,"defaultSendRequestWhen":"immediate"}

### Comments

Hi, I have attempted to add an integration for this new Clipsal (Schneider Electric) fan controller and there are just a few things I am getting stuck with.

As you can see below I have done some sort of analysis and got the below details which is working in z2m!
A few issues I just don't know how to analysis in z2m or whether its somewhat possible?

For the standard Clipsal light switches, they have this LED indicator mode and its based off endpoint 21 (which I have the code in my below example, I just didn't add it in because it's a copy paste from the device integration):
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/schneider_electric.js#L25C11-L25C11

The only issue is my device only has endpoints 7 and 242, it doesn't have this endpoint 21 or perhaps another one to change the LEDs. In the Wise app I added the fan controller to see what is available and it SHOULD have 2 additional settings with the options listed below:
- LED Indicator Mode:
  - Always On
  - On with Timeout but as Locator
  - On with Timeout
- LED Indicator Orientation
  - Horiztonal Left
  - Horiztonal Right
  - Vertical Top
  - Vertical Bottom 

I am more than happy to dive deep into this and finish this integration but I'm not really sure about how the endpoints work here? If the endpoint is not available in z2m is that something I can make available somehow by adding code somewhere? Or are there other changes or limitations that I am unable to see?

There is also this https://github.com/Koenkk/zigbee2mqtt/discussions/18613 discussion which talks about presets, but with this fan controller there are only 4 modes:
- off
- low
- medium
- high

So being able to do this as well would make it neater to use. Either way, I'm more than happy to do the work here, I just don't know if there are limitations somewhere or whether I need to add something to get the additional endpoints this device has into z2m

```js
...
const definition = {
    zigbeeModel: ['CHFAN/SWITCH/1'],
    model: '41ECSFWMZ-VW',
    vendor: 'Schneider Electric',
    description: 'Wiser 40/300-Series Module AC Fan Controller',
    fromZigbee: [fz.on_off, fz.fan, fzLocal.indicator_mode],
    toZigbee: [tz.fan_mode, tz.on_off, tzLocal.indicator_mode],
    exposes: [e.fan().withModes(['off', 'high', 'medium', 'low', 'on']), exposesLocal.indicator_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(7);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'hvacFanCtrl']);
        await reporting.onOff(endpoint);
        await reporting.fanMode(endpoint);
    },
};

module.exports = definition;
```

lucasteligioridis, 2023-08-30:
Just checking back in to see if there is anything I could do to debug this further, much appreciated :)

lucasteligioridis, 2023-09-07:
Just checking back in again :)

lucasteligioridis, 2023-09-18:
Just checking in and keeping issue from going stale

mn7474, 2023-10-14:
I’m using these devices as well, and interested in any help to implement

Koenkk, 2023-10-15:
@lucasteligioridis this device doesn't have an endpoint 21, otherwise you would have seen it in the `epList`, endpoint 242 is for Zigbee green power so you can ignore that one too. Did you try to set the indicator mode on endpoint 7 already?

lucasteligioridis, 2023-10-18:
@Koenkk I tried endpoint 7 but didn't work using the same code in the integration.
To be fair I'm not sure if using the existing code and just changing the endpoint for the indicator mode to 7 will actually work as the data structure looks to be different as explained in the issue. More options for this fan switch.

I don't have enough understanding of how to debug this process further, I'm happy to trial and error a few things, I just don't quite understand which knobs and switches I need to turn to get a feedback cycle here.

This is the error I get when using endpoint 7 on the existing code base:
```
Zigbee2MQTT:error 2023-10-18 10:50:06: Publish 'set' 'indicator_mode' to 'GuestRoom/Fan/Ceiling/Blades' failed: 'Error: Write 0x8cf681fffe7dd8c8/7 elkoSwitchConfigurationClusterServer({"0":{"value":2,"ty
pe":48}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4190,"
transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:debug 2023-10-18 10:50:06: Error: Write 0x8cf681fffe7dd8c8/7 elkoSwitchConfigurationClusterServer({"0":{"value":2,"type":48}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,
"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4190,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUP
PORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:420:28)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:489:22)
    at Object.convertSet (/app/data/extension/externally-loaded.js:33:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:36)
```

Koenkk, 2023-10-18:
@lucasteligioridis so probably we have to send something different indeed, to find this out you have to sniff the traffic when setting this option via the original gateway (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html)

lucasteligioridis, 2023-10-18:
Thanks @Koenkk 
I was afraid you were going to say that and was hoping for an easy magical option 😆 

So I assume to get all this data for things like this, this is the process you have to undertake for a new device and/or addition?

Assume I'll have to connect my fan to the Wise Hub and sniff the zigbee traffic when activating all the new LED indicator options, record those all down then use that data in the payloads within the z2m converter?

Koenkk, 2023-10-20:
> sniff the zigbee traffic when activating all the new LED indicator options, record those all down then use that data in the payloads within the z2m converter?

Exactly!

lucasteligioridis, 2023-10-22:
okay no worries, I'll find sometime over the next week or so to use my spare zigbee stick and flash it with the sniffer and then come back here when I get some results to figure out how to plug those into the code base! 

Thanks for the help so far :)

bjammin, 2023-10-31:
Apparently this is working in ZHA if that is any help.

lennon101, 2023-10-31:
jeeze this whole process sounds like an incredible journey. I thank all those who are investing the time in this and look forward to the finished solution implemented! <3

avalonea, 2023-11-13:
Hi all, just came across this issue myself. Anyway to get some basic functionality for the fans until the issue is resolved? 
thanks!

lucasteligioridis, 2024-01-01:
> @lucasteligioridis so probably we have to send something different indeed, to find this out you have to sniff the traffic when setting this option via the original gateway (https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html)

I finally got around to some time to start looking into this, turns out my SONOFF stick is CC2652P and not compatible with those instructions, for packet sniffer 1. Might have to buy another device to get this working, can't see any simple instructions on how to flash this one with a sniffer to get these answers.

lucasteligioridis, 2024-01-01:
> Hi all, just came across this issue myself. Anyway to get some basic functionality for the fans until the issue is resolved? thanks!

Hey @avalonea feel free to use this in the mean time as a manual import, it gives basic functionality and I've mostly copied the code from the original schneider lib:
```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const constants = require('zigbee-herdsman-converters/lib/constants');

const e = exposes.presets;
const ea = exposes.access;

const exposesLocal = {
    indicator_mode: exposes.enum('indicator_mode', ea.ALL, ['on_with_timeout_but_as_locator', 'on_with_timeout', 'always_on'])
        .withDescription('Led Indicator Mode'),
};

const tzLocal = {
    indicator_mode: {
        key: ['indicator_mode'],
        convertSet: async (entity, key, value, meta) => {
            const endpoint = entity.getDevice().getEndpoint(7);
            const lookup = {'on_with_timeout_but_as_locator': 2, 'on_with_timeout': 0, 'always_on': 1};
            utils.validateValue(value, Object.keys(lookup));
            await endpoint.write(0xFF17, {0x0000: {value: lookup[value], type: 0x30}}, {manufacturerCode: 0x105e});
            return {state: {indicator_mode: value}};
        },
        convertGet: async (entity, key, meta) => {
            const endpoint = entity.getDevice().getEndpoint(7);
            await endpoint.read(0xFF17, [0x0000], {manufacturerCode: 0x105e});
        },
    },
};

const fzLocal = {
    indicator_mode: {
        cluster: '64516',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            const lookup = {2: 'on_with_timeout_but_as_locator', 0: 'on_with_timeout', 2: 'always_on'};
            if ('indicator_mode' in msg.data) {
                result.indicator_mode = lookup[msg.data['indicator_mode']];
            }
            return result;
        },
    },
};

const definition = {
    zigbeeModel: ['CHFAN/SWITCH/1'],
    model: '41ECSFWMZ-VW',
    vendor: 'Schneider Electric',
    description: 'Wiser 40/300-Series Module AC Fan Controller',
    fromZigbee: [fz.on_off, fz.fan, fzLocal.indicator_mode],
    toZigbee: [tz.fan_mode, tz.on_off, tzLocal.indicator_mode],
    exposes: [e.fan().withModes(['off', 'high', 'medium', 'low', 'on']), exposesLocal.indicator_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(7);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'hvacFanCtrl']);
        await reporting.onOff(endpoint);
        await reporting.fanMode(endpoint);
    },
};

module.exports = definition;
```

Koenkk, 2024-01-02:
@lucasteligioridis I've never tried, but check out packet sniffer 2

lucasteligioridis, 2024-01-04:
> @lucasteligioridis I've never tried, but check out packet sniffer 2

Yeah I'll give that one a shot I did some light reading on that, do you mind me asking what you have used on what platform? i.e. Mac OS/Linux and what stick?

Koenkk, 2024-01-04:
@lucasteligioridis Mac with a CC2531.

lucasteligioridis, 2024-01-04:
Thanks @Koenkk will try packet sniffer 2 and if I get no where I'll just go buy a CC2531 for this purpose... any particular one you suggest?

lucasteligioridis, 2024-01-06:
@Koenkk potentially for another issue but you may at least be able to help me if there is a "quick answer", I have tried digging in the code base, but not sure where to look exactly but this line here:
```
    exposes: [e.fan().withModes(['off', 'high', 'medium', 'low', 'on']), exposesLocal.indicator_mode],
```

Is there anyway for me to remap `on` to one of the other modes somehow? For example I want to remap `on` to the `medium` speed for example, which in this case is actually `fanMode: 2`. I understand the requirement for needing something like this and I believe the official Wiser hub has `on` as `medium` anyway. I just can't see anywhere in the docs that highlights how to do this or if its even possible right now.

Any help appreciated if possible :) thanks

Koenkk, 2024-01-06:
@lucasteligioridis why do you want to do this?

lucasteligioridis, 2024-01-06:
@Koenkk because with the current configuration `on` maps to `fanMode: 4` which doesn't exist on this fan controller, it only has 0, 1, 2 ,3 where 0 is `off`. So the `on` preset just gives an error failing with above.

I'm not sure what other fans do in this case, they would surely have to default to some fan speed, this is just a ceiling fan mind you so there is no such thing as just an `on` state it has to be one of the 3 available speeds. But by not having an `on` mapped to something I can't utilise this in HA or Homekit correctly for a quick on behaviour.

Hope thats clear.

Koenkk, 2024-01-06:
Try with: https://gist.github.com/Koenkk/871b5e61dcb7855dfd9be1462c8e10a6

lucasteligioridis, 2024-01-06:
Got this response:
```Publish 'set' 'fan_state' to 'my_clipsal_fan' failed: 'Error: At least one of "brightness" or "state" must have a value: '{"fan_state":"ON"}''```

Error happens for any state/speed now with the above message.

lucasteligioridis, 2024-01-06:
Ah was wondering what that return was about! 

Great that did exactly what I expected! On goes to `medium` and no longer errors, much cleaner to interact with now for a quick `on` behaviour! Thanks for the quick help here! :)

Wonder if this is worth making a configurable value or just being opinionated or replicating how the default Wiser hub does it? 

Either way, all thats left for this integration is to get the sniffer to figure out the indicator/LEDs like the other wiser switches.
Thanks for all your assistance so far, been great :)

Koenkk, 2024-01-07:
> Wonder if this is worth making a configurable value or just being opinionated or replicating how the default Wiser hub does it?

This is fine

> Thanks for all your assistance so far, been great :)

😄

lucasteligioridis, 2024-01-09:
I had another idea but not sure if this information is actually available... can we just default "on" to whatever the last state was set to from the speeds? I don't know if this is data that is available or whether that state is recorded in such a way to refer back to, this way no opinions are really required, on will just go back to whatever the user had last set their fan speed to and will most likely be the speed they'll go back to.

On another note, how do you feel about me submitting a PR with what we have today so people can use this in the mean time? I'm more than happy to wait until we have this fully done with the LEDs, but just wanted to get an idea from your perspective on it and whether that is something you promote or not :)

Koenkk, 2024-01-09:
> can we just default "on" to whatever the last state 

That would be hard as z2m does not have the state history

> On another note, how do you feel about me submitting a PR with what we have today so people can use this in the mean time? 

Yes, please do so!

alexeiw123, 2024-01-15:
So good to see this work taking place - I just ordered one of these /before/ looking if it was supported.
I see that PR is merged, does that mean it's likely to 'just work' once I pair it later this week?

 What does the 'LED Indicator Orientation' do in the wiser app, and is that supported?

lucasteligioridis, 2024-01-16:
@alexeiw123 assuming that there is a new release by then and you use the latest version yes it will be supported. Can always just switch to the dev branch in the temporary short period before the official release. But it's only around the corner 😀 

In terms of the LED, the orientation allows you to move where you want the light to shine for speeds 1, 2, 3 and off. So you can rotate where you would prefer/like it. Also the ability to turn the light off entirely, if you have other wiser switches it's exactly like that. 

Currently not supported yet, need to get some time to install a packet sniffer and figure out the commands that the fan switch is sending so we can import them into z2m. For now the practical functionality of the switch works perfectly, just no ability to change LED behaviour yet.

alexeiw123, 2024-01-16:
@lucasteligioridis great, thanks. I hadn't seen the dev branch. Neat that I can install them side by side (disabling one or the other) and all config comes across.

I haven't got the wiser hub myself but I did discover that a basic $20 tuya zigbee hub is recognised by the wiser app and able to add devices. Seemed to allow most functionality for the mechs that I have, but no firmware updates. Could be a cheaper route to packet sniffing.

lucasteligioridis, 2024-01-16:
Hmmm well if your fans work with the Tuya hub and you can see all the commands in plaintext somewhere then it might be something that we could use to finalise this integration.

Otherwise you might need the official Wise hub to see the raw commands if that's not possible. Which I have mind you, just need to get some time to flash my other stick and do it.

alexeiw123, 2024-01-16:
> Hmmm well if your fans work with the Tuya hub and you can see all the commands in plaintext somewhere then it might be something that we could use to finalise this integration.
>
> Otherwise you might [...]

Couldn't tell you yet. All my other mechs are dimmers or switches. I'm not using the tuya hub, I bought it to 'try' updating the firmware. I'm using my HA yellow as the zigbee coordinator, so the tuya hub is just in a box. I should have my fan switch this week but I'm also installing the fan too, where there's no crawl space so it might take me a few weeks until I'm ready to test.

alexeiw123, 2024-01-20:
@lucasteligioridis well I got my fan installed and the wiser fan controller working. Switched to the edge add-on and like magic it's working.

Have you found a way to change the button light behaviour, even if just a one off? Bedroom fan, no lights preferred.

I read elsewhere that 10 button presses is supposed to toggle the light mode on the switch mechs, doesn't seem to work here.

lucasteligioridis, 2024-01-20:
@alexeiw123 ah nice great it worked immediately!

I have not had any luck and saw that 10 button thing but read that it gets wiped once you switch back to zigbee mode which is what you probably experienced.

Only time I've ever seen it work is with the Wise hub I have, never without it unfortunately.

alexeiw123, 2024-01-20:
Bummer! Was hoping for some magic in the dev console part but it's way over
my head. Thanks for your great work.

alexeiw123, 2024-01-23:
@lucasteligioridis aha, I figured out a workaround! From page 35 here: https://download.schneider-electric.com/files?p_enDocType=User+guide&p_File_Name=41ECSFWMZ-VW_WSE_DUG_EN.pdf&p_Doc_Ref=41ECSFWMZ-VW_WSE_DUG_EN

"Changing protocol: Short press the push-button 3 times and long press once for > 20 s."

I had been using the 20s press to switch from zigbee to BLE, and 12 rapid presses to switch from BLE to zigbee. The 12 rapid presses clears some settings (not device orientation, but definitely clears led behaviour)

If you change the LED settings in BLE mode, then switch to zigbee by doing 3 quick presses then holding the 4th press until the red light changes from a slow blink to a rapid blink, then the led timeout setting is retained.

lucasteligioridis, 2024-01-23:
@alexeiw123 mate this is fantastic! 
I personally don't care about the orientation but the LED is super annoying in my bedroom at night haha. Thanks for digging into this! I read those docs so many times but must have been too deep into it. 

@Koenkk i think this would be good to put as "notes" on the device support page on the site at least until we get this integration fully complete and in control. Is this something I can edit directly on the site repo or is it needed somewhere else for the way auto generate works?

alexeiw123, 2024-01-23:
> I personally don't care about the orientation but the LED is super annoying in my bedroom at night ha! Thanks for digging into this! I read those docs so many times but must have been too deep into it.

exactly the same for me. We like a dark room - orientation is less important. Happy to tick that minor annoyance.

lucasteligioridis, 2024-01-23:
@alexeiw123 I tried this method just now but everytime I switch back and "pair the fan" back to z2m the settings reset.
Can you be very clear on the steps exactly you took after switching back to zigbee mode please? Something I've missed?

Once the fan is back into z2m I can see the settings reset to default and the LED comes back.

alexeiw123, 2024-01-23:
> Also do you mind telling me the firmware version on your fan switch in z2m please

002.002.000 D

> Can you be very clear on the steps exactly you took after switching back to zigbee mode please?

I think before I switched to BLE mode I had put the fan switch in ZigBee pairing mode, which had removed it from Z2M, although when it was re-added it remember the device name.

In BLE mode, I used the wiser apps to change the settings as I like them. Then I did the 3 button press, hold 4th press until the fast red blinking. While the switch was restarting I put Z2M in to pairing mode - now I can't be sure if it went straight in to ZigBee pairing mode and was detected, or if I need to put it in pairing mode manually (by simply holding the button for about 8 seconds).

It would be frustrating if there's a specific way to pair that avoids a settings reset. Good luck.

lucasteligioridis, 2024-01-23:
My firmware is `002.002.001 R` so if incrementing is newer thats funny... 

I'll give this a shot tomorrow, maybe having it ready in pairing mode is the difference. Every time it "paired" thats when I saw the LED indicator reset back to defaults. I definitely saw it resetting to defaults anytime it went back into Zigbee mode.... so will try this out and see what happens. Thanks for the explanation!

lucasteligioridis, 2024-01-24:
Yeah no dice. It doesn't seem to retain the settings, looks like the 3 button press plus the 4th on long press also resets the settings.

Assume it's something to do with my newer firmware, hence the different behaviour.

Where did you read or see this thing about 12 quick button presses?

alexeiw123, 2024-01-24:
> Yeah no dice. It doesn't seem to retain the settings, looks like the 3 button press plus the 4th on long press also resets the settings.

Are you holding long enough for the fast red blinks? It will first blink red slowly at about 10 seconds (device reset) then blink quickly at around 20 seconds (switch mode).

> Where did you read or see this thing about 12 quick button presses?

I read about that one with my other wiser devices; dimmer and 2AX units. Can't remember if it was on forums or documentation. I just assumed the fan would have the same button patterns and this 12 button press to go to ZigBee mode is indeed there.

lucasteligioridis, 2024-01-25:
Yeah I wait until the fast red blinking and then I let go, then it switches into Zigbee mode. But then once it repairs into z2m it resets? 

I kind of assumed the first phase of the slow blinking perhaps is resetting it before going into the switch. Maybe I'll hold it longer when it flashes red quickly instead of letting go immediately?

alexeiw123, 2024-01-28:
Sorry for the slow response - have been pulled away form home on short notice.

> Yeah I wait until the fast red blinking and then I let go, then it switches into Zigbee mode. But then once it repairs into z2m it resets?

So for me it worked first time and hearing your struggles I'm pretty hesitant to re-do my steps, but I'll do the kids rooms soon. Hopefully we've found the cluster and attributes for led button behaviour by then anyway and the rest process won't matter.

> I kind of assumed the first phase of the slow blinking perhaps is resetting it before going into the switch. Maybe I'll hold it longer when it flashes red quickly instead of letting go immediately?

The docs say that the slow red blinking is device reset, the fast red blinking is changing connection mode. Another thing I did in BLE mode was I changed to timer mode and back to switch mode, just to explore options. I guess it's remotely possible that this action could have changed some configurable on reset behaviour?

alexeiw123, 2024-02-19:
@lucasteligioridis hey do you have any of the 2AX switches? model: CH2AX/SWITCH/1
Have just noticed on latest dev build that the switch is no longer exposed.
Trying to figure out what PR broke it.

lucasteligioridis, 2024-02-19:
Hey @alexeiw123 rather than continuing the chat here on that switch, create a new issue and ping me over there. So we don't hijack this thread with non fan related chat.

alexeiw123, 2024-02-19:
@DanielNagy - While I have your attention on the 2AX issue, and given your involvement in getting the led behaviour working on wiser devices, do you have any insight for @lucasteligioridis or myself in adding support for the button behaviour configuration for the wiser AC fan controller?

I've been able to set under bluetooth mode and retain that setting over on zigbee mode but it was _very_ fiddly. Neither of us have been able to identify the cluster responsible for this.

alexeiw123, 2024-02-20:
> I had started shopping for a outdoor fan for this! LOL

I hear you - I installed a fan in our bedroom and went AC over DC based on this product (plus my hate of 433 MHz remotes).

> Looking at the documentation of the new Fan controller shows there is more light indication options around orientation.

Yes, it has different behaviour and options around the lights compared to the mechs. There are 4x seperate LEDs, one on each edge of the button. It illuminates 1x for low, 2x for medium and 3x for high and then shows all 4x when you turn it off.

There are also options for the orientation of those 4x leds and then if the lights dim after a short time, or turn off completely.

> Of interest, do the current indication options exist as per the 2ax for this new fan controller? Does the current z2m implementation work?

No, these are the exposes (noting this is on stable release).

[image]

> In terms of my original work for the 2AX/10AX Led indication. I purchased a Clipsal ZigBee hub purely for this purpose of sniffing / reverse engineering the Led Indications. (please donate to my paypal so I can recover the absorbent cost of the hub! LOL, jokes).

Oh I've been following along and putting far too much time in finding a way to update my zigbee firmwares without the hub. I bought a generic tuya hub which does get picked up through the wiser app and allow adding of devices, but alas no firmware updates shown. I am currently in the middle of a dispute with Schneider about their prevention of expected functionality hiding being a paid product being in breach of consumer law. Will see how that pans out. I may end up with a hub myself as I'm not 50% through me entire house with wiser and feeling pretty commited.

And you know what, I think I will help as I see the work also on getting OTA updates working, which would really unlock the potential of these mechs.

> What does the configuration page look like for this device in the native app? Is the orientation a separate configuration option?

Yes - here's some screenshots of the bluetooth cofig options. Device is in zigbee mode but I think the screenshots still show enough without it being connected.

[image]
[image]
[image]
[image]

> I just need to source a AC Fan controller first

I've been buying from sparkydirect - competitive prices but I've also found their product turnover is high enough that the mechs seem to have reasonably recent zigbee firmware.

DanielNagy, 2024-02-20:
> I hear you - I installed a fan in our bedroom and went AC over DC based on this product (plus my hate of 433 MHz remotes).

Yep - I'm with you on that also! It appears that nearly every fan now comes with a Remote, even the AC ones?

> Yes, it has different behaviour and options around the lights compared to the mechs. There are 4x seperate LEDs, one on each edge of the button. It illuminates 1x for low, 2x for medium and 3x for [...]

Did you try using the existing led indication? interesting to know if it does anything with the Led status option, even if they are wrong.
 
> Oh I've been following along and putting far too much time in finding a way to update my zigbee firmwares without the hub. I bought a generic tuya hub which does get picked up through the wiser app and allow adding of devices, but alas no firmware updates shown. I am currently in the middle of a dispute with Schneider about their prevention of expected functionality hiding being a paid product being in breach of consumer law. Will see how that pans out. I may end up with a hub myself as I'm not 50% through me entire house with wiser and feeling pretty commited.
> 
> And you know what, I think I will help as I see the work also on getting OTA updates working, which would really unlock the potential of these mechs. edit: you might want to consider that link on your profile ;)

Yeah, ota does work with z2mqtt, we just need access to the latest and greatest firmware files so we can package them and upload them to the z2m update repo for all to benefit and consume.

consider the gist link on my profile? how?

> > What does the configuration page look like for this device in the native app? Is the orientation a separate configuration option?
> 
> Yes - here's some screenshots of the bluetooth cofig options. Device is in zigbee mode but I think the screenshots still show enough without it being connected.
> 
Ok, so there's 2 settings, Indiciation which has different functions to previous (interested to understand if the current functions work if not correct)

> > I just need to source a AC Fan controller first
> 
> I've been buying from sparkydirect - competitive prices but I've also found their product turnover is high enough that the mechs seem to have reasonably recent zigbee firmware.

Yep - i usually purchase from mjs, but as i wanted a speed cap at the same time, i looked at sparkydirect, and then found it was cheaper to buy from them via ebay as ebay offered me a $15 off voucher in the mobile app. So that is purchased now.

lucasteligioridis, 2024-02-20:
Well funny enough, I have both the wise hub and the fan, as well as a spare zigbee dongle. 

My main issue was getting the stupid sniffing firmware flashed on my stick which wasn't quite the same as it's the SONOFF P stick.

From the testing I've done, none of the original code works with how the other switches modify the LED behaviour. Looks like it's on the same cluster as the fan config, there is this option on endpoint 7 on the input cluster list: 64516 

I reckon it's that cluster, but we'd need a sniffer to work out the actual data to send.

alexeiw123, 2024-02-20:
> It appears that nearly every fan now comes with a Remote, even the AC ones?

I got the ventair spyda 3 blade -  no remote although I think you can option them with. It's quiet enough, only complaint is that the lowest speed is still fairly high!

> Did you try using the existing led indication? interesting to know if it does anything with the Led status option, even if they are wrong.

I don't think it worked, the original dev work was by @lucasteligioridis before I'd got my mech but I'm sure I read somewhere that the option as with the other mechs didn't work.

DanielNagy, 2024-02-20:
> From the testing I've done, none of the original code works with how the other switches modify the LED behaviour. Looks like it's on the same cluster as the fan config, there is this option on endpoint 7 on the input cluster list: 64516

 Ok, good to know. 

I used a the older original ccXXXX zigbee stick with wireshark quite successfully. There's a custom sniff firmware and app from memory which feeds wireshark from memory. 
Capturing the secret/network key on join was quite the fun part.

DanielNagy, 2024-02-27:
Here is the sniff results of the "LED Indicator Mode" settings.. All 3 options. Looks like they changed from the 2AX/10AX/Dimmer Manufacturer Specific value to this one.

LED Indicator Mode - Always on
```

ZigBee Application Support Layer Data, Dst Endpt: 7, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 7
    Cluster: Manufacturer Specific (0xfc04)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 94
ZigBee Cluster Library Frame, Mfr: Schneider Electric (0x105e), Command: Write Attributes, Seq: 14
    Frame Control Field: Profile-wide (0x04)
    Manufacturer Code: Schneider Electric (0x105e)
    Sequence Number: 14
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 3
        Attribute: 0x0002
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 3 (0x03)
```
LED Indicator Mode - On with timeout by as locator
```
ZigBee Application Support Layer Data, Dst Endpt: 7, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 7
    Cluster: Manufacturer Specific (0xfc04)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 95
ZigBee Cluster Library Frame, Mfr: Schneider Electric (0x105e), Command: Write Attributes, Seq: 15
    Frame Control Field: Profile-wide (0x04)
    Manufacturer Code: Schneider Electric (0x105e)
    Sequence Number: 15
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 4
        Attribute: 0x0002
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 4 (0x04)
```
LED Indicator Mode -On with timeout
````
ZigBee Application Support Layer Data, Dst Endpt: 7, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 7
    Cluster: Manufacturer Specific (0xfc04)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 96
ZigBee Cluster Library Frame, Mfr: Schneider Electric (0x105e), Command: Write Attributes, Seq: 16
    Frame Control Field: Profile-wide (0x04)
    Manufacturer Code: Schneider Electric (0x105e)
    Sequence Number: 16
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 5
        Attribute: 0x0002
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 5 (0x05)
````

DanielNagy, 2024-02-27:
Here is the sniff results of the "LED Indicator Orientation" settings.

LED Indicator Orientation
Horizontal Left -- default

```
ZigBee Application Support Layer Data, Dst Endpt: 7, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 7
    Cluster: Manufacturer Specific (0xfc04)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 101
ZigBee Cluster Library Frame, Mfr: Schneider Electric (0x105e), Command: Write Attributes, Seq: 21
    Frame Control Field: Profile-wide (0x04)
    Manufacturer Code: Schneider Electric (0x105e)
    Sequence Number: 21
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 0
        Attribute: 0x0060
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 0 (0x00)
```

LED Indicator Orientation
Horizontal Right 

```
ZigBee Application Support Layer Data, Dst Endpt: 7, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 7
    Cluster: Manufacturer Specific (0xfc04)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 98
ZigBee Cluster Library Frame, Mfr: Schneider Electric (0x105e), Command: Write Attributes, Seq: 18
    Frame Control Field: Profile-wide (0x04)
    Manufacturer Code: Schneider Electric (0x105e)
    Sequence Number: 18
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 1
        Attribute: 0x0060
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 1 (0x01)
```
LED Indicator Orientation
Vertical Top
```
ZigBee Application Support Layer Data, Dst Endpt: 7, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 7
    Cluster: Manufacturer Specific (0xfc04)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 99
ZigBee Cluster Library Frame, Mfr: Schneider Electric (0x105e), Command: Write Attributes, Seq: 19
    Frame Control Field: Profile-wide (0x04)
    Manufacturer Code: Schneider Electric (0x105e)
    Sequence Number: 19
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 3
        Attribute: 0x0060
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 3 (0x03)
```
LED Indicator Orientation
Vertical Bottom	
```
ZigBee Application Support Layer Data, Dst Endpt: 7, Src Endpt: 1
    Frame Control Field: Data (0x40)
    Destination Endpoint: 7
    Cluster: Manufacturer Specific (0xfc04)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 100
ZigBee Cluster Library Frame, Mfr: Schneider Electric (0x105e), Command: Write Attributes, Seq: 20
    Frame Control Field: Profile-wide (0x04)
    Manufacturer Code: Schneider Electric (0x105e)
    Sequence Number: 20
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 2
        Attribute: 0x0060
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 2 (0x02)
```

Someone can implement these details as per the zigbee sniff data i've provided. Sorry, i'm a little time poor lately.

closed 2024-02-29 by Koenkk