Koenkk/zigbee2mqtt #18624: Eco-Dim .05 0x100W Dimmer is not responding to state setting (timeout)

kennyboy55, 2023-08-15:
### What happened?

When trying to set the state of my Eco-Dim .05 double dimmer in Z2M or HA nothing happens. After 10 seconds an error message occurs and the state of the light does switch. Then, when trying to change the state again, the light keeps turning itself back to the state it was set to first. (So if you failed to turn it on, it will keep turning itself on every x seconds even if you turn it off locally). 

I get the following error message: 
`Publish 'set' 'state' to 'Living Room Dinner Lights' failed: 'Error: Command 0x84b4dbfffe08a150/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 46286 - 1 - 27 - 6 - 11 after 10000ms)'`

### What did you expect to happen?

Dimmer to respond/switch state immediately.

### How to reproduce it (minimal and precise)

Pair EcoBright EcoDim .05 and try to switch state in Z2M or HA.

### Zigbee2MQTT version

1.32.2-dev commit: 3af130a

### Adapter firmware version

20210708

### Adapter

Sonoff Zb Dongle-P Zigbee 3.0 Usb

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12345140/log.txt)

rdeneer, 2023-08-25:
I'm experiencing the same issue here. It seems to respond but only after about 10 seconds and several time-out warnings.

rdeneer, 2023-08-26:
For me the issue is solved by explicitly setting a transition time:

[image]

Without setting this manually after pairing, I get time-outs.

kennyboy55, 2023-08-27:
> For me the issue is solved by explicitly setting a transition time:
> 
> [image]
> 
> Without setting this manually after pairing, I get time-outs.

Thank you! This seems to have solved the issue for me as well.

closed 2023-08-27 by kennyboy55

edwinh, 2024-10-13:
Yes, that works, but when turning on the lights, the brightness is always set to 100%, so not remembering the the last state. Any idea how to solve that?

Geertidow, 2025-11-23:
I fixed it by setting the transition time to 1 s in settings (speficic) (zigbee@mqtt)