Koenkk/zigbee2mqtt #18628: Vimar 14594 roller shutter updating status doesn't work

danielescali73, 2023-08-15:
### What happened?

I'm running Home Assistant OS (2023.8.2) with Zigbee2MQTT addon.
I paired my 9 devices and I can control them, but they aren't reporting the current status if shutters are moved with physical buttons.
If I go to "Exposes" page of each device and refresh, the current status is successfully updated.

@Koenkk please help us. :)

### What did you expect to happen?

The status of the roller shutter is updated if opened or closed with physical buttons on the device.is 

### How to reproduce it (minimal and precise)

If the roller shutter is closed, open it with the physical button. Check back in Home Assistant, it's still reported closed (even hours later).

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle E

danielescali73, 2023-08-15:
I wrote to Vimar technical support, the answer is below:
"Thank you for contacting us.
From some internal checks, we can confirm that our product xx594.0 provides status and percentage data but the management of this information (widget) must be sought in the supervisors.
In the case of Home Assistant, an open-source application, we have noticed that there is a specific setting for the rolling shutters defined as "go_to_lift_percentage" but we have noticed that a punctual update of the widget is not managed; we assume due to the limitations of the supervision/management software.
That said we are to confirm that our products are certified to work with Zigbee Hub 3.0 but sadly we cannot oversee the reliability of third-party manufacturers handling.
As indicated in the Bluetooth mode with our gateway, the status information and shutter opening percentage are promptly managed."

lanny318, 2023-08-16:
Hi you can retrieve the status  in the dev console ->  closures-  'closuresWindowCovering': {"currentPositionLiftPercentage":49}.
It is not update in realtime, only when the motor stops.

danielescali73, 2023-08-17:
> Hi you can retrieve the status in the dev console -> closures- 'closuresWindowCovering': {"currentPositionLiftPercentage":49}. It is not update in realtime, only when the motor stops.

Sorry I'm not an expert. How can I get the information to home assistant?

hexstudy, 2023-08-21:
Hi,

I will add that when the percentuage is 0 the state remaining to stato OPEN

lanny318, 2023-08-21:
Hi, in the past I put this code in an external converter  and it works fine.

Original code in vimar.ts

`    {
        zigbeeModel: ['Window_Cov_v1.0'],
        model: '14594',
        vendor: 'Vimar',
        description: 'Roller shutter with slat orientation and change-over relay',
        fromZigbee: [fz.cover_position_tilt],
        toZigbee: [tz.cover_state, tz.cover_position_tilt],
        exposes: [e.cover_position()],
    },

New code in external converter:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    zigbeeModel: ['Window_Covering_v1.0'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '14594', // Vendor model number, look on the device for a model number
    vendor: 'vimar', // Vendor of the device (only used for documentation and startup logging)
    description: 'Roller shutter with slat orientation and change-over relay', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'open, close, stop, position, tilt',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    exposes: [e.cover_position()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(10);
        const binds = ['closuresWindowCovering'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
};

module.exports = definition;`
`

hexstudy, 2023-08-21:
@lanny318 thank you
I have a container installation, can you tell me where can i found this file?
Or I can put inside some folder?

Thanks

closed 2023-09-29 by github-actions[bot]

danielescali73, 2023-09-29:
@lanny318, @Koenkk Will the fix of this issue be released in the next version?

lanny318, 2023-09-29:
Perhaps someone should confirm that it still works well; I no longer have this device.

danielescali73, 2023-09-29:
Confirmed! Works!

hexstudy, 2023-09-29:
Confirmed!

LeoeLeoeL, 2023-11-09:
Any news about this fix?
I just migrated this device from Deconz to Z2M via Sonoff "E" and I have the same behaviour.
It's very strange in Deconz it worked. :)

Koenkk, 2023-11-09:
Integrated the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

LeoeLeoeL, 2023-11-15:
It works.
Tks

LeoeLeoeL, 2025-09-08:
Yes it works as it should.

tonytony00, 2025-09-08:
Thank you,

How do I check that similar code (arké series)  19594 also works correctly?