Koenkk/zigbee2mqtt #18632: [New device support]:    TS0601  _TZE204_kapvnnlk - Human Presence Sensor

Diddlik, 2023-08-16:
### Link

https://de.aliexpress.com/item/1005005834366702.html?spm=a2g0o.order_list.order_list_main.11.112d5c5fiP3E61&gatewayAdapt=glo2deu

### Database entry

{"id":83,"type":"Router","ieeeAddr":"0xa4c13871b329107b","nwkAddr":33819,"manufId":4417,"manufName":"_TZE204_kapvnnlk","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_kapvnnlk","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1692174674158,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to replace one of the existing motion detector, but without luck ;(

geertvercamer, 2023-08-31:
same here, can't link it to HA

kkossev, 2023-09-06:
Presence State				         1
Human Motion State			11
Presence Time				        12 
Motion Far Detection			13
Motion Detection Sensitivity		15
Motionless Detection Sensitivity	16
Current Distance			        19
Battery level				        101

===============

Code	Type	Values
presence_state	Enum	
{
  "range": [
    "none",
    "presence"
  ]
}
human_motion_state	Enum	
{
  "range": [
    "none",
    "small_move",
    "large_move"
  ]
}
presence_time	Integer	
{
  "unit": "s",
  "min": 3,
  "max": 600,
  "scale": 0,
  "step": 1
}
motion_far_detection	Integer	
{
  "unit": "cm",
  "min": 150,
  "max": 600,
  "scale": 0,
  "step": 75
}
motion_sensitivity_value	Integer	
{
  "unit": "",
  "min": 0,
  "max": 7,
  "scale": 0,
  "step": 1
}
motionless_sensitivity	Integer	
{
  "unit": "",
  "min": 0,
  "max": 7,
  "scale": 0,
  "step": 1
}
dis_current	Integer	
{
  "unit": "cm",
  "min": 0,
  "max": 1000,
  "scale": 0,
  "step": 1
}

===============

Code	Type	Values
presence_time	Integer	
{
  "unit": "s",
  "min": 3,
  "max": 600,
  "scale": 0,
  "step": 1
}
motion_far_detection	Integer	
{
  "unit": "cm",
  "min": 150,
  "max": 600,
  "scale": 0,
  "step": 75
}
motion_sensitivity_value	Integer	
{
  "unit": "",
  "min": 0,
  "max": 7,
  "scale": 0,
  "step": 1
}
motionless_sensitivity	Integer	
{
  "unit": "",
  "min": 0,
  "max": 7,
  "scale": 0,
  "step": 1
}

=================

closed 2023-10-14 by github-actions[bot]

Diddlik, 2024-01-17:
I tried to create a converter, sofar ok.

one point to fix is human_motion, How can I define enum?
and the other problem is to set value. If I try to change some values I get this error:

No converter available for 'motionless_detection_sensitivity' (1)

human_motion_state Enum
{
"range": [
"none",
"small_move",
"large_move"
]
}

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_kapvnnlk',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Fill in a description of the device here',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [e.presence(),
            e.numeric('current_distance', ea.STATE).withDescription('Current Distance').withUnit('cm'),
			
            e.numeric('motionless_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
                .withDescription('Motionless Detection Densitivity'),
			e.numeric('motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
                .withDescription('Motion Detection Densitivity'),
            e.numeric('motion_far_detection', ea.STATE_SET).withValueMin(150).withValueMax(600).withValueStep(75)
                .withDescription('Motion Far Detection').withUnit('cm'),
            e.numeric('human_motion_state', ea.STATE).withDescription('Human Motion State'),
			e.numeric('presence_time', ea.STATE_SET).withValueMin(3).withValueMax(600).withValueStep(1)
                .withDescription('Presence Time').withUnit('s'),
			e.battery(),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
		 [1, 'presence', tuya.valueConverter.trueFalse1],
         [19, 'current_distance', tuya.valueConverter.raw],
         [11, 'human_motion_state', tuya.valueConverter.trueFalse1],
		 [13, 'motion_far_detection', tuya.valueConverter.raw],
         [15, 'motion_detection_sensitivity', tuya.valueConverter.raw],
         [16, 'motionless_detection_sensitivity', tuya.valueConverter.raw],
         [12, 'presence_time', tuya.valueConverter.raw],
         [101, 'battery', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```