Koenkk/zigbee2mqtt #18636: Philips Hue white and color ambiance 9290022166 not responding due to cluster 2

accelle17, 2023-08-17:
### What happened?

Z2m version: 1.32.2
Adapter: conbee2 firmware 0x26780700
Devices: 150
End devices: 106
Router: 44
Started: when I upgraded the zigbee2mqtt add-on from 1.32.0-2 to : 1.32.2-1.
Hue firmware: 1.53.3_r27175, also upgraded to 1.104.2

### What did you expect to happen?

On/off to happen and there should only be two endpoint (11 and 242) but there is now 3 which includes 2.
Tried removing the device normally and by force but endpoint 2 is still there.
I have two more hues of the same kind but I don't want to rejoin it to see if it acts the same.

[image]

### How to reproduce it (minimal and precise)

1. remove device
2. allow permit join to all or specific device
3. device should rejoin (also tried using hue switch to reset it)
4. device rejoins and with endpoint 2,11, 242.
5. It even has reporting configured for endpoints 11 which doesn't exist on the other 2 devices with the same version/device model.
6. try to power on and it tries to publish it on endpoint 2 instead of 11

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

0x26780700

### Adapter

conbee2

### Debug log

JOINING:

Z2M@hass.dayritcaldozo.family
Devices
Dashboard
Map
Settings
Groups
OTA
Touchlink
Logs
Extensions
English
Show only
all
Filter by text
0x0017880108aceff5
Log level
debug
Logging level
 
Info 2023-08-17 08:19:18Accepting joining not in blocklist device '0x0017880108aceff5'
Info 2023-08-17 08:19:18Device '0x0017880108aceff5' joined
Info 2023-08-17 08:19:18MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880108aceff5","ieee_address":"0x0017880108aceff5"},"type":"device_joined"}'
Info 2023-08-17 08:19:18MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0017880108aceff5"},"type":"device_connected"}'
Info 2023-08-17 08:19:18MQTT publish: topic 'homeassistant/light/0x0017880108aceff5/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/0x0017880108aceff5/set","device":{"identifiers":["zigbee2mqtt_0x0017880108aceff5"],"manufacturer":"Philips","model":"Hue white and color ambiance E26/E27 (9290022166)","name":"0x0017880108aceff5","sw_version":"1.104.2"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","candle","fireplace","colorloop","finish_effect","stop_effect","stop_hue_effect"],"json_attributes_topic":"zigbee2mqtt/0x0017880108aceff5","max_mireds":500,"min_mireds":153,"name":"0x0017880108aceff5","schema":"json","state_topic":"zigbee2mqtt/0x0017880108aceff5","supported_color_modes":["xy","color_temp"],"unique_id":"0x0017880108aceff5_light_zigbee2mqtt"}'
Info 2023-08-17 08:19:18MQTT publish: topic 'homeassistant/sensor/0x0017880108aceff5/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880108aceff5"],"manufacturer":"Philips","model":"Hue white and color ambiance E26/E27 (9290022166)","name":"0x0017880108aceff5","sw_version":"1.104.2"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/0x0017880108aceff5","name":"0x0017880108aceff5 power on behavior","state_topic":"zigbee2mqtt/0x0017880108aceff5","unique_id":"0x0017880108aceff5_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
Info 2023-08-17 08:19:18MQTT publish: topic 'homeassistant/select/0x0017880108aceff5/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x0017880108aceff5/set/power_on_behavior","device":{"identifiers":["zigbee2mqtt_0x0017880108aceff5"],"manufacturer":"Philips","model":"Hue white and color ambiance E26/E27 (9290022166)","name":"0x0017880108aceff5","sw_version":"1.104.2"},"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/0x0017880108aceff5","name":"0x0017880108aceff5 power on behavior","options":["off","on","toggle","previous"],"state_topic":"zigbee2mqtt/0x0017880108aceff5","unique_id":"0x0017880108aceff5_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
Info 2023-08-17 08:19:18MQTT publish: topic 'homeassistant/select/0x0017880108aceff5/effect/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x0017880108aceff5/set/effect","device":{"identifiers":["zigbee2mqtt_0x0017880108aceff5"],"manufacturer":"Philips","model":"Hue white and color ambiance E26/E27 (9290022166)","name":"0x0017880108aceff5","sw_version":"1.104.2"},"enabled_by_default":false,"icon":"mdi:palette","json_attributes_topic":"zigbee2mqtt/0x0017880108aceff5","name":"0x0017880108aceff5 effect","options":["blink","breathe","okay","channel_change","candle","fireplace","colorloop","finish_effect","stop_effect","stop_hue_effect"],"unique_id":"0x0017880108aceff5_effect_zigbee2mqtt"}'
[... 12 lines cut ...]
Debug 2023-08-17 08:19:19Received MQTT message on 'homeassistant/select/0x0017880108aceff5/effect/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x0017880108aceff5/set/effect","device":{"identifiers":["zigbee2mqtt_0x0017880108aceff5"],"manufacturer":"Philips","model":"Hue white and color ambiance E26/E27 (9290022166)","name":"0x0017880108aceff5","sw_version":"1.104.2"},"enabled_by_default":false,"icon":"mdi:palette","json_attributes_topic":"zigbee2mqtt/0x0017880108aceff5","name":"0x0017880108aceff5 effect","options":["blink","breathe","okay","channel_change","candle","fireplace","colorloop","finish_effect","stop_effect","stop_hue_effect"],"unique_id":"0x0017880108aceff5_effect_zigbee2mqtt"}'
Debug 2023-08-17 08:19:19Received MQTT message on 'homeassistant/sensor/0x0017880108aceff5/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880108aceff5"],"manufacturer":"Philips","model":"Hue white and color ambiance E26/E27 (9290022166)","name":"0x0017880108aceff5","sw_version":"1.104.2"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x0017880108aceff5","name":"0x0017880108aceff5 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x0017880108aceff5","unique_id":"0x0017880108aceff5_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Debug 2023-08-17 08:19:19Received MQTT message on 'homeassistant/sensor/0x0017880108aceff5/last_seen/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880108aceff5"],"manufacturer":"Philips","model":"Hue white and color ambiance E26/E27 (9290022166)","name":"0x0017880108aceff5","sw_version":"1.104.2"},"device_class":"timestamp","enabled_by_default":true,"entity_category":"diagnostic","icon":"mdi:clock","json_attributes_topic":"zigbee2mqtt/0x0017880108aceff5","name":"0x0017880108aceff5 last seen","state_topic":"zigbee2mqtt/0x0017880108aceff5","unique_id":"0x0017880108aceff5_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Debug 2023-08-17 08:19:19Received MQTT message on 'homeassistant/sensor/0x0017880108aceff5/update_state/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880108aceff5"],"manufacturer":"Philips","model":"Hue white and color ambiance E26/E27 (9290022166)","name":"0x0017880108aceff5","sw_version":"1.104.2"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x0017880108aceff5","name":"0x0017880108aceff5 update state","state_topic":"zigbee2mqtt/0x0017880108aceff5","unique_id":"0x0017880108aceff5_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Debug 2023-08-17 08:19:19Received MQTT message on 'homeassistant/binary_sensor/0x0017880108aceff5/update_available/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880108aceff5"],"manufacturer":"Philips","model":"Hue white and color ambiance E26/E27 (9290022166)","name":"0x0017880108aceff5","sw_version":"1.104.2"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x0017880108aceff5","name":"0x0017 [... cut ...]
Info 2023-08-17 08:19:44MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x0017880108aceff5' failed: 'Error: Command 0x0017880108aceff5/2 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (210))'","meta":{"friendly_name":"0x0017880108aceff5"},"type":"zigbee_publish_error"}'

Koenkk, 2023-08-17:
Pushed a fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-17 by Koenkk

accelle17, 2023-08-18:
Great. It is working now via HA and z2m ui. There's also that 64516 on the input cluster of endpoint 11. I am not able to bind it now too to tradfri switch. Let me test further on the binding.

accelle17, 2023-08-18:
Binding now works to tradfri switch after I removed the entries on reporting and the coordinator bind. This was not present on my two similar hue bulbs. It was added during the rejoin of the problematic bulb. Again, thank you for the quick work.

Note: I am using z2m addon on HA supervised. I just copied the philips.js on the fix and add it as external converter (need to replace the .. with the static path (/app/node_modules/zigbee-herdsman-converters) to work.