Koenkk/zigbee2mqtt #18642: [Wrong device]: GIRIER Tuya ZigBee Plug 20A Smart Outlet Socket EU with Power Monitoring

sereda-oleg, 2023-08-17:
### Link

https://aliexpress.ru/item/1005005105319027.html

### Model

JR-ZPM03

### Description

Smart Plug

### Vendor

Girier

### Picture (link)

https://ae04.alicdn.com/kf/Sb635c77754984e97aacca60effdd0340I.jpg_640x640.jpg

### Database entry

{"id":38,"type":"Router","ieeeAddr":"0x70b3d52b60027e03","nwkAddr":54921,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"manufacturerName":"_TZ3000_okaz9tjs","zclVersion":2,"appVersion":100,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,251]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"moesStartUpOnOff":2,"tuyaBacklightMode":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31BS102","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2123427102},"lastSeen":1692270089524,"defaultSendRequestWhen":"immediate"}

sereda-oleg, 2023-08-17:
[image]

closed 2023-09-25 by github-actions[bot]

S474N, 2024-05-12:
Same "problem" here, only it identifies as `BSD29_1`. Is it possible to fix it? Also with correct photo of device?
[image]

tonbor, 2024-08-10:
nice if there was some action on this after one year

b2un0, 2024-09-16:
I have the original device from [Elivico](https://www.zigbee2mqtt.io/devices/LSPA9.html#elivco-lspa9) which has exactly this Tuya identifier (`_TZ3000_okaz9tjs`).

What do you think should happen now?

If we now change the name and the picture to your device, it will be wrong for all others who have the Elivico device (like me).

As our devices were on the market earlier, it can be assumed that your manufacturer ‘`GIRIER`’ used the same firmware