Koenkk/zigbee2mqtt #18643: Zigbee2mqtt crashes regularly

bikeymouse, 2023-08-17:
### What happened?

I'm using Zigbee2mqtt with the standard Docker images. 
Normally I'm using the koenkk/zigbee2mqtt:latest Docker-image, but I experimented with the "latest-dev" version to get rid of the warnings in Home Assistant about the entity/device naming.

This worked Ok, but after a day or 2 Zigbee2Mqtt went crashing, so I reverted back to "latest" which is now 6.10.3.0 build 297

Unfortunately this didn't help, now Zigbee2Mqtt keeps on crashing after every couple of days. Restarting the container, or the whole machine does not help, I have to physically remove the USB adapter from the machine.

It looks like the Sonoff Zigbee Dongle Plus "P" is crashing.

If I look at the logs it starts with all my devices not responding to a ping, then the MQTT-broker being not available (not true) and then a "FATAL ERROR: Reached heap limit Allocation failed - JavaScript heap out of memory" 

### What did you expect to happen?

To keep working.

### How to reproduce it (minimal and precise)

Rebuild the container, pulled the image again.
Restarted Zigbee2Mqtt. Restarted machine.

### Zigbee2MQTT version

1.3.2.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonof Zigbee Dongle-P

### Debug log

Zigbee2MQTT:error 2023-08-17 11:57:44: Not connected to MQTT server!
Zigbee2MQTT:warn  2023-08-17 11:58:02: Failed to ping 'Piano Spot1' (attempt 1/1, Read 0x001788010cf38aa0/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Zigbee2MQTT:error 2023-08-17 11:59:27: Not connected to MQTT server!
Zigbee2MQTT:error 2023-08-17 11:59:57: Not connected to MQTT server!
Zigbee2MQTT:error 2023-08-17 12:00:22: Not connected to MQTT server!
Zigbee2MQTT:warn  2023-08-17 12:01:56: Failed to ping 'TV Lightstrip' (attempt 1/1, Read 0x0017880103f545cd/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Zigbee2MQTT:error 2023-08-17 12:03:55: Not connected to MQTT server!
<--- Last few GCs --->
[7:0x7f4c4d57f330] 317717108 ms: Scavenge 1907.1 (1993.7) -> 1899.8 (1993.7) MB, 3.8 / 0.0 ms  (average mu = 0.451, current mu = 0.300) allocation failure; 
[7:0x7f4c4d57f330] 317717132 ms: Scavenge 1907.4 (1993.7) -> 1900.0 (1994.2) MB, 3.8 / 0.0 ms  (average mu = 0.451, current mu = 0.300) allocation failure; 
[7:0x7f4c4d57f330] 317717157 ms: Scavenge 1907.6 (1994.2) -> 1900.2 (2010.2) MB, 5.0 / 0.0 ms  (average mu = 0.451, current mu = 0.300) allocation failure; 
<--- JS stacktrace --->
FATAL ERROR: Reached heap limit Allocation failed - JavaScript heap out of memory

william-sy, 2023-08-18:
Hi, 

The first error is: 
`Zigbee2MQTT:error 2023-08-17 11:57:44: Not connected to MQTT server!`
Is your docker mqtt server up and running and still configured correctly ?

Seccond thing is that yes, The update to the dev branch might be not backwards compatible, but do not quote me on this.

bikeymouse, 2023-08-18:
Hi,
Yes it is connected all the time to MQTT-broker that is on-line 24/7 as I'm using the for all other stuff as well. 
Disconnecting to the MQTT-server sees to me a result not a cause of the issue. If it would, just restarting Zigbee2Mqtt would solve the issue.

The fact that I really physically have to remove the USB-stick suggest to me that the stick itself gets into a state from which it can only recover after a stick-reboot. Really unfortunately that I cannot reset is remotely as I'm about to go on vacation.

What could do to remove data that is not backwards compatible without having to re-pair all my devices?

The firmware on my Sonoff ZBdongle-P is 6.10.3.0 build 297 as I indicated in the initial post.

william-sy, 2023-08-18:
`which it can only recover after a stick-reboot. Really unfortunately that I cannot reset is remotely as I'm about to go on vacation.` 

Technically you can, the firmware updater does this for example for you.  It s a bit complex but you can find it here: 
`https://github.com/william-sy/sonoff-zbdongle-p-fw-mac-Lin/blob/main/script/uartLog_en.py`
You would have to adjust the script to just reset the device

Please make sure z2m is off before attempting to update firmware (Note: This will likely not solve your problem tough)

The best thing is to restore a backup of the config folder if you have one and see if that fixes your problem as I am not sure what to delete exactly.. Perhaps someone else might know that

bikeymouse, 2023-08-27:
I'm afraid that this script also doesn't work as the complete USB device is no longer available. 
E.g. lsusb no longer shows the Snonoff USB dongle. It's just like the stick completely locks up.
Looking at dmesg: I see that it is no longer available (from inserting the stick until it suddenly is no longer working):
`
[91510.152255] usb 1-4: new full-speed USB device number 7 using xhci_hcd
[91510.316083] usb 1-4: New USB device found, idVendor=1a86, idProduct=55d4, bcdDevice= 4.42
[91510.316087] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[91510.316089] usb 1-4: Product: SONOFF Zigbee 3.0 USB Dongle Plus V2
[91510.316091] usb 1-4: Manufacturer: ITEAD
[91510.316092] usb 1-4: SerialNumber: 20220817082147
[91510.317869] cdc_acm 1-4:1.0: ttyACM0: USB ACM device
....
[107682.260705] usb 1-4: USB disconnect, device number 7
[107682.566028] usb 1-4: new full-speed USB device number 8 using xhci_hcd
[107682.698050] usb 1-4: device descriptor read/64, error -71
[107682.934061] usb 1-4: device descriptor read/64, error -71
[107683.178117] usb 1-4: new full-speed USB device number 9 using xhci_hcd
[107683.306045] usb 1-4: device descriptor read/64, error -71
[107683.542198] usb 1-4: device descriptor read/64, error -71
[107683.650074] usb usb1-port4: attempt power cycle
[107684.062055] usb 1-4: new full-speed USB device number 10 using xhci_hcd
[107684.062188] usb 1-4: Device not responding to setup address.
[107684.291254] usb 1-4: device descriptor read/8, error -71
[107684.419544] usb 1-4: device descriptor read/8, error -71
[107684.654041] usb 1-4: new full-speed USB device number 11 using xhci_hcd
[107684.674181] usb 1-4: Invalid ep0 maxpacket: 0
[107684.674265] usb usb1-port4: unable to enumerate USB device
[139017.555024] fwbr120i0: port 2(tap120i0) entered disabled state
[139017.582715] fwbr120i0: port 1(fwln120i0) entered disabled state
[139017.582788] vmbr0: port 2(fwpr120p0) entered disabled state
[139017.582920] device fwln120i0 left promiscuous mode
[139017.582923] fwbr120i0: port 1(fwln120i0) entered disabled state
[139017.611044] device fwpr120p0 left promiscuous mode
[139017.611052] vmbr0: port 2(fwpr120p0) entered disabled state
`

william-sy, 2023-08-27:
Sorry to see that, Might be time for a reinstall of the OS or a new zigbee stick..

bikeymouse, 2023-08-27:
Yes, l already bought a new Sonoff Type P stick and flashed it with the same IEEE address but then it also doesn’t start saying the that the PanId is probably already in use. Flashed it 3 times now, also with different firmware but keep having the same error.
I also have a Sonoff Type-E but that is not  a 1-1 replacement. 
One way or another it I looks like I have to re-pair al my 35 devices 😬

william-sy, 2023-08-27:
For that problem see: https://github.com/Koenkk/zigbee2mqtt/issues/2893

closed 2023-10-05 by github-actions[bot]

tobias-digital, 2024-07-24:
I have the same issue. zigbee2mqtt cashes after several days. Only unplugging the usb-stick works. Reboot of the raspberry etc. doesnt show any effect. It seems the stick has crashed ...