Koenkk/zigbee2mqtt #18644: [New device support]: NIKO 552-720x1 Wireless battery-powered switch, 1 button

netweaver1970, 2023-08-17:
### Link

https://www.niko.eu/en/article/552-72001

### Database entry

{"id":78,"type":"EndDevice","ieeeAddr":"0x003c84fffe8bf3a0","nwkAddr":50561,"manufId":4703,"manufName":"NIKO NV","powerSource":"Battery","modelId":"Battery switch, 1 button","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":6,"inClusterList":[0,1,3,2821,4096,64512,64513],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"Battery switch, 1 button","manufacturerName":"NIKO NV","powerSource":3,"zclVersion":3,"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20220223","swBuildId":"314-446-00"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20220223","swBuildId":"314-446-00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1692273101793,"defaultSendRequestWhen":"immediate"}

### Comments

thanks for having a look.

netweaver1970, 2023-08-17:
I removed the 2 last lines from the configuration.yaml in the zigbee2mqtt folder file and now z2m_edge starts again.

But I'm still puzzled why adding the proper  convertor would cause this bad start-up behaviour.
Not touching it anymore till someone has a clever idea :)

netweaver1970, 2023-08-17:
I got a step further, as I couldn't let go. Now I don't have an error anymore, I think I got a possible config (compared with other devices) but no luck.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Battery switch, 1 button'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '552-720x1' , // Vendor model number, look on the device for a model number
    vendor: 'NIKO NV', // Vendor of the device (only used for documentation and startup logging)
    description: 'NIKO Wireless battery-powered switch, 1 button', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.battery, fz.brightness],
    toZigbee: [],
    exposes: [e.battery(), e.action(['single', 'double', 'hold'])],
};

module.exports = definition;
```

Battery percentage is flashing red in the dashboard (possible) but the action field is not filled when pressing a button.
Also the bind page is empty. when manually making a bind between endpoint 1 and the coordinator, for clusters on/off and levelcontrol, then it's properly accepted by the switch.
When pressing the button long and short I'm getting corresponding actions in the log but no change in the exposes tab.
It seems I might be going in the right direction but 'with a blindfold on'. Iets met een klok en een klepel... (Dutch expression)

Can you help me further?

Thanks.

**extract from the log:**
 ```
Debug 2023-08-17 22:26:31Received Zigbee message from '0x003c84fffe8bf6de', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0

Info 2023-08-17 22:26:31MQTT publish: topic 'zigbee2mqtt/0x003c84fffe8bf6de', payload '{"battery":null,"device":{"applicationVersion":1,"dateCode":"20220223","friendlyName":"0x003c84fffe8bf6de","hardwareVersion":0,"ieeeAddr":"0x003c84fffe8bf6de","manufacturerID":4703,"manufacturerName":"NIKO NV","model":"552-720x1","networkAddress":25773,"powerSource":"Battery","softwareBuildID":"314-446-00","stackVersion":6,"type":"EndDevice","zclVersion":3},"last_seen":"2023-08-17T22:26:31+02:00","linkquality":32}'

Debug 2023-08-17 22:26:58Received Zigbee message from '0x003c84fffe8bf6de', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0

Info 2023-08-17 22:26:58MQTT publish: topic 'zigbee2mqtt/0x003c84fffe8bf6de', payload '{"battery":null,"device":{"applicationVersion":1,"dateCode":"20220223","friendlyName":"0x003c84fffe8bf6de","hardwareVersion":0,"ieeeAddr":"0x003c84fffe8bf6de","manufacturerID":4703,"manufacturerName":"NIKO NV","model":"552-720x1","networkAddress":25773,"powerSource":"Battery","softwareBuildID":"314-446-00","stackVersion":6,"type":"EndDevice","zclVersion":3},"last_seen":"2023-08-17T22:26:58+02:00","linkquality":36}'

Debug 2023-08-17 22:27:14Received Zigbee message from '0x003c84fffe8bf6de', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0

Info 2023-08-17 22:27:14MQTT publish: topic 'zigbee2mqtt/0x003c84fffe8bf6de', payload '{"battery":null,"device":{"applicationVersion":1,"dateCode":"20220223","friendlyName":"0x003c84fffe8bf6de","hardwareVersion":0,"ieeeAddr":"0x003c84fffe8bf6de","manufacturerID":4703,"manufacturerName":"NIKO NV","model":"552-720x1","networkAddress":25773,"powerSource":"Battery","softwareBuildID":"314-446-00","stackVersion":6,"type":"EndDevice","zclVersion":3},"last_seen":"2023-08-17T22:27:14+02:00","linkquality":36}'

Debug 2023-08-17 22:27:20Received Zigbee message from '0x003c84fffe8bf6de', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":0,"rate":50}' from endpoint 1 with groupID 0

Info 2023-08-17 22:27:20MQTT publish: topic 'zigbee2mqtt/0x003c84fffe8bf6de', payload '{"battery":null,"device":{"applicationVersion":1,"dateCode":"20220223","friendlyName":"0x003c84fffe8bf6de","hardwareVersion":0,"ieeeAddr":"0x003c84fffe8bf6de","manufacturerID":4703,"manufacturerName":"NIKO NV","model":"552-720x1","networkAddress":25773,"powerSource":"Battery","softwareBuildID":"314-446-00","stackVersion":6,"type":"EndDevice","zclVersion":3},"last_seen":"2023-08-17T22:27:20+02:00","linkquality":36}'

Debug 2023-08-17 22:27:20Received Zigbee message from '0x003c84fffe8bf6de', type 'commandStop', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0

Info 2023-08-17 22:27:20MQTT publish: topic 'zigbee2mqtt/0x003c84fffe8bf6de', payload '{"battery":null,"device":{"applicationVersion":1,"dateCode":"20220223","friendlyName":"0x003c84fffe8bf6de","hardwareVersion":0,"ieeeAddr":"0x003c84fffe8bf6de","manufacturerID":4703,"manufacturerName":"NIKO NV","model":"552-720x1","networkAddress":25773,"powerSource":"Battery","softwareBuildID":"314-446-00","stackVersion":6,"type":"EndDevice","zclVersion":3},"last_seen":"2023-08-17T22:27:20+02:00","linkquality":36}'

```

netweaver1970, 2023-08-17:
**Some success!** 
with the following converter it works. But the rate of up/down dimming is too fast: now in 1 (max 2, hard to measure) second I'm from 0 to 255 (0% to 100% brightness).
Probably this line in the log is related: But how to interfere?
`'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":0,"rate":50}'`

the -kind of- working converter is this one:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Battery switch, 1 button'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '552-720x1' , // Vendor model number, look on the device for a model number
    vendor: 'NIKO NV', // Vendor of the device (only used for documentation and startup logging)
    description: 'NIKO Wireless battery-powered switch, 1 button', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_move, fz.command_on, fz.command_off, fz.identify, fz.battery],
    toZigbee: [],
    exposes: [e.action(['on', 'off', 'brightness_move_up', 'brightness_move_down', 'brightness_stop']), e.battery()],
};

module.exports = definition;
```

Koenkk, 2023-08-18:
AFAIK you cannot change the rate of a dimmer (at least not from the coordinator perspective). This comes from the firmware of the dimmer.

netweaver1970, 2023-08-18:
> AFAIK you cannot change the rate of a dimmer (at least not from the coordinator perspective). This comes from the firmware of the dimmer.

the NIKO dimmer switch has some vendor specific endpoints so maybe there's a possibility that using the Niko gateway + app, these things can be changed. I looked at the existing convertors for the niko but no luck, nothing related to dimming rate in z2m so far.
[image]
Niko is saying that their button/switch/dimmer can be used with other devices and gateways, so the behaviour must be somehow 'nice'. And a 1-2 s dimming time is not really to be called nice.

I don't know how this 'rate' works, is this the steps/s between the/any move_start type message and the move_stop type? 
In that case the dimming interval should be about 5s, (255/50). That would be a lot more like it.

Another potential thing involved is the fact that my physical Zigbee dimmer is not recognized as the correct device at all. Mine is this one (a zigbee 2 0-10V output, used to dim LED lights): https://www.aliexpress.com/item/1005005523348129.html?spm=a2g0o.order_list.order_list_main.5.291ed2d6HjxUHC 

While the one z2m wrongly recognises is: https://www.zigbee2mqtt.io/devices/YSR-MINI-01_dimmer.html#ysrsai-ysr-mini-01_dimmer

my (incorrectly recognized) database.db entry is: 
`{"id":80,"type":"Router","ieeeAddr":"0xa4c1382005268d13","nwkAddr":20866,"manufId":4107,"manufName":"eWeLight","powerSource":"Mains (single phase)","modelId":"ZB-DL01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZB-DL01","manufacturerName":"eWeLight","powerSource":1,"zclVersion":3,"appVersion":6,"stackVersion":2,"hwVersion":0,"dateCode":"20221027","swBuildId":"2.0.1"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":133}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca15655","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca15655","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}}},"appVersion":6,"stackVersion":2,"hwVersion":0,"dateCode":"20221027","swBuildId":"2.0.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1692370616095,"defaultSendRequestWhen":"immediate"}`

The main function kind of works but maybe not 100% correct, specifically the too fast dimming. Could that be related?

Also, what do I need to do/change to get the NIKO dimmer switch 'action' also relayed to HA? Right now I have bound the NIKO switch to the LED dimmer directly in z2m, as this is for me enough but for proper inclusion I'm sure the Action would have to be visible in HA.

Thanks.

Koenkk, 2023-08-19:
From the Zigbee spec:

[image]

So rate is amount of units to move per seconds (where the max for the bulb is 254, min = 0)

> Also, what do I need to do/change to get the NIKO dimmer switch 'action' also relayed to HA? Right now I have bound the NIKO switch to the LED dimmer directly in z2m

If you bind a dimmer, directly to the device (not to a group), the dimmer will not send any actions to the coordinator anymore. Therefore you will not see actions in HA.

netweaver1970, 2023-08-21:
giving up on this solution (for my specific usecase), as the rate on dimming is too fast and the 2 NIKO switches are actually (soft)fighting with each other and with the central zigbee light dimmer), as each wants to have control over the brightness level & on/off state. 

The 'smartness' needs to be moved to a central point, not to 2 (even 3) individual points. 

This switch is totally fine in a single control point to simple zigbee light(s) scenario, but not in a multiple control point to simple zigbee light(s) or multiple control point to zigbee dimmer scenario, as not configurable enough and too many logic control points (#switches +1)

Instead I'll try going for this centralized zigbee dimmer [Ali Link](https://www.aliexpress.com/item/1005005925119433.html?spm=a2g0o.productlist.main.13.534e54393Ck4zW&algo_pvid=ccc01d4b-df1f-4d79-b16c-38082ed89d3c&aem_p4p_detail=202308210452563147895711089320010756725&algo_exp_id=ccc01d4b-df1f-4d79-b16c-38082ed89d3c-6&pdp_npi=4%40dis%21EUR%2134.26%2123.98%21%21%21266.43%21%21%40211b88f116926187761178745e0207%2112000034875300304%21sea%21BE%21119382455%21&curPageLogUid=we0pSCU8uGPe&search_p4p_id=202308210452563147895711089320010756725_7) and use my old 2 pre-existing hardwired 220V pushbutton switches, so only leaving me with 1 logic control point (the new zigbee control module) and there won't be state sync issues as I'm seeing now during testing the double (actually triple) zigbee dimmer situation: using 2x the niko 552-720X1 + Zigbee dimmer (0-10V). 

Maybe a different setup (z2m converter) can change the behaviour but as long as the NIKO switch persistently holds on to it's state (on/off+brightness) there'll be sync mismatches the moment one increases the number of switches. And that's a no-no for me. Maybe setup zigbee reporting between all 3 devices involved could als solve the sync issues but I would still end up with too fast dimming rate.

Closing this request, maybe the external converter is useful to someone, as it works ok-ish. Or to be  used as a basis for further work.

closed 2023-08-21 by netweaver1970