Koenkk/zigbee2mqtt #18646: Loginovo Zigbee Mmwave Human Presence Sensor Tuya Smart Presence Detector 5V Motion Sensor With Luminance/Distance Detection[New device support]:

genik70, 2023-08-17:
### Link

https://www.aliexpress.com/item/1005005677110270.html?spm=a2g0o.order_list.order_list_main.23.21ef5c5fvAgRrN&gatewayAdapt=glo2deu

### Database entry

{"id":27,"type":"Router","ieeeAddr":"0x385cfbfffeb432c3","nwkAddr":29397,"manufId":4098,"manufName":"_TZE204_yensya2c","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_yensya2c","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1692287198022,"defaultSendRequestWhen":"immediate"}

### Comments

Datapoint 115 not defined for '_TZE204_yensya2c' with data {"dp":115,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Datapoint 116 not defined for '_TZE204_yensya2c' with data {"dp":116,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
Datapoint 1 not defined for '_TZE204_yensya2c' with data {"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Datapoint 19 not defined for '_TZE204_yensya2c' with data {"dp":19,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
Datapoint 20 not defined for '_TZE204_yensya2c' with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,2]}}
Datapoint 113 not defined for '_TZE204_yensya2c' with data {"dp":113,"datatype":1,"data":{"type":"Buffer","data":[0]}}
Datapoint 114 not defined for '_TZE204_yensya2c' with data {"dp":114,"datatype":1,"data":{"type":"Buffer","data":[0]}}
I tried to make an external converter similar to ZY-M100-S_1 but no success

genik70, 2023-08-18:
Good afternoon. The manufacturer has provided me with the points
[image]

PunkRa1n, 2023-08-27:
Hi @genik70 thanks for the datapoints,
I've written a WIP code. That will work. But I haven't figured out how to send the commands back correctly. I don't even know if "radar_sensitivy" will arrive or accept the values.
Many strings will not work. I took many references from [here] (https://github.com/Koenkk/zigbee-herdsman-converters/blob/23eba9da00fdb0b6743bec86f4bb534b28e05c4b/src/devices/tuya.ts#L4009C24-L4009C41)  and many other lines and did the best I could.
Many thinks need to be adjusted. 

The only problem is that the device is very chatty and I don't know how to minimize that.

``` javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_yensya2c',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Fill in a description of the device here',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    //configure: tuya.configureMagicPacket,

    exposes: [
        // Here you should put all functionality that your device exposes
        e.presence(),
        e.illuminance_lux(),
        e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        e.binary('breathe_flag', ea.STATE, ['true', 'false']).withDescription('breathe flag'),
        e.binary('small_flag', ea.STATE, ['true', 'false']).withDescription('small flag'),
        e.binary('large_flag', ea.STATE, ['true', 'false']).withDescription('large flag'),
        e.numeric('presence_delay_time', ea.STATE).withDescription('presence_delay_time'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1], //presence_state
            //[12, 'detection_delay', tuya.valueConverter.raw],
            [19, 'target_distance', tuya.valueConverter.divideBy100],
            [20, 'illuminance_lux', tuya.valueConverter.raw],
            [101, 'radar_sensitivity', tuya.valueConverter.raw],
            //[102, 'presence_delay', tuya.valueConverter.raw],
            //[111, 'breath_min', tuya.valueConverter.raw],
            //[112, 'breath_max', tuya.valueConverter.raw],
            [113, 'breathe_flag', tuya.valueConverter.raw],
            [114, 'small_flag', tuya.valueConverter.raw],
            [115, 'large_flag', tuya.valueConverter.raw],
            [116, 'presence_delay_time', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

genik70, 2023-08-27:
Thanks, PunkRa1n for trying. With your code it takes a long time to determine busy and a long time to free. I also experimented with the code and I have errors with presence_delay and presence_time ' No converter available for 'presence_delay' (424)'.

tribalRu, 2023-09-05:
I have the same device. Any progress on the converter?

tribalRu, 2023-09-28:
With the help of Iliya today the converter has obtained the following view:

Almost all the functionality works except for setting "breath_*" and "presence_time" values, because according to the DP numbers shown in the picture, these parameters are missing.
For further development we need a Tui gateway to find out DP numbers of missing parameters. 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const localToZigbee = {
  datapoints: {
      ...tuya.tz.datapoints,
      //key: [...tuya.tz.datapoints.key, 'breathe_flag', 'small_flag', 'large_flag'],
  }
};

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_yensya2c',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Fill in a description of the device here',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [localToZigbee.datapoints],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    //configure: tuya.configureMagicPacket,

    exposes: [
        // Here you should put all functionality that your device exposes
        e.presence(),
        // e.numeric('presence_time', ea.STATE).withDescription('Presence time').withUnit('s'),
        e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        e.illuminance_lux(),
        e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        e.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(3600).withValueStep(1)
            .withDescription('Detection delay').withUnit('s'),
        e.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        e.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        e.binary('breathe_flag', ea.STATE, 'true', 'false').withDescription('Breathe flag'),
        e.binary('small_flag', ea.STATE, 'true', 'false').withDescription('Small flag'),
        e.binary('large_flag', ea.STATE, 'false', 'false').withDescription('Large flag'),
        e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(3600).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            // [12, 'presence_time', tuya.valueConverter.raw],
            [19, 'target_distance', tuya.valueConverter.divideBy100],
            [20, 'illuminance_lux', tuya.valueConverter.raw],
            [101, 'radar_sensitivity', tuya.valueConverter.raw],
            [102, 'detection_delay', tuya.valueConverter.raw],
            [111, 'maximum_range', tuya.valueConverter.divideBy100],
            [112, 'minimum_range', tuya.valueConverter.divideBy100],
            [113, 'breathe_flag', tuya.valueConverter.raw],
            [114, 'small_flag', tuya.valueConverter.raw],
            [115, 'large_flag', tuya.valueConverter.raw],
            [116, 'fading_time', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;`

Teunhen, 2023-09-28:
what i could find with tuya gateway and iot. if we need something else please tell me where to look. :-)

data points:

{"1":"Presence State","12":"Presence Trigger Time","19":"Current Distance","20":"Illuminance Value","101":"Sensitivity","102":"Presence Delay Time","111":"Sensing Distance Min","112":"Sensing Distance Max","113":"breathe flag","114":"small flag","115":"large","116":"Presence Delay Time"}

standard instruction set:

presence_time | Integer | {   "unit": "s",   "min": 1,   "max": 3600,   "scale": 0,   "step": 1 }
-- | -- | --
sensitivity | Integer | {   "unit": "",   "min": 0,   "max": 10,   "scale": 0,   "step": 1 }
presence_delay | Integer | {   "unit": "s",   "min": 5,   "max": 3600,   "scale": 0,   "step": 1 }
breath_min | Integer | {   "unit": "cm",   "min": 0,   "max": 1000,   "scale": 0,   "step": 50 }
breath_max | Integer | {   "unit": "cm",   "min": 50,   "max": 1000,   "scale": 0,   "step": 50 }
breathe_flag | Boolean | "{true,false}"
small_flag | Boolean | "{true,false}"
large_flag | Boolean | "{true,false}"
presence_delay_time | Integer | {   "unit": "S",   "min": 0,   "max": 3600,   "scale": 0,   "step": 1 }

standard status set:

presence_state | Enum | {   "range": [     "none",     "presence"   ] }
-- | -- | --
presence_time | Integer | {   "unit": "s",   "min": 1,   "max": 3600,   "scale": 0,   "step": 1 }
dis_current | Integer | {   "unit": "cm",   "min": 0,   "max": 1000,   "scale": 0,   "step": 1 }
illuminance_value | Integer | {   "unit": "lux",   "min": 0,   "max": 10000,   "scale": 0,   "step": 1 }
sensitivity | Integer | {   "unit": "",   "min": 0,   "max": 10,   "scale": 0,   "step": 1 }
presence_delay | Integer | {   "unit": "s",   "min": 5,   "max": 3600,   "scale": 0,   "step": 1 }
breath_min | Integer | {   "unit": "cm",   "min": 0,   "max": 1000,   "scale": 0,   "step": 50 }
breath_max | Integer | {   "unit": "cm",   "min": 50,   "max": 1000,   "scale": 0,   "step": 50 }
breathe_flag | Boolean | "{true,false}"
small_flag | Boolean | "{true,false}"
large_flag | Boolean | "{true,false}"
presence_delay_time | Integer | {   "unit": "S",   "min": 0,   "max": 3600,   "scale": 0,   "step": 1 }

Westcott1, 2023-10-01:
Thanks Teunhen :)
I've made a couple of small changes to TribalRu's converter.
The detection delay is now in 0.1s increments.
'Presence' is rather slow, but 'target_distance' seems quicker.
Setting debounce to 1 makes it much less chatty.
Sorry, I can't seem to get the insert code working properly.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const localToZigbee = {
   datapoints: {
      ...tuya.tz.datapoints,
      //key: [...tuya.tz.datapoints.key, 'breathe_flag', 'small_flag', 'large_flag'],
  }
};

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_yensya2c',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Loginovo Zigbee Mmwave Human Presence Sensor YENSYA2C',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [localToZigbee.datapoints],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    //configure: tuya.configureMagicPacket,

    exposes: [
        // Here you should put all functionality that your device exposes
        e.presence(),
        e.numeric('presence_time', ea.STATE).withDescription('Presence time').withUnit('s'),
        e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        e.illuminance_lux(),
        e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        e.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        e.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.5)
            .withDescription('Minimum range').withUnit('m'),
        e.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.5)
            .withDescription('Maximum range').withUnit('m'),
        e.binary('breathe_flag', ea.STATE, 'true', 'false').withDescription('Breathe flag'),
        e.binary('small_flag', ea.STATE, 'true', 'false').withDescription('Small flag'),
        e.binary('large_flag', ea.STATE, 'false', 'false').withDescription('Large flag'),
        e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(3600).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [12, 'presence_time', tuya.valueConverter.raw],
            [19, 'target_distance', tuya.valueConverter.divideBy100],
            [20, 'illuminance_lux', tuya.valueConverter.raw],
            [101, 'radar_sensitivity', tuya.valueConverter.raw],
            [102, 'detection_delay', tuya.valueConverter.divideBy10],
            [111, 'minimum_range', tuya.valueConverter.divideBy100],
            [112, 'maximum_range', tuya.valueConverter.divideBy100],
            [113, 'breathe_flag', tuya.valueConverter.raw],
            [114, 'small_flag', tuya.valueConverter.raw],
            [115, 'large_flag', tuya.valueConverter.raw],
            [116, 'fading_time', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

tribalRu, 2023-10-02:
> 'fading_time' can't be set, it always reverts to 30.

experimentally, I have found that "fading_time" is countdown timer and depends from "detection_delay". If sensor detects any movemets "fading_time" takes value from "detection_delay" and starts countdown to 0. All this time "presence_state" takes "true" (present) value, if "fading_time" takes 0 "presence_state" takes "false" (none) value.

Thus, the sensor has no configurable detection time, at least this setting is not included in data points.

kkossev, 2023-10-10:
@tribalRu do you know what the 'breathe_flag', 'small_flag' and 'large_flag' do?

Jonno12345, 2023-10-10:
Hi all,

This is my take on it. Allows for writing the keep_time (how long before presence is marked as False) and detection_delay (how long before presence is marked as True).

fading_time is only intended to be a readonly countdown to how long until the presence is marked as false. 5s seems to be the minimum value for this.

I'm using the latest edge version of Z2M and all parts appear to be functional.

```
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
  // Since a lot of TuYa devices use the same modelID, but use different datapoints
  // it's necessary to provide a fingerprint instead of a zigbeeModel
  fingerprint: [
    {
      // The model ID from: Device with modelID 'TS0601' is not supported
      // You may need to add \u0000 at the end of the name in some cases
      modelID: "TS0601",
      // The manufacturer name from: Device with modelID 'TS0601' is not supported.
      manufacturerName: "_TZE204_yensya2c",
    },
  ],
  model: "TS0601_new",
  vendor: "TuYa",
  description: "Loginovo Zigbee Mmwave Human Presence Sensor YENSYA2C",
  fromZigbee: [tuya.fz.datapoints],
  toZigbee: [tuya.tz.datapoints],
  exposes: [
    e.presence(),
    e.numeric("detection_delay", ea.STATE_SET).withValueMin(1).withValueMax(3600).withValueStep(1)
      .withDescription("Presence Trigger Time").withUnit("s"),
    e.numeric("target_distance", ea.STATE).withDescription("Distance to target").withUnit("m"),
    e.illuminance_lux(),
    e.numeric("radar_sensitivity", ea.STATE_SET).withValueMin(0).withValueMax(10)
      .withValueStep(1).withDescription("Sensitivity of the radar"),
    e.numeric("keep_time", ea.STATE_SET).withValueMin(5).withValueMax(3600).withValueStep(1)
      .withDescription("Presence Delay Time").withUnit("s"),
    e.numeric("minimum_range", ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.5)
      .withDescription("Minimum range").withUnit("m"),
    e.numeric("maximum_range", ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.5)
      .withDescription("Maximum range").withUnit("m"),
    e.binary("breathe_flag", ea.STATE, "true", "false").withDescription("Breathe flag"),
    e.binary("small_flag", ea.STATE, "true", "false").withDescription("Small flag"),
    e.binary("large_flag", ea.STATE, "true", "false").withDescription("Large flag"),
    e.numeric("fading_time", ea.STATE).withDescription("Fading time").withUnit("s"),
  ],
  meta: {
    // All datapoints go in here
    tuyaDatapoints: [
      [1, "presence", tuya.valueConverter.trueFalse1],
      [12, "detection_delay", tuya.valueConverter.raw],
      [19, "target_distance", tuya.valueConverter.divideBy100],
      [20, "illuminance_lux", tuya.valueConverter.raw],
      [101, "radar_sensitivity", tuya.valueConverter.raw],
      [102, "keep_time", tuya.valueConverter.raw],
      [111, "minimum_range", tuya.valueConverter.divideBy100],
      [112, "maximum_range", tuya.valueConverter.divideBy100],
      [113, "breathe_flag", tuya.valueConverter.raw],
      [114, "small_flag", tuya.valueConverter.raw],
      [115, "large_flag", tuya.valueConverter.raw],
      [116, "fading_time", tuya.valueConverter.raw],
    ],
  },
};

module.exports = definition;

```

Jonno12345, 2023-10-10:
> @tribalRu do you know what the 'breathe_flag', 'small_flag' and 'large_flag' do?

These indicate the level of motion detected. I have yet to see breathe_flag trigger, so I'm not sure if these are sensitive enough for this small level of motion as I believe they're 5.8ghz devices rather than 24ghz. Otherwise, small_flag triggers if you're mostly motionless but still present, large_flag triggers if you are making larger motions.

tribalRu, 2023-10-10:
> e.numeric("detection_delay", ea.STATE_SET).withValueMin(1).withValueMax(3600).withValueStep(1)
> e.numeric("fading_time", ea.STATE).withDescription("Fading time").withUnit("s")

It's strange because I have these parameters for the exact opposite roles. I wrote above how I found it. 
Are you sure that fading_time starts the countdown timer?

Jonno12345, 2023-10-10:
> It's strange because I have these parameters for the exact opposite roles. I wrote above how I found it. Are you sure that fading_time starts the countdown timer?

Seems to work perfectly on the devices I have. The detection_delay is purely how long it needs to detect motion for before it treats it as 'presence'. The 'keep_time' sets what the fading_time starts at, and only when that value counts down to zero does it set presence to false. As said, it seems to be working as intended for me, unless the are two iterations of the same device 😟

tribalRu, 2023-10-10:
> Otherwise, small_flag triggers if you're mostly motionless but still present, large_flag triggers if you are making larger motions

I can't confirm.
I specifically tested changing the flags in these fields. This is the result I get:
**'Small flag'** - _'true'_ when motion is detected within the set  **'Minimum range'** value
**'Large flag'** - _'true'_ when motion is detected within the **'Maximum range'** parameter.
**'Breathe flag'** - _'false'_ when there are no objects within the sensor's range, _'true'_ - as long as any motion/object is detected. Somewhat similar to **'Presence'** but the value is changed immediately after the event instead of the **'fading_time'** timer, but with a small delay, I guess it is a factory setting.

Westcott1, 2023-10-10:
Should "detection_delay" be in 0.1 second increments?

Jonno12345, 2023-10-10:
> > Otherwise, small_flag triggers if you're mostly motionless but still present, large_flag triggers if you are making larger motions
> 
> I can't confirm. I specifically tested changing the flags in these fields. This is the result I get: **'Small flag'** - _'true'_ when motion is detected within the set **'Minimum range'** value **'Large flag'** - _'true'_ when motion is detected within the **'Maximum range'** parameter. **'Breathe flag'** - _'false'_ when there are no objects within the sensor's range, _'true'_ - as long as any motion/object is detected. Somewhat similar to **'Presence'** but the value is changed immediately after the event instead of the **'fading_time'** timer, but with a small delay, I guess it is a factory setting.
> 
> Logically, it should be exactly as you wrote, but for some reason I can't prove it by experience.

This is a bit confusing then, we must have different hardware variants with the same fingerprint. I've even tried mine within the Tuya app, which labels them as per what I experience:

[image]

Have you been able to use yours within the app to see if the action is as you expect there too?

tribalRu, 2023-10-10:
> Have you been able to use yours within the app to see if the action is as you expect there too?

sorry, but i don't have tuya hub (

I trust you that it should work as you described. But the my real test doesn't match the application's scenario. That's where I'm confused too. If there are several hardware revisions, it will not be very comfortable. I will try to continue to test, perhaps I have not quite correctly reproduced these events  to make the flags work exactly as you describe.

Teunhen, 2023-10-11:
I have the same interface if i use the hub and app.

Chartreusito, 2023-10-18:
Just received mine today, and tried the converter, my only problem concerns flags which tends to go unavailable.
It's this item : https://fr.aliexpress.com/item/1005006020464433.html
My brand device is Gleco and the model number is GS200 but the manufacturer model is _TZE204_yensya2c too

> Hi all,
>
> This is my take on it. Allows for writing the keep_time (how long before presence is marked as False) and detection_delay (how long before presence is marked as True).
>
> fading_time is only [...]

dexif, 2023-10-22:
Does anyone know if it is possible to turn off the LED in this device?

Chartreusito, 2023-10-25:
I don't know about you folks but this device is flooding my zigbee network so much i'm about to unplug it !
GUI doesn't allow to modify reporting.
Did anyone open a PR for this device ?

upais, 2023-10-25:
> I don't know about you folks but this device is flooding my zigbee network so much i'm about to unplug it !
> GUI doesn't allow to modify reporting.

The same thing, a lot of spam. And the presence determines quite poorly. I removed it from the network.

tribalRu, 2023-10-26:
> Did anyone open a PR for this device ?

It's not the converter, it's the way the device firmware is organized. I'm not sure if anyone here will agree to rewrite it from scratch.

tribalRu, 2023-10-26:
> And the presence determines quite poorly

I didn't notice anything like that. Sensitivity is quite good, detects movements correctly.

Chartreusito, 2023-10-27:
> > And the presence determines quite poorly
> 
> I didn't notice anything like that. Sensitivity is quite good, detects movements correctly.

It works well with me too. It just floods the network doing so. Replaced it with esp32+ ld2410

makeitworktech, 2023-11-02:
I'm noticing that the target distance on this sensor drops to zero and then back up to true distance. Not sure if there's a way to filter out zero data. This is a comparison with the Everything Presence Lite. Here stood at 1m distances from the sensors, and you can see that the _TZE204_yensya2c target distance keeps dropping to zero.
[image]

dexif, 2024-05-27:
Has anyone achieved normal operation of this device? Please share your configuration.

closed 2025-01-13 by github-actions[bot]