Koenkk/zigbee2mqtt #18650: [New device support]: MOES Star Ring ZigBee Smart Dimmer Switch Dimmable 1-3Gang

demcas, 2023-08-18:
### Link

https://moeshouse.com/products/moes-star-ring-zigbee-smart-dimmer-switch-for-light-dimming-work-with-alexa-google-home-dimmable-1-3gang?variant=46040268276027

### Database entry

I don't know how to access this as I'm new to Home Assistant

### Comments

[image]

New Switch WIll not pair

demcas, 2023-08-21:
i get this comment when trying to pair. 
Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_zenj4lxv

farmerkz, 2023-08-26:
> i get this comment when trying to pair. Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_zenj4lxv

Hi!
Same problem.

martindell, 2023-08-31:
Same here with a new 2-gang Moes dimmer.  Error message points to this page: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.htm but it's way beyond my pay grade to figure that out.  Anyone had success creating a suitable convertor?

martindell, 2023-09-05:
Looks like this has just been added to the dev branch:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6136
https://github.com/Koenkk/zigbee-herdsman-converters/blob/23bee5ddfca1f62fc0a03d40fbf788b0b7fe2fc3/CHANGELOG.md?plain=1#L16
https://github.com/Koenkk/zigbee-herdsman-converters/blob/23bee5ddfca1f62fc0a03d40fbf788b0b7fe2fc3/src/devices/tuya.ts#L1265

Koenkk, 2023-09-05:
@martindell thanks for confirming!

closed 2023-09-05 by Koenkk

kuzaxak, 2023-09-22:
> Just installed the edge version of the Home Assistant add-on

@martindell could you give a hint how to do that?

martindell, 2023-09-22:
I'm away from home at the moment, but you have to uninstall the HA
Zigbee2MQTT add-on and reinstall the edge version.  From memory, I
structuons are on the add-on home page

kuzaxak, 2023-09-22:
Added a support for min\max for 3-gang device that I have [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6190)

demcas, 2023-10-14:
@Koenkk @kuzaxak I've been testing the 1,2 & 3 Gang Dimmer Switch and it works great.  An issue I can across is that the device type is listed at LIGHT not SWITCH.  I can't changed this in HA.  Your assistance will be appreciated 
[image]

Koenkk, 2023-10-14:
These are dimmers, so they should be listed as light (a switch doesn't support dimming)

demcas, 2023-10-24:
@Koenkk @kuzaxak Hi, I have been testing the light switches for our community and found a few bugs I want to make you aware of.  Let me know if you need me to open a new ticket.  

Problem 1: The model numbers are inconsistent 
1 Gang  - TS0601_dimmer 1
2 Gang - ZS-SR-EUD-2
3 Gang - ZS-SR-EUD-3

Problem 2: Lighting dimming with a HA dashboard card & Z2MQTT Dashbaord
1 Gang 
- Works flawlessly on/off & diming is very responsive
- Physical wall actions respond and display in Z2MQTT & HA Dashboard
- HA dashboard card and the physical wall switch always in sync with zero latency

2 Gang  
- Only the on/off light function works
- Dimming up and down on the HA dashboard card does dim the light
- Dimming with the physical wall switch works as normal and dimming values change up and down on HA dashboard card
- HA & Z2MQTT dashboard card and the physical wall switch not in sync at times

3 Gang 
- Only the on/off light function works
- Dimming up and down on the HA dashboard card does dim the light
- Dimming with the physical wall switch works as normal and dimming values change up and down on HA dashboard card
- HA & Z2MQTT dashboard card and the physical wall switch not in sync at times

I don't know any thing about coding but the 1 gang identified at TS0601_dimmer operates differently to ZS-SR-EUD.  Could the coding for TS0601 be replicated for the other two switches to fice the issues?

[image]
[image]

kvn1351, 2023-10-26:
I can confirm that the 2-gang Switch can only be turned on/off right now. But I'm very grateful for the efforts so far.

Koenkk, 2023-10-27:
Can you provide the data/database.db entry of your device?

demcas, 2023-10-30:
> Can you provide the data/database.db entry of your device?
@Koenkk 

Here is the Data/Databce.db entries for all three devices

**1 Gang - TS0601_dimmer 1**
{"id":12,"type":"Router","ieeeAddr":"IEEE ADDRESS REMOVED","nwkAddr":51740,"manufId":4417,"manufName":"_TZE204_hlx9tnzb","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"l\u0001\u0000\u0000im\u0001\u0000\u0000in\u0001\u0000\u0000io\u0001\u0000\u0000ip\u0001\u0000\u0000iq\u0001\u0000\u0000ir\u0001\u0000\u0000is\u0001\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE204_hlx9tnzb","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1698114446883,"defaultSendRequestWhen":"immediate"}

**2 Gang - ZS-SR-EUD-2**
{"id":10,"type":"Router","ieeeAddr":"IEE ADDRESS REMOVED","nwkAddr":33865,"manufId":4417,"manufName":"_TZE204_zenj4lxv","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"!��,f\"��,\u0012\"��,e","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE204_zenj4lxv","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1698114443625,"defaultSendRequestWhen":"immediate"}

**3 Gang - ZS-SR-EUD-3**
{"id":13,"type":"Router","ieeeAddr":"IEEE ADDRESS REMOVED","nwkAddr":29034,"manufId":4417,"manufName":"_TZE204_1v1dxkck","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":".\u0000\u0000\u0000i/\u0000\u0000\u0000i1\u0000\u0000\u0000i2\u0000\u0000\u0000i4\u0000\u0000\u0000il\u0001\u0000\u0000im\u0001\u0000\u0000i(\u0000\u0000\u0000i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE204_1v1dxkck","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1698102776937,"defaultSendRequestWhen":"immediate"}

gorstj, 2023-10-30:
I can replicate the issue with teh 2 Gang - ZS-SR-EUD-2 only being able to switch on/off via HA 
(oddly, can control via the zigbee2MQTT web gui but isn't very reliable)

Koenkk, 2023-10-30:
@demcas I'm a bit confused:

> Only the on/off light function works
Dimming up and down on the HA dashboard card does dim the light

First you say on/off only works, next sentence you indicate dimming also works?

Could you clearly state what is not working yet?

demcas, 2023-10-31:
> @demcas I'm a bit confused:
> 
> > Only the on/off light function works
> > Dimming up and down on the HA dashboard card does dim the light
> 
> First you say on/off only works, next sentence you indicate dimming also works?
> 
> Could you clearly state what is not working yet?

Sorry for the confusion.  Hope the issue is clearer with the summary below.\

1 Gang Light Switch
- Light on/off & diming is operating as normal pressing the physical switch 
- On the HA dashboard the On/off is working 
- On the HA dashboard dimming is working 

2 Gang Light Switch
- Light on/off & diming is operating as normal pressing the physical switch
- On the HA dashboard the On/off is working 
- On the HA dashboard **dimming not working** 

3 Gang Light Switch
- Light on/off & diming is operating as normal pressing the physical switch
- On the HA dashboard the On/off is working 
- On the HA dashboard **dimming not working**

Koenkk, 2023-10-31:
@demcas

For the 2/3 gang:
- Does the dimming work from the z2m frontend?
- Can you provide the debug log when trying to dim?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

demcas, 2023-11-01:
> @demcas
> 
> For the 2/3 gang:
> 
> * Does the dimming work from the z2m frontend?
> * Can you provide the debug log when trying to dim?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Hi @Koenkk 

- The Dimming does work from the Z2M frontend only but only if you turn the light on from this screen.  If the light was powered on via the physical switch the dimming function does not work on the Z2M frontend, only the ON/Off works. 
- On the HA dashboard light cards, only the on/off works, dimming does not work.
- Debug Files attached, let me know if i have followed the process correctly.
- 
[Debug File 2.docx](https://github.com/Koenkk/zigbee2mqtt/files/13223541/Debug.File.2.docx)
[Debug File 1.docx](https://github.com/Koenkk/zigbee2mqtt/files/13223542/Debug.File.1.docx)

Koenkk, 2023-11-03:
Let's focus on the z2m frontend first. Could you provide 2 separate debug logs (in txt file, not in docx). Note that I don't need the zigbee-herdsman-debug logging, just the z2m one is enough:

1. Turn on via frontend -> dim (works)
2. Turn on physically -> dim (doesn't work)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

demcas, 2023-11-05:
> Let's focus on the z2m frontend first. Could you provide 2 separate debug logs (in txt file, not in docx). Note that I don't need the zigbee-herdsman-debug logging, just the z2m one is enough:
>
> 1. [...]

Hi @Koenkk 
This is all new to me so I hope I have completed this correctly for you.

[2Gang Dimmer Switch - Turn on frontend dim does not work.txt](https://github.com/Koenkk/zigbee2mqtt/files/13258208/2Gang.Dimmer.Switch.-.Turn.on.frontend.dim.does.not.work.txt)
[2Gang Dimmer Switch - Turn on physically - dim works.txt](https://github.com/Koenkk/zigbee2mqtt/files/13258209/2Gang.Dimmer.Switch.-.Turn.on.physically.-.dim.works.txt)

Koenkk, 2023-11-05:
I'm confused now, I thought dimming doesn't work when turning on physically? The log file name says otherwise

>  If the light was powered on via the physical switch the dimming function does not work on the Z2M frontend, only the ON/Off works.

demcas, 2023-11-05:
> I'm confused now, I thought dimming doesn't work when turning on physically? The log file name says otherwise
> 
> > If the light was powered on via the physical switch the dimming function does not work on the Z2M frontend, only the ON/Off works.

Apologies for any confusion.

Physical Switch - when the light is powered directly from the switch both Dimming and On/Off state works, on the Z2M frontend it displays the switch actions On/Off along the the dimming value change up or down percentage.

Z2M Frontend - Dimming does not work, only ON/OFF state works.

Koenkk, 2023-11-05:
I don't understand why it doesn't work, all looks good. @kirovilya since you added this device in https://github.com/Koenkk/zigbee-herdsman-converters/pull/6136, could you help further here?

demcas, 2023-11-07:
> I don't understand why it doesn't work, all looks good. @kirovilya since you added this device in [Koenkk/zigbee-herdsman-converters#6136](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6136), could you help further here?

It is a strange one.  

Hi @kirovilya let me know how I can assist you in isolating this issue on the 2 Gang and 3 Gang version.  The 1 Gang switch operates flawlessly.  Is it possible to replicate the code from the 1 Gang over to the 2 and 3 Gang switches?

dankocrnkovic, 2023-11-24:
I can confirm that dimming is not working from HA HUI or from Z2M GUI. On/Off is ok. (2 gang)

lverti, 2023-11-25:
I have a 1 gang version and there are 2 issue. 
1) Z2M GUI In switch state off the set brightness is not working. E.g.: state on, brightness 200, switch off, set brightness 100, switch on, the Z2M say brightness is 100 but the real brightness is 200. The others (max brightness, backlightmode) are working in state off.
2) Domoticz, switch on is not working, switch off and toggle are ok. Dimming is same as in Z2M, in state on is ok, in state off not.

dwainegallimore, 2023-11-27:
> I can replicate the issue with teh 2 Gang - ZS-SR-EUD-2 only being able to switch on/off via HA 
> 
> (oddly, can control via the zigbee2MQTT web gui but isn't very reliable)

I have this same experience.

ncodee, 2024-01-04:
Really disappointing to hear. I took several hours trying to find a 3-gang dimmable light switch that works well with HA/Z2M and this was the only one available in the market in the UK.

I'm guessing there is no progress going forwards?

demcas, 2024-01-05:
Just tested and no it has not been fixed.. Not sure what to do here, I installed 40 of these light switched my house and my mother's house, frustrating I can't get the dimming to work on the HA light cards.

beam-d, 2024-01-11:
Hi,

I've been testing a 2-gang Moes dimmer and I got the same issue (or at least I think). I tried to investigate it myself but I have no knowledge of neither z2mqtt nor zigbee and tuya protocols, so maybe it will help someone more knowledgeable. 

Anyway, the behavior is the following :

- On/Off always work, the only issue is with the dimmers
- Dimming on the physical switches works (at first)
- Dimming through Z2MQTT UI will work (at first)
- Dimming on HA always fail.
- After you tried dimming in HA, you will not be able to dim (neither with Z2M UI or physically) until you turn OFF then ON the light.

Now I've tried to investigate what's happening with Z2QMTT logs and I observed the mqtt calls used by HA and z2m UI are different, so I tried doing manual publishes on mqtt and here's what I found:

- Z2M UI uses those kind of publish : `zigbee2mqtt/dimmer1/set` with data `{"brightness_l1":114}`
- HA UI uses those kind of publish : `zigbee2mqtt/dimmer1/l1/set` with data `{"state": "ON", "brightness_l1":114}`

The second kind seems to have issues. Furthermore, the following work perfectly:
- `zigbee2mqtt/dimmer1/l1/set` with data `{"state":"ON"}`
- `zigbee2mqtt/dimmer1/l1/set` with data `{"brightness":100}`

The issue will only arise when publishing both state and brightness. Here, the brightness is ignored, only the state is set (I tried with state ON and OFF and different brightness values).
Note that if I send `{"brightness":100,"state":"ON"}` (inverting the order of the field), it is now brightness that is sent, and state is ignored.

Here are some logs of Z2M. Note the `'set' 'state'`/`'set' 'brightness'`, and also the `dp` (1 being the tuya datapoint for state_l1 and 2 for brightness_l1 - and I get 7 and 8 for l2)
```
Received MQTT message on 'zigbee2mqtt/dimmer1/l1/set' with data '{"state":"ON"}'
Publishing 'set' 'state' to 'dimmer1'
Received Zigbee message from 'dimmer1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":33792}' from endpoint 1 with groupID 0
...
Received MQTT message on 'zigbee2mqtt/dimmer1/l1/set' with data '{"brightness":100}'
Publishing 'set' 'brightness' to 'dimmer1'
Received Zigbee message from 'dimmer1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,138],"type":"Buffer"},"datatype":2,"dp":2}],"seq":34304}' from endpoint 1 with groupID 0
...
Received MQTT message on 'zigbee2mqtt/dimmer1/l1/set' with data '{"state":"ON","brightness":100}'
Publishing 'set' 'brightness' to 'dimmer1'
Received Zigbee message from 'dimmer1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":37120}' from endpoint 1 with groupID 0
...
Received MQTT message on 'zigbee2mqtt/dimmer1/l1/set' with data '{"brightness":100,"state":"ON"}'
Publishing 'set' 'state' to 'dimmer1'
Received Zigbee message from 'dimmer1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,138],"type":"Buffer"},"datatype":2,"dp":2}],"seq":34816}' from endpoint 1 with groupID 0
```

Now, there is a second, different issue that come into the mix, and make thing difficult : if you set state to "ON" while it is already "ON", subsequent commands to the brightness will fail. You'll have to cycle the state OFF -> ON so that it can work again.

Example:
- initial state: light is Off
- `zigbee2mqtt/dimmer1/l1/set` with data `{"state":"ON"}`
-- light is now ON
- `zigbee2mqtt/dimmer1/l1/set` with data `{"brightness":100}`
-- brightness changes to 100
- `zigbee2mqtt/dimmer1/l1/set` with data `{"brightness":10}`
-- brightness changes to 10
- `zigbee2mqtt/dimmer1/l1/set` with data `{"state":"ON"}`
-- nothing changes, it is already on
- `zigbee2mqtt/dimmer1/l1/set` with data `{"brightness":100}`
-- Problem, brightness stays at 10 (with "optimistic" disabled)

- `zigbee2mqtt/dimmer1/l1/set` with data `{"state":"OFF"}`
-- light is now OFF
- `zigbee2mqtt/dimmer1/l1/set` with data `{"state":"ON"}`
-- dimmer is now ON
- `zigbee2mqtt/dimmer1/l1/set` with data `{"brightness":100}`
-- brightness changes to 100 again

Note: this also affect the physically dimmer: after sending `{"state": "ON"}` twice, you need to do an off/on cycle before being able to dim physically (work similarly if you use the physical switch or mqtt for the ON/OFF).

So in fact, with HA UI, this is what happens : 
1. Only the state is send, not the brightness
2. when changing the brightness, we actually send a ON state when already ON, which bugs the brightness control

Koenkk, 2024-01-13:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/19874