Koenkk/zigbee2mqtt #18654: Devices become unreachable randomly

aperezva, 2023-08-18:
### What happened?

My davices every 3 o 4 days become unreachable. After one hour they recover the functionality.

These are the errors:
error 2023-08-17 21:39:08: Publish 'set' 'state' to 'Enchufe Lavavajillas Lexman' failed: 'Error: Command 0x4c5bb3fffe707b9e/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:39:09: Publish 'set' 'state' to 'Enchufe Lavavajillas Lexman' failed: 'Error: Command 0x4c5bb3fffe707b9e/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:39:11: Publish 'set' 'state' to 'Enchufe Lavavajillas Lexman' failed: 'Error: Command 0x4c5bb3fffe707b9e/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:39:15: Publish 'set' 'state' to 'Enchufe Lavavajillas Lexman' failed: 'Error: Command 0x4c5bb3fffe707b9e/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:39:29: Publish 'set' 'state' to 'Luz Piscina' failed: 'Error: Command 0xdc8e95fffe0d4f62/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:39:33: Publish 'set' 'state' to 'Guirnaldas' failed: 'Error: Command 0xa4c138d32d531575/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:39:36: Publish 'set' 'state' to 'Enchufe Televisión Cocina' failed: 'Error: Command 0xa4c138f6c11e1aee/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:40:35: Publish 'set' 'state' to 'Guirnaldas' failed: 'Error: Command 0xa4c138d32d531575/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:40:40: Publish 'set' 'state' to 'Luz Piscina' failed: 'Error: Command 0xdc8e95fffe0d4f62/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:40:44: Publish 'set' 'state' to 'Luz Piscina' failed: 'Error: Command 0xdc8e95fffe0d4f62/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:41:55: Publish 'set' 'state' to 'Depuradora Piscina' failed: 'Error: Command 0xdc8e95fffe3f2fdb/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:42:02: Publish 'set' 'state' to 'Depuradora Piscina' failed: 'Error: Command 0xdc8e95fffe3f2fdb/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:42:02: Publish 'set' 'state' to 'Depuradora Piscina' failed: 'Error: Command 0xdc8e95fffe3f2fdb/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:42:08: Publish 'set' 'state' to 'Depuradora Piscina' failed: 'Error: Command 0xdc8e95fffe3f2fdb/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:42:12: Publish 'set' 'state' to 'Depuradora Piscina' failed: 'Error: Command 0xdc8e95fffe3f2fdb/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:42:18: Publish 'set' 'state' to 'Depuradora Piscina' failed: 'Error: Command 0xdc8e95fffe3f2fdb/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:42:25: Publish 'set' 'state' to 'Depuradora Piscina' failed: 'Error: Command 0xdc8e95fffe3f2fdb/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:42:36: Publish 'set' 'state' to 'Televisión Porche' failed: 'Error: Command 0xa4c138ca85790ac0/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
error 2023-08-17 21:42:37: Publish 'set' 'state' to 'Depuradora Piscina' failed: 'Error: Command 0xdc8e95fffe3f2fdb/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'un [... cut ...] nt":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (25))'
info  2023-08-17 23:40:41: Successfully changed options

### What did you expect to happen?

I expect that no issues regarding this problem appears

### How to reproduce it (minimal and precise)

In this moment , I cnt´reproduce it.

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

20230507

### Adapter

Sonoff Dongle P

jsvensson, 2023-08-22:
#16350 is possibly related. I have similar issues -- devices aren't entirely unreachable, but the Z2M UI stops updating their state. Some devices still react to actions (such as turning a power plug on/off), but the UI doesn't reflect the change and I get similar error messages as above.

My bypass has been to restart Z2M when this happens, but it reappears within hours or days. I'll update this with error messages next time it happens, but they ought to be the same ones as above.

Koenkk, 2023-08-22:
Does it also happen with the `20221226` firmware? (https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin)

aperezva, 2023-08-22:
I don’t know. Any reason to downgraded?

if it needed I can do it. But please give any reason to it. I’m very worried about Zigbee behaviors

Koenkk, 2023-08-22:
> Any reason to downgraded?

This firmware may resolve the instabilities you are experiencing.

aperezva, 2023-08-22:
Ok thanks I will try it tomorrow, and I will give you feedback. 

Thanks for
Your help

jsvensson, 2023-08-22:
Just had it pop up for me again, a couple of hours after the most recent restart.

```
Publish 'set' 'state' to 'power/plug01/couch' failed: 'Error: Command 0xa4c138551cef072c/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
```

So that's a different error message than OP's.

In my case I'm running Z2M in Kubernetes, with a [Zigstar Stick v4](https://zig-star.com/projects/zigbee-stick-v4/). Pretty sure I haven't done any firmware updates since I got it, I can try a different one if that would be helpful for investigating.

aperezva, 2023-08-23:
Hi
I have downgraded to 20221226 version and  in this moment Zigbee is stable. Please don t close this issue before I could confirm in the next days that network is stable. 

@Koenkk why actual coordinator version is not stable or could generate instabilities?

Thanks again

Koenkk, 2023-08-24:
@aperezva because the SDK has been updated (from Texas Instruments). If 20221226 is stable, also try with https://github.com/Koenkk/Z-Stack-firmware/files/12063461/coordinator_20230716.zip

aperezva, 2023-08-24:
Thanks @Koenkk  I prefer to keep this version for a two o three weeks before do another change. I also see in the forums taht this version you are provide me, has some problems after some days. I don´t know maybe is only one person. 

I keep this version on mind to flash it in two weeks.

I guess this new versions solve the problems taht 20230507 version had , correct?

Any other  solved bugs or new functionalities?

closed 2023-10-01 by github-actions[bot]