Koenkk/zigbee2mqtt #18677: [New device support]: _TZE200_v6ossqfy mmwave sensor

thibp, 2023-08-20:
### Link

https://fr.aliexpress.com/item/1005005865536713.html

### Database entry

{"id":45,"type":"Router","ieeeAddr":"0xaaaaaaaaaaaaaaaa","nwkAddr":29231,"manufId":4098,"manufName":"_TZE204_clrdrnya","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000e\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000e\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_clrdrnya","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1692568976161,"defaultSendRequestWhen":"immediate"}

### Comments

Found this post that seems to be a good base : https://github.com/zigpy/zha-device-handlers/issues/2059.
Javicalle posted a good candidate but not for zigbee2mqtt :/

It is like the https://www.zigbee2mqtt.io/devices/TS0601_human_presence_sensor.html but with a relay in addition

thibp, 2023-08-21:
[image]
[image]

evgenln, 2023-08-22:
Title seems incorrect, based on data you posted device is _TZE204_clrdrnya

evgenln, 2023-08-22:
I also looking for data points of this device

evgenln, 2023-08-23:
If you have a Tuya hab you can collect data about datapoints, https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_8-find-your-data-point

thibp, 2023-08-23:
Sorry I don't have any TuYa Hub however I got it working with docker image latest-dev and custom converter based on the source on the issue you sent :
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_clrdrnya']),
        model: 'MTG235-ZB-RL',
        vendor: 'TuYa',
        description: '24G human presence sensor with relay',
        configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        exposes: [
            e.presence(), e.illuminance_lux(),
            e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
            e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1).withDescription('Sensitivity of the radar'),
            e.numeric('entry_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1).withDescription('Entry sensitivity'),
            e.numeric('illuminance_threshold', ea.STATE_SET).withValueMin(0).withValueMax(420).withValueStep(0.1).withUnit('lx')
                .withDescription('Illumination threshold for switching on'),
            e.numeric('detection_range', ea.STATE_SET).withValueMin(0).withValueMax(8).withValueStep(0.1)
                .withUnit('m').withDescription('Detection range'),
            e.numeric('shield_range', ea.STATE_SET).withValueMin(0).withValueMax(8).withValueStep(0.1).withUnit('m')
                .withDescription('Shield range of the radar'),
            e.numeric('entry_distance_indentation', ea.STATE_SET).withValueMin(0).withValueMax(8).withValueStep(0.1).withUnit('m')
                .withDescription('Entry distance indentation'),
            e.numeric('entry_filter_time', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1).withUnit('s')
                .withDescription('Entry filter time'),
            e.numeric('departure_delay', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(1).withUnit('s').
                withDescription('Turn off delay'),
            e.numeric('block_time', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1).withUnit('s').withDescription('Block time'),
            e.binary('breaker_status', ea.STATE_SET, 'ON', 'OFF').withDescription('Breaker status changes with breaker_mode->standard'),
            e.enum('breaker_mode', ea.STATE_SET, ['standard', 'local']).withDescription('Status breaker mode: standard is external, local is auto'),
            e.enum('status_indication', ea.STATE_SET, ['OFF', 'ON']).withDescription('Led backlight when triggered'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'presence', tuya.valueConverter.trueFalse1],
                [2, 'radar_sensitivity', tuya.valueConverter.raw],
                [3, 'shield_range', tuya.valueConverter.divideBy100],
                [4, 'detection_range', tuya.valueConverter.divideBy100],
                [6, 'equipment_status', tuya.valueConverter.raw],
                [9, 'target_distance', tuya.valueConverter.divideBy100],
                [101, 'entry_filter_time', tuya.valueConverter.divideBy10],
                [102, 'departure_delay', tuya.valueConverter.raw],
                [103, 'cline', tuya.valueConverter.raw],
                [104, 'illuminance_lux', tuya.valueConverter.divideBy10],
                [105, 'entry_sensitivity', tuya.valueConverter.raw],
                [106, 'entry_distance_indentation', tuya.valueConverter.divideBy100],
                [107, 'breaker_mode', tuya.valueConverterBasic.lookup({'standard': tuya.enum(0), 'local': tuya.enum(1)})],
                [108, 'breaker_status', tuya.valueConverterBasic.lookup({'OFF': tuya.enum(0), 'ON': tuya.enum(1)})],
                [109, 'status_indication', tuya.valueConverterBasic.lookup({'OFF': tuya.enum(0), 'ON': tuya.enum(1)})],
                [110, 'illuminance_threshold', tuya.valueConverter.divideBy10],
                [111, 'breaker_polarity', tuya.valueConverterBasic.lookup({'NC': tuya.enum(0), 'NO': tuya.enum(1)})],
                [112, 'block_time', tuya.valueConverter.divideBy10],
                [113, 'parameter_setting_result', tuya.valueConverter.raw],
                [114, 'factory_parameters', tuya.valueConverter.raw],
                [115, 'sensor', tuya.valueConverter.onOff],
            ],
        },
    };

module.exports = definition;
```

thibp, 2023-08-23:
I only have issue on modifying illuminance threshold "No converter available for 'illuminance_threshold' ({value here})"
I also noticed that distance is not refreshed often and radar sensitivity or other sensivity parameters does not seems to have an impact.
Otherwise it seems to be working well.

evgenln, 2023-08-24:
Sorry, I'm not z2m user so cannot help with it. I was looking for the data points info in order to create ZHA handler. Also, from my tests, this device data points accepts a bit different values compared to the source converter.  You can update on your side if want 

departure_delay > min 2 - max 1500 s
block_time >15-600 ms
entry_distance_indentation > 0-1220 cm
detection_range > 240-1250 cm
shield_range > 30-1240 cm

dp 110, the illuminance threshold accepts any value but resets to 4124, 1446, 446, 3445 and so on, on overloaded values like 999999 so I believe the currently used range is  OK

evgenln, 2023-08-25:
@thibp Get a chance to take a look at the code you posted, the issue with "No converter available for 'illuminance_threshold' ({value here})" because if used
`  toZigbee: [tuya.tz.datapoints],  `
 threshold attribute should be 
```
 e.numeric('illumin_threshold', ea.STATE_SET).withValueMin(0).withValueMax(420).withValueStep(0.1).withUnit('lx')
                .withDescription('Illumination threshold for switching on')
```
If you want a more sexy name for it add const like this
`const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [...tuya.tz.datapoints.key, 'illuminance_threshold']
}`
and refer to it in toZigbee
`toZigbee: [tzDatapoints],`

OakfieldDrive, 2023-08-27:
I’m also stuck with this device that isn’t yet supported by Zm2mqtt. Is there a way to add support to it without customisation?

evgenln, 2023-08-27:
no, use this converter yet

OakfieldDrive, 2023-08-27:
> no, use this converter yet

Thank you for your feedback. Any tutorial around on how to that? Apologies but I’m not an advanced user.

OakfieldDrive, 2023-09-02:
@evgenln @thibp thanks for this post. with a bit of research found my way to set-up the external converter (.js file next to zigbee2mqtt with the code above). I also got the same issue with regards "No converter available for 'illuminance_threshold'; the parts of the code from @evgenln seemed identical to the original but I replaced it anyway and guess what? it fixed the illuminance issue. it seems to be working fine for me too. Tested speed of detection and works fine. Relays works fine too. the Target distance doesn't not seem to be able to measure accurately. It goes from zero to 2.83m and doesn't vary.

It there a process to add native support and avoid these manual work around steps?

thibp, 2023-09-04:
@OakfieldDrive Here a working converter. I tried to adapt the code based of @evgenln messages.
It seems to have this issues :
- Approximative target distance issue, like you mentionned
- Backlight parameter not lighting the light on the cover. Is this working? (mine is installed and I cannot flip it)

To have the device working :
1. Create a file `_TZE204_clrdrnya.js`, in the same folder that contains config, and paste this content :
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_clrdrnya']),
        model: 'MTG235-ZB-RL',
        vendor: 'TuYa',
        description: '24G human presence sensor with relay',
        configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        exposes: [
            e.presence(), e.illuminance_lux(),
            e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
            e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1).withDescription('Sensitivity of the radar'),
            e.numeric('entry_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1).withDescription('Entry sensitivity'),
            e.numeric('illumin_threshold', ea.STATE_SET).withValueMin(0).withValueMax(420).withValueStep(0.1).withUnit('lx')
                .withDescription('Illumination threshold for switching on'),
            e.numeric('detection_range', ea.STATE_SET).withValueMin(240).withValueMax(1250).withValueStep(10)
                .withUnit('cm').withDescription('Detection range'),
            e.numeric('shield_range', ea.STATE_SET).withValueMin(30).withValueMax(1240).withValueStep(10).withUnit('cm')
                .withDescription('Shield range of the radar'),
            e.numeric('entry_distance_indentation', ea.STATE_SET).withValueMin(0).withValueMax(1220).withValueStep(10).withUnit('cm')
                .withDescription('Entry distance indentation'),
            e.numeric('entry_filter_time', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1).withUnit('s')
                .withDescription('Entry filter time'),
            e.numeric('departure_delay', ea.STATE_SET).withValueMin(2).withValueMax(1500).withValueStep(1).withUnit('s').
                withDescription('Turn off delay'),
            e.numeric('block_time', ea.STATE_SET).withValueMin(1.5).withValueMax(60).withValueStep(0.1).withUnit('s').withDescription('Block time'),
            e.binary('breaker_status', ea.STATE_SET, 'ON', 'OFF').withDescription('Breaker status changes with breaker_mode->standard'),
            e.enum('breaker_mode', ea.STATE_SET, ['standard', 'local']).withDescription('Status breaker mode: standard is external, local is auto'),
            e.enum('status_indication', ea.STATE_SET, ['OFF', 'ON']).withDescription('Led backlight when triggered'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'presence', tuya.valueConverter.trueFalse1],
                [2, 'radar_sensitivity', tuya.valueConverter.raw],
                [3, 'shield_range', tuya.valueConverter./*divideBy100*/raw], 
                [4, 'detection_range', tuya.valueConverter./*divideBy100*/raw],
                [6, 'equipment_status', tuya.valueConverter.raw],
                [9, 'target_distance', tuya.valueConverter./*divideBy100*/raw],
                [101, 'entry_filter_time', tuya.valueConverter.divideBy10],
                [102, 'departure_delay', tuya.valueConverter.raw],
                [103, 'cline', tuya.valueConverter.raw],
                [104, 'illuminance_lux', tuya.valueConverter.divideBy10],
                [105, 'entry_sensitivity', tuya.valueConverter.raw],
                [106, 'entry_distance_indentation', tuya.valueConverter./*divideBy100*/raw],
                [107, 'breaker_mode', tuya.valueConverterBasic.lookup({'standard': tuya.enum(0), 'local': tuya.enum(1)})],
                [108, 'breaker_status', tuya.valueConverterBasic.lookup({'OFF': tuya.enum(0), 'ON': tuya.enum(1)})],
                [109, 'status_indication', tuya.valueConverterBasic.lookup({'OFF': tuya.enum(0), 'ON': tuya.enum(1)})],
                [110, 'illumin_threshold', tuya.valueConverter.divideBy10],
                [111, 'breaker_polarity', tuya.valueConverterBasic.lookup({'NC': tuya.enum(0), 'NO': tuya.enum(1)})],
                [112, 'block_time', tuya.valueConverter.divideBy10],
                [113, 'parameter_setting_result', tuya.valueConverter.raw],
                [114, 'factory_parameters', tuya.valueConverter.raw],
                [115, 'sensor', tuya.valueConverter.onOff],
            ],
        },
    };

module.exports = definition;

```
2. Add the file name `_TZE204_clrdrnya.js` to External converters section in Parameters on the web interface.
Hope this will help.
3. Reboot the server

However, if somebody can add this converter, or enhance it to have it natively, this would be great!

evgenln, 2023-09-10:
Inaccurate target distance seems issue of the device itself I see the same behavior in zha, at least it is less spammy compared to _TZE204_sxm7l9xa. 
When I bought the device Backlight seemed to be off but exploring data points I turned it on and now cannot set it off as well. 
One more thing that can be improved is the Sensor option, it is not only onOff but allows 4 states: 'turn_on', 'turn_off', 'occupied', and 'unoccupied'. I'm not sure someone will use it but it's a way to e.g. turn on/off the light fast regardless of the situation. To get it working you can add/change the next lines

`e.enum('sensor', ea.STATE_SET, ['turn_on', 'turn_off', 'occupied', 'unoccupied']).withDescription('Status breaker mode: standard is external, local is auto'),`

`[115, 'sensor', tuya.valueConverterBasic.lookup({'turn_on': tuya.enum(0), 'turn_off': tuya.enum(1), 'occupied': tuya.enum(2), 'unoccupied': tuya.enum(3)})],
`

LuisAlbertoFP, 2023-10-03:
This code worked for me:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [...tuya.tz.datapoints.key, 'radar_sensitivity', 'entry_sensitivity', 'illumin_threshold', 'detection_range','shield_range','entry_distance_indentation','entry_filter_time','departure_delay','block_time', 'breaker_polarity', 'status_indication','breaker_mode','se sensor']
}

const definition = {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_clrdrnya']),
        model: 'MTG235-ZB-RL',
        vendor: 'TuYa',
        description: '24G human presence sensor with relay',
        configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tzDatapoints],
        exposes: [
            e.presence(), e.illuminance_lux(),
            e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
            e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1).withDescription('Sensitivity of the radar'),
            e.numeric('entry_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1).withDescription('Entry sensitivity'),
            e.numeric('illumin_threshold', ea.STATE_SET).withValueMin(0).withValueMax(420).withValueStep(0.1).withUnit('lx')
                .withDescription('Illumination threshold for switching on'),
            e.numeric('detection_range', ea.STATE_SET).withValueMin(240).withValueMax(1250).withValueStep(10)
                .withUnit('cm').withDescription('Detection range'),
            e.numeric('shield_range', ea.STATE_SET).withValueMin(30).withValueMax(1240).withValueStep(10).withUnit('cm')
                .withDescription('Shield range of the radar'),
            e.numeric('entry_distance_indentation', ea.STATE_SET).withValueMin(0).withValueMax(1220).withValueStep(10).withUnit('cm')
                .withDescription('Entry distance indentation'),
            e.numeric('entry_filter_time', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1).withUnit('s')
                .withDescription('Entry filter time'),
            e.numeric('departure_delay', ea.STATE_SET).withValueMin(2).withValueMax(1500).withValueStep(1).withUnit('s').
                withDescription('Turn off delay'),
            e.numeric('block_time', ea.STATE_SET).withValueMin(1.5).withValueMax(60).withValueStep(0.1).withUnit('s').withDescription('Block time'),
            e.binary('breaker_status', ea.STATE_SET, 'ON', 'OFF').withDescription('Breaker status changes with breaker_mode->standard'),
            e.enum('breaker_mode', ea.STATE_SET, ['standard', 'local']).withDescription('Status Breaker mode'),
            e.enum('status_indication', ea.STATE_SET, ['ON', 'OFF']).withDescription('Led backlight when triggered'),
			e.enum('sensor', ea.STATE_SET, ['turn_on', 'turn_off', 'occupied', 'unoccupied']).withDescription('The radar sensor can be set in four states: turn on, turn off, occupied, and unoccupied. For example, if occupied is set up, it will continue to maintain a presence regardless of whether someone present or not. If unoccupied is set, the unoccupied state will be maintained continuously.'),
		
			
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'presence', tuya.valueConverter.trueFalse1],
                [2, 'radar_sensitivity', tuya.valueConverter.raw],
                [3, 'shield_range', tuya.valueConverter./*divideBy100*/raw], 
                [4, 'detection_range', tuya.valueConverter./*divideBy100*/raw],
                [6, 'equipment_status', tuya.valueConverter.raw],
                [9, 'target_distance', tuya.valueConverter./*divideBy100*/raw],
                [101, 'entry_filter_time', tuya.valueConverter.divideBy10],
                [102, 'departure_delay', tuya.valueConverter.raw],
                [103, 'cline', tuya.valueConverter.raw],
                [104, 'illuminance_lux', tuya.valueConverter.divideBy10],
                [105, 'entry_sensitivity', tuya.valueConverter.raw],
                [106, 'entry_distance_indentation', tuya.valueConverter./*divideBy100*/raw],
			    [107, 'breaker_mode', tuya.valueConverterBasic.lookup({'standard': tuya.enum(0), 'local': tuya.enum(1)})],
                [108, 'breaker_status', tuya.valueConverterBasic.lookup({'ON': tuya.enum(1), 'OFF': tuya.enum(0)})],
                [109, 'status_indication', tuya.valueConverterBasic.lookup({'ON': tuya.enum(1), 'OFF': tuya.enum(0)})],
                [110, 'illumin_threshold', tuya.valueConverter.divideBy10],
                [112, 'block_time', tuya.valueConverter.divideBy10],
                [113, 'parameter_setting_result', tuya.valueConverter.raw],
                [114, 'factory_parameters', tuya.valueConverter.raw],
[... cut ...] XPn50rxcyXYD3UkfSaQUvCqQhujNfoEudX3/E7y6quxr/jKNkolFY1H+Xfd+o7d+UbGcZqeNFb4XsgrYqQbKyGN1f6GqiAOGOFEAjeit/CaHm0QgoE1171AjR0offRIuRdc/Ufw2hVn+15/uvebHoYBGEwvBF1m8ABTYsNMcTEe///vxvVnWay6QIe9h65vHy3oQ9HhWBIZ6gBBQf4y9xjTTqH/4VaBDUeHXyfj0DQ0YGvdu7DFLabB6gr5URQyI7gd26DCsF32SkssvcU5Zqv7gomb4owYGSdgADDdh2W4DjKXZ+i8tAWI9B0MMgzpyysur1di43ywDwpE4tCiBSNmmaRFEVR3oUxpSLM0gLOZjnwnztga9OlVHexfIkZVG/qGIrbC14dFvTV2rXQLAycjdJGdMaPkvV5nDbS+iT3/9ITvYOPrlg8bjYAAAAASUVORK5CYII=',
    };
	
module.exports = definition;
`

Koenkk, 2023-10-03:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-03 by Koenkk

CH-Tefi, 2024-01-04:
Has anyone a working zha py file for TS0601 _TZE204_clrdrnya mmwave sensor?

bsaurusrex, 2024-01-29:
I find the sensitivity to not work in the main branch.  This sensor detects my dog even when on sensitivity = 0.

patrykgiera, 2024-02-23:
Rowniez czekam na plik do ZHA PY