Koenkk/zigbee2mqtt #18681: Crashes with BUFFER_FULL

tkarlsson, 2023-08-21:
### What happened?

I'm starting to see these crashes on a regular basis relating to buffer full

### Zigbee2MQTT version

1.32.2-dev commit: 71c4211

### Adapter firmware version

20230507

### Adapter

Sonoff

### Debug log

error 2023-08-21 00:23:27: Exception while calling fromZigbee converter: CommandResponse 0x00000000e203e547/242 greenPower.response({"options":0,"tempMaster":1085,"tempMasterTx":0,"srcID":3791906119,"gpdCmd":254,"gpdPayload":{"commandID":254,"buffer":{"type":"Buffer","data":[0]}}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":242,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":202,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[25,211,6,0,61,4,0,71,229,3,226,254,1,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))}
debug 2023-08-21 00:23:27: Error: CommandResponse 0x00000000e203e547/242 greenPower.response({"options":0,"tempMaster":1085,"tempMasterTx":0,"srcID":3791906119,"gpdCmd":254,"gpdPayload":{"commandID":254,"buffer":{"type":"Buffer","data":[0]}}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":242,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":202,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[25,211,6,0,61,4,0,71,229,3,226,254,1,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

yayitazale, 2023-09-18:
I have this issue since various month and I have to reboot the container almost every 2 days. 

Zigbee2mqtt version:
    [1.33.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.0) commit: [7ee207f](https://github.com/Koenkk/zigbee2mqtt/commit/7ee207f)

Coordinator type
    zStack3x0

Coordinator version
    20230507

Coordinator IEEE Address
    0x00124b0021cbff52

Total  devices: 102

End devices: 52
Routers: 50

Koenkk, 2023-09-18:
Please check https://github.com/Koenkk/Z-Stack-firmware/discussions/474#discussioncomment-7031686

closed 2023-10-27 by github-actions[bot]

mremedi2023, 2024-08-30:
Same problem here