Koenkk/zigbee2mqtt #18682: [New device support]: TS0202 Presence sensor  _TZ3000_jmrgyl7o

matze20211, 2023-08-21:
### Link

https://zigbee.blakadder.com/Luminea_ZX-5311.html

### Database entry

"id":54,"type":"EndDevice","ieeeAddr":"0xa4c138fdce2d95c2","nwkAddr":19935,"manufId":4417,"manufName":"_TZ3000_jmrgyl7o","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b0029dede2c","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1692611277927,"defaultSendRequestWhen":"immediate"}

### Comments

Hi, these device is already in [deconz](https://dresden-elektronik.github.io/deconz-rest-doc/devices/tuya/ts0202_presence_sensor/)

`{
   "schema":"devcap1.schema.json",
   "manufacturername": ["_TYZB01_tv3wxhcz","_TYZB01_hqbdru35","TZ3000_tiwq83wk","_TZ3000_ykwcwxmz","_TZ3000_hgu1dlak","_TZ3000_hktqahrq","_TZ3000_kmh5qpmb",
                        "_TZ3000_jmrgyl7o","_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3000_6ygjfyll", "_TZ3040_6ygjfyll", "_TZ3000_mcxw5ehu","_TYZB01_ef5xlc9q",
                        "_TYZB01_vwqnz1sn","_TYZB01_2b8f6cio","_TZE200_bq5c8xfe","_TYZB01_dl7cejts","_TYZB01_qjqgmqxr","_TZ3000_mmtwjmaq","_TYZB01_zwvaj5wy",
                        "_TZ3000_bsvqrxru"],
   "modelid":["TS0202", "TS0202", "TS0202", "TS0202", "TS0202","TS0202", "TS0202", "TS0202", "TS0202", "TS0202","TS0202", "TS0202", "TS0202", "TS0202",
             "TS0202", "TS0202", "TS0202", "TS0202", "TS0202","TS0202", "TS0202", "TS0202"],
   "product":"TS0202 Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint": {
           "profile": "0x0104",
           "device": "0x0402",
           "endpoint": "0x01",
           "in": [
             "0x0000",
             "0x0001",
             "0x0500"
           ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
              "name": "attr/swversion",                   
              "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
              "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "default":0
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":3600,
               "max":14400,
               "change":"0x00000001"
            }
         ]
      }
   ]
}`

matze20211, 2023-09-02:
i try  to parse messages it with this converter, but the only message i receive is:

`2023-09-02 12:19:14Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
`
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0202'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: ['ZX-5311'], // Vendor model number, look on the device for a model number
    vendor: ['Luminea'], // Vendor of the device (only used for documentation and startup logging)
    description: ['Oudoor PIR Sensor'], // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.presence(), e.battery_low(), e.battery(), e.voltage(), e.linkquality()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

```

Koenkk, 2023-09-02:
Try with: `fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],`

matze20211, 2023-09-02:
Now, i received this:
```

debug 2023-09-02 14:46:16Device '0xa4c138fdce2d95c2' announced itself
info 2023-09-02 14:46:16MQTT publish: topic 'zigbee2mqDD/bridge/event', payload '{"data":{"friendly_name":"0xa4c138fdce2d95c2","ieee_address":"0xa4c138fdce2d95c2"},"type":"device_announce"}'
debug 2023-09-02 14:46:19Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
debug 2023-09-02 14:46:19Received Zigbee message from '0xa4c138fdce2d95c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2023-09-02 14:46:21Received Zigbee message from '0xa4c138fdce2d95c2', type 'attributeReport', cluster 'ssIasZone', data '{"61441":1}' from endpoint 1 with groupID 0
debug 2023-09-02 14:46:21Received Zigbee message from '0xa4c138fdce2d95c2', type 'attributeReport', cluster 'ssIasZone', data '{"currentZoneSensitivityLevel":1}' from endpoint 1 with groupID 0
debug 2023-09-02 14:46:29Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
debug 2023-09-02 14:47:29Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
debug 2023-09-02 14:48:29Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
debug 2023-09-02 14:46:21Received Zigbee message from '0xa4c138fdce2d95c2', type 'attributeReport', cluster 'ssIasZone', data '{"61441":1}' from endpoint 1 with groupID 0
debug 2023-09-02 14:46:21Received Zigbee message from '0xa4c138fdce2d95c2', type 'attributeReport', cluster 'ssIasZone', data '{"currentZoneSensitivityLevel":1}' from endpoint 1 with groupID 0
debug 2023-09-02 14:46:29Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
debug 2023-09-02 14:47:29Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
debug 2023-09-02 14:48:29Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
debug 2023-09-02 14:49:47Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
debug 2023-09-02 14:50:09Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
debug 2023-09-02 14:50:28Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
```

Koenkk, 2023-09-03:
I would expect MQTT messages to be published now, is this not the case?

matze20211, 2023-09-03:
There is no MQTT message published

matze20211, 2023-09-03:
here the actual converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0202'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZX-5311', // Vendor model number, look on the device for a model number
    vendor: 'Luminea', // Vendor of the device (only used for documentation and startup logging)
    description: 'Oudoor PIR Sensor', // Description of the device, copy from vendor site. (only used fo>
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_>
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.occupancy(), e.battery_low(), e.battery(), e.voltage(), e.linkquality()], // Defines wha>
};

module.exports = definition;

```

Koenkk, 2023-09-04:
Did you add it to the `configuration.yaml`?

Could you check if the issue is fixed with the following external converter: 

- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

matze20211, 2023-09-05:
its already in configuration.yaml:

```
...
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
  external_converters:
    - ZX-5311.js
device_options:
  legacy: false
..
```

Koenkk, 2023-09-05:
It should not be under advanced, change it to:

```yaml
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
external_converters:
  - ZX-5311.js
device_options:
  legacy: false
```

matze20211, 2023-09-06:
now it works:

```
debug 2023-09-06 08:48:27Device '0xa4c138fdce2d95c2' announced itself
info 2023-09-06 08:48:27MQTT publish: topic 'zigbee2mqDD/bridge/event', payload '{"data":{"friendly_name":"0xa4c138fdce2d95c2","ieee_address":"0xa4c138fdce2d95c2"},"type":"device_announce"}'
debug 2023-09-06 08:48:29Retrieving state of '0xa4c138fdce2d95c2' after reconnect
debug 2023-09-06 08:48:30Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
info 2023-09-06 08:48:30MQTT publish: topic 'zigbee2mqDD/0xa4c138fdce2d95c2', payload '{"battery_low":false,"linkquality":25,"occupancy":false,"tamper":false}'
debug 2023-09-06 08:48:30Received Zigbee message from '0xa4c138fdce2d95c2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2023-09-06 08:48:32Received Zigbee message from '0xa4c138fdce2d95c2', type 'attributeReport', cluster 'ssIasZone', data '{"61441":1}' from endpoint 1 with groupID 0
info 2023-09-06 08:48:32MQTT publish: topic 'zigbee2mqDD/0xa4c138fdce2d95c2', payload '{"battery_low":false,"linkquality":32,"occupancy":false,"tamper":false}'
debug 2023-09-06 08:48:32Received Zigbee message from '0xa4c138fdce2d95c2', type 'attributeReport', cluster 'ssIasZone', data '{"currentZoneSensitivityLevel":1}' from endpoint 1 with groupID 0
info 2023-09-06 08:48:32MQTT publish: topic 'zigbee2mqDD/0xa4c138fdce2d95c2', payload '{"battery_low":false,"linkquality":36,"occupancy":false,"tamper":false}'
debug 2023-09-06 08:48:41Received Zigbee message from '0xa4c138fdce2d95c2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
info 2023-09-06 08:48:41MQTT publish: topic 'zigbee2mqDD/0xa4c138fdce2d95c2', payload '{"battery_low":false,"linkquality":40,"occupancy":true,"tamper":false}'
```

Koenkk, 2023-09-06:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-06 by Koenkk