Koenkk/zigbee2mqtt #18684: (0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS) with CC2652RB_coordinator_20230507.zip

adekloet, 2023-08-21:
### What happened?

igbee2MQTT:error 2023-08-20 23:31:15: Publish 'set' 'state' to 'L8 fontijn Stopcontact Z2MQTT' failed: 'Error: Command 0x7cb03eaa0a0c06ae/3 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":59576,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'

[CC2652RB_coordinator_20230507.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC2652RB_coordinator_20230507.zip)

95 Devices / routers and sensors
Stick: RB Sleash stick

[image]

### What did you expect to happen?

No error on this amount of devices. It would be nice if Z2MQTT goes into a stop state or restarts as it "hangs" in a running state. (with a delay avoiding a race condition with stop/starts)

### How to reproduce it (minimal and precise)

random it seems. Z2MQTT Add-on stays in a running state. Memory starts increasing. I reset Z2MQTT with a automation once it hits 8% mem use on my system now as a resolution.

### Zigbee2MQTT version

1.32.2-1 

### Adapter firmware version

20230507

### Adapter

Slaesh RB stick

### Debug log

There are no other indication apart from the log: (0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS) and clearly Z2MQTT hangs and increases it's memery used. After the event it does not recover and "hangs" in a running state.

BradleyFord, 2023-08-25:
This is an open issue with the Z-Stack firmare; please check that thread.
https://github.com/Koenkk/Z-Stack-firmware/issues/439

BradleyFord, 2023-08-25:
Ah now I see it has been posted in 2 places; can I request that you close this issue?

adekloet, 2023-08-25:
@BradleyFord  Indeed; forgot to close this one, I will do it right away thanks for the heads up!

adekloet, 2023-08-25:
Closing see other thread.

closed 2023-08-25 by adekloet