Koenkk/zigbee2mqtt #18685: Zemismart ZM25RX-08/30 (TS0601/_TZE200_7eue9vhc): No battery info

chris1h1, 2023-08-21:
### What happened?

This issue follows on from the closed (stale) [https://github.com/Koenkk/zigbee2mqtt/issues/16942](https://github.com/Koenkk/zigbee2mqtt/issues/16942).

I have a tubular motor from Zemismart (see below); everything works fine, but I can't get the battery info to work, I just get a question mark.
Manufacturer: Zemismart
Model: ZM25RX-08/30
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_7eue9vhc

### What did you expect to happen?

Show the battery info in percent.

I bought a Tuya Zigbee gateway and followed the instructions at https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html; the resultant data follows:

```json
{
    "1": "Control",
    "101": "设置限位点",
    "106": "行程状态",
    "11": "Situation_set",
    "12": "Fault",
    "13": "Battery Percentage",
    "2": "Curtain position setting",
    "3": "Current curtain position",
    "4": "Mode",
    "5": "Motor Direction",
    "7": "Work State"
}
```

Per Google Translate:

- 设置限位点 = set limit point
- 行程状态 = trip status

I tried to create a local external converter definition for this device, but was unsuccessful.

### How to reproduce it (minimal and precise)

Add a Zemismart ZM25RX-08/30 to Zigbee2MQTT; the "Power" attribute will be "Battery ?".

### Zigbee2MQTT version

1.32.1 commit: 83205e43

### Adapter firmware version

20230507

### Adapter

smlight-slzb-05 (CC2652P)

### Debug log

N/A

nimmerpeople, 2023-08-21:
Thanks for opening a new one. Chris. Appreciated

chris1h1, 2023-08-21:
Possibly related to [https://github.com/Koenkk/zigbee2mqtt/issues/7630](https://github.com/Koenkk/zigbee2mqtt/issues/7630).

chris1h1, 2023-08-28:
Possible starting point for a working external converter: [https://github.com/Koenkk/zigbee2mqtt/issues/17436#issuecomment-1600901453](https://github.com/Koenkk/zigbee2mqtt/issues/17436#issuecomment-1600901453).

chris1h1, 2023-08-28:
Also related: [https://github.com/Koenkk/zigbee2mqtt/issues/18413](https://github.com/Koenkk/zigbee2mqtt/issues/18413).

chris1h1, 2023-09-28:
> This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 7 days

Commenting to avoid closure.

DariBer, 2023-10-03:
anyone who can help to edit an converter with the info above?
to this one https://www.zigbee2mqtt.io/devices/ZM25RX-08_30.html

chris1h1, 2023-10-24:
To avoid this being marked as stale, perhaps [https://gist.github.com/Koenkk/79780d4380c01fff06c93769e99e2e2c](https://gist.github.com/Koenkk/79780d4380c01fff06c93769e99e2e2c) could be used as a template.

donbobka, 2023-10-26:
@DariBer  
Can you try this one https://gist.github.com/donbobka/513167a4599c7dc1f74fde0de147bee7? If it's ok, I'll create PR.

Just create file `tuya-blinds.js` in same folder with configuration.yaml and then add external converted to your config 
```
external_converters:
  - tuya.js
```

I've added set limits through zigbee (tested on my _TZE200_bv1jcqqu) and also added battery, but mine motor doesn't send it

donbobka, 2023-10-26:
@DariBer, @chris1h1
Can you try this one https://gist.github.com/donbobka/513167a4599c7dc1f74fde0de147bee7? If it's working for you, I'll create PR.

Just create file `tuya-blinds.js` in same folder with configuration.yaml and then add external converted to your config 
```
external_converters:
  - tuya-blinds.js
```

I've added set limits through zigbee (tested on my _TZE200_bv1jcqqu) and also added battery, but mine motor doesn't send it

DariBer, 2023-10-28:
@donbobka, I tested now, and unfortunatly, it doesn't work. after I  added the external converter, I got instead of "Battery ?" now it's an red blinking ?

This is how it was before I installed your converter 
[image]

This is how it looks like after I installed your converter
[image]

donbobka, 2023-10-28:
@DariBer have you tried to use remote to control motor? According to https://github.com/Koenkk/zigbee2mqtt/issues/11251#issuecomment-1161218120 some motors send battery info only when you use remote buttons

DariBer, 2023-10-30:
@donbobka, That didn't help. I tried on both that I have. Doesn't help. I don't get any battery info/status

donbobka, 2023-10-30:
It looks like it's same as my blinds: motor just never sends it. Do you have tuya hub? Is it visible there?

JB75France, 2023-11-02:
Yes it is. And it's relevant on Tuya.  :)

donbobka, 2023-11-18:
@JB75France If you can [sniff zigbee traffic](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) between tuya hub and motor then maybe we can figure out how make it work in zigbee2mqtt?

JB75France, 2023-11-20:
yes I will try in a few days.

stevenjev, 2023-11-21:
I believe this is a related issue, but apologies if not. 

I too have these roller shades and I'm experiencing pretty fast battery drain on them once paired to Z2M. Looking in my logs, it seems the shades are frequently messaging and (maybe?) not receiving a response due to the missing converter. I wonder if the shades are expecting a response. Roughly every minute, the lights on the roller shade light up without any commands issued.

```

Received Zigbee
message from 'Master Bedroom Shades', type
'commandMcuVersionResponse', cluster
'manuSpecificTuya', data
'{"seq" :256, "version":65}' from endpoint 1 with groupID 0
Debug
2023-11-20 22:05:26 No converter available
for 'ZM25RX-08/30' with cluster
'manuSpecificTuya' and type
'commandMcuVersionResponse' and data
'{"seq" :256, "version": 65}'

```

donbobka, 2023-11-21:
@stevenjev I think it's a little bit unrelated issue, but i also found this when playing with some PRs for that blinds.
And i think I even found the reason: [this line](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/tuya.ts#L46) triggers almost infinite loop of requests for version. @Koenkk do you know why we need that line? For me it looks strange that when we receive message `commandMcuVersionResponse` we again trigger `mcuGatewayConnectionStatus`

Koenkk, 2023-11-21:
@donbobka the naming of the commands is wrong I think, but the temperature sensor that was added here required this (and I think it's according to the TuYa protocol). If I understand correctly, with this line commented battery reporting started working for this device?

donbobka, 2023-11-21:
> If I understand correctly, with this line commented battery reporting started working for this device?
Nope :) Let me create separate issue for that

TRusselo, 2023-12-02:
> yes I will try in a few days.

any update?
I would sniff traffic if i could, but I dont have a tuya hub.

deku-m, 2024-01-13:
Any luck in solving this? Would love to see this too.

cannondale0815, 2024-02-09:
I have the same blinds. Funny enough, the [z2m page](https://www.zigbee2mqtt.io/devices/ZM25RX-08_30.html#zemismart-zm25rx-08%252F30) for this model says that battery status should appear within 24 hours, but it doesn't. And every time I move the blinds manually using the included RF remote, I can see the battery percentage symbol blinking in the z2m web UI, but then it goes back to a question mark after some time.

cannondale0815, 2024-02-19:
I bought a Tuya Zigbee gateway to see if battery info shows up in the official Tuya/SmartLife app. It does, but it's buried in the settings for each device (not a front-and-center attribute) and is also just a crude battery symbol, not a percentage number.

And when I use the native Tuya integration in HA, there's no battery information shown there for these shades either.

iSetebos, 2024-05-26:
I have a problem with TS0601
_TZE200_9yapgbuv. It constantly consumes 4-6 mA and does not go into sleep mode.

donbobka, 2024-06-03:
All my blinds got battery level somehow.

[image]
[image]

donbobka, 2024-06-17:
@Koenkk 👋  i think we can close this issue

closed 2024-06-17 by Koenkk

DariBer, 2024-08-26:
@donbobka Did you do anything spec, like an external converter or anything else, cos for me it's not working, this is the one I have
[image]

preciserobot, 2024-10-02:
> @donbobka Did you do anything spec, like an external converter or anything else, cos for me it's not working, this is the one I have ![Screenshot 2024-08-26 at 14 55 33](https://private-user-images.githubusercontent.com/26138185/361439065-6fa0d626-c6e9-4fd3-93cc-4271b9b8e711.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc4NzkyNDAsIm5iZiI6MTcyNzg3ODk0MCwicGF0aCI6Ii8yNjEzODE4NS8zNjE0MzkwNjUtNmZhMGQ2MjYtYzZlOS00ZmQzLTkzY2MtNDI3MWI5YjhlNzExLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDAyVDE0MjIyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEyODI2OTNlOGU5YmU0YTViNzQxM2MyZDA0NmY1Mzc0MGY1NTc3NTY5MDJiNWFhNmNhYmVjNGE2YmMzNWQyMTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.jtrqxNH-u3Ux0JB8O6GB59x2tXJa_7TyAsMesy0Ym8c)

I have exactly the same problem with missing battery info. Also sending any messages to the device remain without consequence (no movement or anything).