Koenkk/zigbee2mqtt #18686: 1.32 update not working for me, problem starting zigbee-herdsman

diazovitch, 2023-08-21:
### What happened?

I start by updating node (v12->v16) and npm (v6.14 -> v8.19) with good version.
Next, i try to start npm and i have a problem.

`Zigbee2MQTT:info  2023-08-21 18:26:40: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-08-21.18-26-40' filename: log.txt
Zigbee2MQTT:info  2023-08-21 18:26:41: Starting Zigbee2MQTT version 1.27.2 (commit #7dc48fb6)
Zigbee2MQTT:info  2023-08-21 18:26:41: Starting zigbee-herdsman (0.14.53)
**Zigbee2MQTT:error 2023-08-21 18:26:41: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-08-21 18:26:41: Failed to start zigbee
Zigbee2MQTT:error 2023-08-21 18:26:41: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-08-21 18:26:41: Exiting...
Zigbee2MQTT:error 2023-08-21 18:26:41: Error: Could not locate the bindings file. Tried:**

Before, i update everything was ok.

Could you help me please. I d'ont have a high level.
`

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20201127

### Adapter

CC2531

### Debug log

info  2023-08-21 18:26:40: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2023-08-21.18-26-40' filename: log.txt
info  2023-08-21 18:26:41: Starting Zigbee2MQTT version 1.27.2 (commit #7dc48fb6)
info  2023-08-21 18:26:41: Starting zigbee-herdsman (0.14.53)
error 2023-08-21 18:26:41: Error while starting zigbee-herdsman
error 2023-08-21 18:26:41: Failed to start zigbee
error 2023-08-21 18:26:41: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-08-21 18:26:41: Exiting...
error 2023-08-21 18:26:41: Error: Could not locate the bindings file. Tried:
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/build/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/build/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/out/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/build/default/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/compiled/16.20.2/linux/arm/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/@serialport/bindings/lib/binding/node-v93-linux-arm/bindings.node
    at bindings (/opt/zigbee2mqtt/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (node:internal/modules/cjs/loader:1198:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1252:10)
    at Module.load (node:internal/modules/cjs/loader:1076:32)
    at Function.Module._load (node:internal/modules/cjs/loader:911:12)
    at Module.require (node:internal/modules/cjs/loader:1100:19)
    at require (node:internal/modules/cjs/helpers:119:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/@serialport/bindings/lib/index.js:14:22)
    at Module._compile (node:internal/modules/cjs/loader:1198:14)

Koenkk, 2023-08-22:
Did you execute `npm ci` after updating z2m?

thetrevy, 2023-08-22:
@Koenkk, I'm having the same issue. Could you explain how to run that command? I'm new to HA and still learning.

diazovitch, 2023-08-22:
I tried one more time to upgrade and it works. I think i forget to execute npm ci.
But today my zigbee devices don't work.
I have to reboot system. What's wrong do you think ?

lux73, 2023-08-23:
@diazovitch 

please consider upgrade your CC2531 - they are cheap, weak & unstable for production use -> https://www.zigbee2mqtt.io/guide/adapters/#not-recommended

**in my opinion it's not worth to deal with it**

diazovitch, 2023-08-23:
> @diazovitch
> 
> please consider upgrade your CC2531 - they are cheap, weak & unstable for production use -> https://www.zigbee2mqtt.io/guide/adapters/#not-recommended
> 
> **in my opinion it's not worth to deal with it**

indeed, I have seen. What model do you recommend? Will I have to pair all the devices again?

lux73, 2023-08-25:
i would recommend any CC2652x based Adapter

I personally use CC2652RB from slae.sh - its really stable in my Environment with +50 Devices 😇

closed 2023-10-02 by github-actions[bot]