Koenkk/zigbee2mqtt #18688: Turning on/off not working since device update Awox 33957

jeroenvansch, 2023-08-21:
### What happened?

I've updated my device from version 2.2.10 to 2.7.0 because of stability issues (also reported on the [device page](https://www.zigbee2mqtt.io/devices/33957.html)) The stability issue seems to be fixed but a new problem arose.
When turning the bulb OFF through Home Assistant (as part of a group) of through the Z2M frontend(the bulb directly), the bulb dimms to fully dimmed, but doesn't turn off.
Luckilly I've also Bound the on/off cluster from a remote to this Bulb so I can still turn off the bulb directly.
Since direct control does work correctly I am guessing there is somthing wrong with the converter for the bulb with this firmware.

### What did you expect to happen?

The light to turn off

### How to reproduce it (minimal and precise)

- Update bulb to 2.7.0 using the Awox homecontrol app 
- Turn Bulb on
- Try to turn off bulb

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20221226

### Adapter

zzh!

### Debug log

debug 2023-08-21 21:18:25: Received MQTT message on 'zigbee2mqtt/badkamer/set' with data '{"state":"ON"}'
debug 2023-08-21 21:18:25: Publishing 'set' 'state' to 'badkamer'
info  2023-08-21 21:18:26: MQTT publish: topic 'zigbee2mqtt/CROSSLINK spots badkamer', payload '{"brightness":56,"color_mode":"color_temp","color_temp":370,"color_temp_startup":65535,"last_seen":"2023-08-21T19:13:41.816Z","linkquality":36,"power_on_behavior":null,"state":"ON"}'
info  2023-08-21 21:18:26: MQTT publish: topic 'zigbee2mqtt/badkamer', payload '{"brightness":56,"color_mode":"color_temp","color_temp":370,"state":"ON"}'
debug 2023-08-21 21:18:27: Received Zigbee message from 'CROSSLINK spots badkamer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2023-08-21 21:18:27: MQTT publish: topic 'zigbee2mqtt/badkamer', payload '{"brightness":56,"color_mode":"color_temp","color_temp":370,"state":"ON"}'
info  2023-08-21 21:18:27: MQTT publish: topic 'zigbee2mqtt/CROSSLINK spots badkamer', payload '{"brightness":56,"color_mode":"color_temp","color_temp":370,"color_temp_startup":65535,"last_seen":"2023-08-21T19:18:27.762Z","linkquality":47,"power_on_behavior":null,"state":"ON"}'
debug 2023-08-21 21:18:27: Received Zigbee message from 'CROSSLINK spots badkamer', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":56}' from endpoint 1 with groupID 0
info  2023-08-21 21:18:27: MQTT publish: topic 'zigbee2mqtt/badkamer', payload '{"brightness":56,"color_mode":"color_temp","color_temp":370,"state":"ON"}'
info  2023-08-21 21:18:27: MQTT publish: topic 'zigbee2mqtt/CROSSLINK spots badkamer', payload '{"brightness":56,"color_mode":"color_temp","color_temp":370,"color_temp_startup":65535,"last_seen":"2023-08-21T19:18:27.825Z","linkquality":51,"power_on_behavior":null,"state":"ON"}'

debug 2023-08-21 21:18:28: Received MQTT message on 'zigbee2mqtt/badkamer/set' with data '{"state":"OFF"}'
debug 2023-08-21 21:18:28: Publishing 'set' 'state' to 'badkamer'
info  2023-08-21 21:18:29: MQTT publish: topic 'zigbee2mqtt/CROSSLINK spots badkamer', payload '{"brightness":56,"color_mode":"color_temp","color_temp":370,"color_temp_startup":65535,"last_seen":"2023-08-21T19:18:27.825Z","linkquality":51,"power_on_behavior":null,"state":"OFF"}'
info  2023-08-21 21:18:29: MQTT publish: topic 'zigbee2mqtt/badkamer', payload '{"brightness":56,"color_mode":"color_temp","color_temp":370,"state":"OFF"}'
debug 2023-08-21 21:18:31: Received Zigbee message from 'CROSSLINK spots badkamer', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0
info  2023-08-21 21:18:31: MQTT publish: topic 'zigbee2mqtt/badkamer', payload '{"brightness":1,"color_mode":"color_temp","color_temp":370,"state":"OFF"}'
info  2023-08-21 21:18:31: MQTT publish: topic 'zigbee2mqtt/CROSSLINK spots badkamer', payload '{"brightness":1,"color_mode":"color_temp","color_temp":370,"color_temp_startup":65535,"last_seen":"2023-08-21T19:18:31.888Z","linkquality":43,"power_on_behavior":null,"state":"OFF"}'

closed 2023-09-28 by github-actions[bot]

addellll, 2023-10-08:
Have same light and did recent update with Awox app to firmware is 2.7.1, still same behaviour that it never turn off from Z2M.

Mysa95, 2023-12-06:
Same issue here 👋

dula34, 2024-07-17:
Same problem fw version V2.7.5