Koenkk/zigbee2mqtt #18691: Switches/button functions not showing in HA after migration from ZHA to Zigbee2MQTT (Re-paired)

SpectralXSC, 2023-08-21:
### What happened?

I "migrated" from ZHA to Zigbee2MQTT and I therefore had re-pair my devices. They get paired and configured fine after 2 tries.
But when I try to setup/resetup my automation I do not have the triggers/sensors (Not sure what you call these?) in my automation when selecting the device.

Using:
Aqara Wireless Mini Switch & Aqara Wireless Remote Switch H1 (double rocker).

HOWEVER if I press through the functions of the buttons (Single press, double press, etc.) it now shows those when trying to setup automations.

### What did you expect to happen?

I expected that it would show all "triggers/sensors" after re-pairing the devices but it only shows "Battery Level changes" and "temperature changes"

But it doesn't show up with the switches/buttons functions.

### How to reproduce it (minimal and precise)

Not sure if it's my setup only but:

Set up Aqara switches & lightbulbs in ZHA and enabled some automation with light bulbs and the switches in ZHA

Switch to Zigbee2MQTT and re-pair devices

Try to set up the previous automation.

Then the buttons trigger/sensors should be missing? (Until you press through them after pair but before setting automation

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102(N)

kauthmbt, 2023-08-23:
Just a wild guess but did you try the following already?
https://www.zigbee2mqtt.io/devices/WXKG11LM.html#deprecated-click-event

Sounds like MQTT is not exposing the `action` event that HA expects

SpectralXSC, 2023-08-24:
Nothing shows up - not even action or click, so I found some blueprints and trying to use them atm.

closed 2023-10-01 by github-actions[bot]