Koenkk/zigbee2mqtt #18693: Vimar 03981 shows non-existing attribute

stefa168, 2023-08-22:
### What happened?

I'm not sure from which update this happened, but I noticed that Z2M shows an attribute for Vimar 03981 Relays that the device doesn't have (`startUpOnOff`), and I sometimes get this error:

`ERROR: Publish 'set' 'power_on_behavior' to 'Luci Scale' failed: 'Error: Write 0xf4ce360a9db9369e/10 genOnOff({"startUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

I sent the original suggestion for the converter last year and it didn't have any issue, I checked the commit history on the adapters repository however I couldn't find any change that suggests the introduction of the attribute.

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20220726

### Adapter

Sonoff ‎ZBdongle-P

Koenkk, 2023-08-22:
Thanks, disabled it!

closed 2023-08-22 by Koenkk

stefa168, 2023-08-22:
Thank you, as always!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)