Koenkk/zigbee2mqtt #18694: [New device support]:

claudiocsm88, 2023-08-22:
### Link

https://www.ebay.it/itm/166262760742?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=ybo8jpmfq-m&sssrc=4429486&ssuid=hhhuyy0xtkg&var=&widget_ver=artemis&media=COPY

### Database entry

{"id":38,"type":"EndDevice","ieeeAddr":"0xa4c138e0d0f8b3b4","nwkAddr":62858,"manufId":4417,"manufName":"_TZ3210_j4pdtz9v","powerSource":"Battery","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[6,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3210_j4pdtz9v","powerSource":3,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xdc8e95fffe08b084","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":153093895},"lastSeen":1692707987360,"defaultSendRequestWhen":"immediate"}

### Comments

The following device Is signed like a normal switch TS0001 not like a fingerbot in z2m. Only function working Is on/off, other commands are without entity. I'm on last versione of z2m on 22/08/2023

Koenkk, 2023-08-22:
`_TZ3210_j4pdtz9v` will be supported in the next release (1 September)

closed 2023-08-22 by Koenkk

claudiocsm88, 2023-09-01:
Good evening, today September 1st I still don't see any update for zigbee2mqtt to be able to use the fingerbot. Will the next update be released soon?

SimonBard, 2024-10-10:
Can you give me a link where to buy a fingerbot? Mine is not even interviewed. 
Does it work for you in the meantime?