Koenkk/zigbee2mqtt #18696: Device identified incorrectly

moodyblue, 2023-08-22:
### What happened?

I've paired this device https://www.zigbee2mqtt.io/devices/KMPCIL-tag-001.html but it was identified as https://www.zigbee2mqtt.io/devices/STS-PRS-251.html

### What did you expect to happen?

The device should be identified as https://www.zigbee2mqtt.io/devices/KMPCIL-tag-001.html

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

SMLIGHT SLZB-06

moodyblue, 2023-08-22:
Two months ago the device was identified correctly

Koenkk, 2023-08-22:
Can you provide the data/database.db entry of this device?

moodyblue, 2023-08-22:
Here is my data folder
[data.zip](https://github.com/Koenkk/zigbee2mqtt/files/12413411/data.zip)

Koenkk, 2023-08-24:
Can you also provide the current entry? (when it is not recognised correctly)

moodyblue, 2023-08-24:
This is the incorrect entry
```
{"id":10,"type":"EndDevice","ieeeAddr":"0x00124b000f642c7e","nwkAddr":275,"manufId":0,"manufName":"SmartThings","powerSource":"Battery","modelId":"tagv4","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":12,"inClusterList":[0,1,3,15,32],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"tagv4","manufacturerName":"SmartThings","powerSource":3,"zclVersion":1,"appVersion":4,"stackVersion":73,"hwVersion":3,"dateCode":"130925"}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryVoltage":27}},"genBinaryInput":{"attributes":{"statusFlags":3,"presentValue":3}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b00257c8ed2","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00257c8ed2","endpointID":1},{"cluster":15,"type":"endpoint","deviceIeeeAddress":"0x00124b00257c8ed2","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":15,"attrId":85,"minRepIntval":10,"maxRepIntval":60,"repChange":1}],"meta":{}}},"appVersion":4,"stackVersion":73,"hwVersion":3,"dateCode":"130925","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1692727397793,"defaultSendRequestWhen":"active","checkinInterval":3600}
```

Koenkk, 2023-08-25:
I'm not sure how it how it happened but for some reason the modelId changed, I suggest to force remove the device (through the frontend) and re-pair it.

moodyblue, 2023-09-07:
In the meantime I've used the correct entry to include in the db file and the device is working. Maybe I will have a problem if I ever need to force remove and re-pair so I will keep a backup of the db file.

Thanks

closed 2023-10-16 by github-actions[bot]