Koenkk/zigbee2mqtt #18698: MCCGQ01LM devices lose link to z2m with ZBDongle-E

lcorsini, 2023-08-22:
### What happened?

Installed Zigbee2MQTT 1.32.1 with EZSP v8 Coordinator (ZBDongle-E v2 with firmware 6.10.3.0 build 297)
and tried to pair my devices (previous coordinator was a raspbee) pairing works fine, but after a while my door sensors
Xiaomi MCCGQ01LM stop sending updates (they blink at state change) to z2m
there is no particular log, they just seems to go deep sleep and won't reconnect to coordinator at wake

### What did you expect to happen?

Xiaomi MCCGQ01LM to continue sending events to z2m

### How to reproduce it (minimal and precise)

take a Xiaomi MCCGQ01LM (with fresh battery)
pair it to z2m
wait at least 30 minutes
the device stops sending events

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ZBDongle-E v2 (EZSP)

closed 2023-09-29 by github-actions[bot]

villmatt, 2024-02-02:
Did you ever find a solution to this? I have just stopped using ZHA and moved to Zigbee2MQTT and sort of kicking myself because I have about 8 of these around the house. They all just drop off the network. When I connect them they last maybe an hour and then nothing. New batteries, I have them close and far from coordinators. I literally said out loud, I thought z2m was meant to be more stable, but I feel like I had a really reliable network with zha.

lcorsini, 2024-02-02:
Well it must be said that these Xiaomi sensors don't follow the zigbee
standard completely, so it's not really Z2m fault, also ZBDongle-E support
os still marked as experimental.

First workaround I found is to pair the sensors with a router (like a smart
plug) but in the end i
moved to a more stable dongle, the ZBDongle-P seems more stable has better
support and works fine with these old sensors

villmatt, 2024-02-02:
😭😭😭

Thanks for the quick response. I'll try and figure out how to add them to another router. I have smart plugs around just unsure how to force them to add to them. 

Thanks again

villmatt, 2024-02-03:
@lcorsini 

Thanks so much. Just tested 3 of them. 2 joined to a power plug and one to the sonoff dongle. You were right. The sonoff one disconnects while the others just stay on working. You are a legend. Thanks heaps

lcorsini, 2024-02-03:
In the Z2m web ui, if you have routers paired in your network, the permit
join button allows you to choose which device you want to permit join on.
Choose a router and pair them, at least in my case they stay active in the
network