Koenkk/zigbee2mqtt #18699: Ikea Trådfri and other models ignoring color_temp and color_temp_startup

h8hr1pzdo3rmiqv5, 2023-08-23:
### What happened?

setup: Home Assistant last stable version and zigbee2mqtt last-dev version (both from official Docker images)

My lights CCT is controlled by an automation that change the temperature depending on time/sun.
In order to avoid having briefly cool light during the night when I turn on lights that were cool last time they were on, I sent a payload with the color_temp to lights that are off. This work great with [Lonsonho ZB-RGBCW](https://www.zigbee2mqtt.io/devices/ZB-RGBCW.html) with eWeLight ZB-CL01 controller: when turned on, the light has the temperature set when it was off.

This doesn't work with Lonsonho ZB-RGBCW (the same) with eWeLink Z102LG03-1 controller.
Neither with Ikea [LED1949C5](https://www.zigbee2mqtt.io/devices/LED1949C5.html#ikea-led1949c5) and [LED2003G10](https://www.zigbee2mqtt.io/devices/LED2003G10.html#ikea-led2003g10).
Haven't tested yet but [LED2005R5](https://www.zigbee2mqtt.io/devices/LED2005R5.html#ikea-led2005r5) may have the same issue.

Observed behavior for the abovesaid models:
publishing the below message when the light is off seem to effectively change the value at the light (it can be retrieved by the command get);

topic: zigbee2mqtt/bulb_e14_470lm/set
payload: '{"color_temp":250}'
qos: 1

but once turned on, the light emit a light of the previous temperature. Worse, if still on and requested to set the wanted temperature again (exactly the same), it will not change its temperature. It will only change it if receiving a request for a different temperature.

### What did you expect to happen?

The light should emit a white temperature equal to the value set while it was turned off and saved in its data.

### How to reproduce it (minimal and precise)

Use a bulb of the involved model, with CCT capability.
Turn the light on.
Set a warm temperature, for example.
Turn the light off.
Set a cool temperature, for example. It can be done with zigbee2MQTT webinterface, by clicking on the temperature bar.
Turn the light on.
Observe the light doesn't emit the set temperature.

### Zigbee2MQTT version

1.32.2-dev

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

Sonoff ZBDongle-E EZSP v8

### Debug log

debug 2023-08-23 02:03:23: Received MQTT message on 'zigbee2mqtt/bulb_e27_1/set' with data '{"color_temp":166}'
debug 2023-08-23 02:03:23: Publishing 'set' 'color_temp' to 'bulb_e27_1'
info  2023-08-23 02:03:23: MQTT publish: topic 'zigbee2mqtt/bulb_e27_1', payload '{"brightness":177,"color":{"x":0.3217,"y":0.3314},"color_mode":"color_temp","color_temp":166,"level_config":{"on_level":"previous"},"linkquality":160,"state":"OFF"}'
debug 2023-08-23 02:03:28: Received MQTT message on 'zigbee2mqtt/bulb_e27_1/set' with data '{"state":"on"}'
debug 2023-08-23 02:03:28: Publishing 'set' 'state' to 'bulb_e27_1'
info  2023-08-23 02:03:28: MQTT publish: topic 'zigbee2mqtt/bulb_e27_1', payload '{"brightness":177,"color":{"x":0.3217,"y":0.3314},"color_mode":"color_temp","color_temp":166,"level_config":{"on_level":"previous"},"linkquality":160,"state":"ON"}'

h8hr1pzdo3rmiqv5, 2023-08-23:
Just tried.

With the Ikea Trådfri: it work, even if zigbee2mqtt complain "there is no converter" for those parameters.
However, those settings don't survive a loss of power. It work again once the power has been reestablished and the settings pushed again. Is that instability the reason those parameters aren't documented nor supported?

With the Lonsonho eWeLink: doesn't work. I suppose they do not exist on the firmware part.

Thank you @sjorge.

sjorge, 2023-08-24:
Mostly because they are optional and most bulbs implement them weird. Tradfri tends to forget them on powerloss. For some (i though innr) you can write but not read them, …~ sjorge

protyposis, 2023-09-14:
> mosquitto_pub -t zigbee2mqtt/FRIENDLY_NAME/set/level_options -m '{"execute_if_off": true}'

Didn't validate myself, but the sources indicate this is called `level_config`.

h8hr1pzdo3rmiqv5, 2023-09-14:
The Ikea Trådfris however don't need additional configuration for brightness change while off; they handle it by default.

closed 2023-10-22 by github-actions[bot]