Koenkk/zigbee2mqtt #18702: Lonsonho QS-Zigbee-D02-TRIAC-L recognized as TuYa TS110E_1gang_2

mrmaximas, 2023-08-23:
### What happened?

I've tried several times to add Lonsonho QS-Zigbee-D02-TRIAC-L 
https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-L.html#lonsonho-qs-zigbee-d02-triac-l 
but it's recognized as TuYa TS110E_1gang_2 
https://www.zigbee2mqtt.io/devices/TS110E_1gang_2.html#tuya-ts110e_1gang_2
Why?

### Zigbee2MQTT version

1.32.1 commit: 69638d79

### Adapter firmware version

20221226

### Adapter

zStack3x0

c0mplex1, 2023-08-24:
I think I have the same issue with my dimmer, but so far I have not received a response to my problem report from @Koenkk
https://github.com/Koenkk/zigbee2mqtt/issues/18480#issue-1830841550

mrmaximas, 2023-09-01:
> I think I have the same issue with my dimmer, but so far I have not received a response to my problem report from @Koenkk
> [#18480 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18480#issue-1830841550)

you closed your ticket, did you find a solution?

Koenkk, 2023-09-02:
Fixed in z2 1.33.0

closed 2023-09-02 by Koenkk

mrmaximas, 2023-09-04:
🙏🏻

mrmaximas, 2023-11-28:
it's not recognized correctly. it's recognized like https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-L_1.html#lonsonho-qs-zigbee-d02-triac-l_1 but this is not the same.
json from db here. my model is   
[image]

{"id":49,"type":"Router","ieeeAddr":"0xa4c138b5bb325444","nwkAddr":38256,"manufId":4417,"manufName":"_TZ3210_weaqkhab","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[4,5,6,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS110E","manufacturerName":"_TZ3210_weaqkhab","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":257,"inClusterList":[4,5,6,8,768,61184],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onTime":0,"onOff":1}},"genLevelCtrl":{"attributes":{"61440":1000,"64515":10}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":

Koenkk, 2023-11-28:
The `_1` postfix is added because there is a conflicting definition.

mrmaximas, 2023-12-01:
Have problem with two exposes.
When I try to set Light type or Power-on behavior  I get errors. Are they generally readable and writable?

2023-12-01 11:55:36 Publish 'set' 'light_type' to 'Kitchen Main' failed: 'Error: Write 0xa4c138b5bb325444/1 genLevelCtrl({"64514":{"value":0,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Error 2023-12-01 11:55:37 Publish 'set' 'power_on_behavior' to 'Kitchen Main' failed: 'Error: Write 0xa4c138b5bb325444/1 genOnOff({"moesStartUpOnOff":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Koenkk, 2023-12-01:
`UNSUPPORTED_ATTRIBUTE` means the device doesn't support this.

Koenkk, 2023-12-01:
@mrmaximas some versions of this device support it some don't, so I would leave it as is.

mrmaximas, 2023-12-01:
ok. undertsand.
please pay attention to https://github.com/Koenkk/zigbee2mqtt/issues/19901
thanks.

mrmaximas, 2023-12-07:
@Koenkk please pay attention. in this video https://youtu.be/ubml-b0YooI?t=555 Light type and Power-on behavior works in Tuya App, which suggests that there are bugs in converter.

Koenkk, 2023-12-07:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/b402cb48a8f88915c92b1d1fa8252db7): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed