Koenkk/zigbee2mqtt #18704: [New device support]: Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL- _TZE200_v1jqz5cy

cloudbr34k84, 2023-08-23:
### Link

https://www.aliexpress.com/item/1005005575336871.html?spm=a2g0o.order_list.order_list_main.11.28e31802Wxn6Wl

### Database entry

{"id":115,"type":"EndDevice","ieeeAddr":"0xe0798dfffeb87446","nwkAddr":42479,"manufId":4098,"manufName":"_TZE200_v1jqz5cy","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u00195y,f\u00195y,\u0012\u001a5y,\u0012\u001e5y,\u0011\"5y,\u0011","65506":31,"65508":0,"appVersion":65,"modelId":"TS0601","manufacturerName":"_TZE200_v1jqz5cy","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1692789799130,"defaultSendRequestWhen":"immediate"}

### Comments

Would someone be willing to make a converter> here are the Data Points

1: TDS
2: temperature
101: ORP
102: CL
7: Battery Percentage
105: backlight Value
10: ph Value
11: EC Value
108: EC Max Value
109: EC Min Value
110: ORP max
111: ORP min
112: CL Max Value
113: CL Min Value
114: PH??
115: EC??
116: ORP??
117: Salt

cloudbr34k84, 2023-08-24:
im giving it a crack

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'BLE-YL01',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: 'TZE200_v1jqz5cy',
        },
    ],
    model: 'WiFi Chlorine Meter BLE-YL01',
    vendor: 'yieryi',
    description: 'Smart WiFi Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Swimming Pool Water Quality Analyzer USB Solar Powered',
    fromZigbee: [tuya.fz.datapoints, tuya.fz.gateway_connection_status],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.tds(),
        e.temperature()
        e.orp()
        e.cl()
        e.batterypercentage()
        e.backlightvalue()
        e.pHvalue()
        e.ecvalue()
        e.ecmaxvalue()
        e.ecminvalue()
        e.orpmaxvalue()
        e.orpminvalue()
        e.clmaxvalue()
        e.clminvalue()
        e.salt()
        // Here you should put all functionality that your device exposes
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'tds', tuya.valueConverter.raw],
            [2, 'temperature', tuya.valueConverter.raw.divideBy10],
            [101, 'orp', tuya.valueConverter.raw],
            [102, 'cl', tuya.valueConverter.raw],
            [7, 'batterypercentage', tuya.valueConverter.raw],
            [105, 'backlightvalue', tuya.valueConverter.raw],
            [10, 'pHvalue', tuya.valueConverter.raw],
            [11, 'ecvalue', tuya.valueConverter.raw],
            [108, 'ecmaxvalue', tuya.valueConverter.raw],
            [109, 'ecminvalue', tuya.valueConverter.raw],
            [110, 'orpmaxvalue', tuya.valueConverter.raw],
            [111, 'orpminvalue', tuya.valueConverter.raw],
            [112, 'clmaxvalue', tuya.valueConverter.raw],
            [113, 'clminvalue', tuya.valueConverter.raw],
            [117, 'salt', tuya.valueConverter.raw],

        ],
    },
};

module.exports = definition;
```

Gor my first error
```
[09:44:12] INFO: Preparing to start...
[09:44:13] INFO: Socat not enabled
[09:44:13] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:30
        e.tds(),
          ^
TypeError: e.tds is not a function
    at /app/data/extension/externally-loaded.js:30:11
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:152:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:159:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:169:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

cloudbr34k84, 2023-08-24:
@Koenkk where is a list of the supported datapoints?

Koenkk, 2023-08-25:
Here is a list of all [standard exposes](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a25c4b71d032a2a29dbb4e59dba52313f95f05b2/src/lib/exposes.ts#L603), but you can easily create new ones: [example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a25c4b71d032a2a29dbb4e59dba52313f95f05b2/src/devices/tuya.ts#L1402)

cloudbr34k84, 2023-08-25:
Thanks @Koenkk , i have removed all the others and just starting with Temperature. I have looked at he other examples and when I tried and connect nothing is happening, not even the logs are giving much. Any pointers? i don't code for a living, so trying to understand is not easy for me.
```
info  2023-08-26 09:13:44: Device '0xe0798dfffeb87446' joined
info  2023-08-26 09:13:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xe0798dfffeb87446","ieee_address":"0xe0798dfffeb87446"},"type":"device_joined"}'
info  2023-08-26 09:13:44: Starting interview of '0xe0798dfffeb87446'
info  2023-08-26 09:13:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xe0798dfffeb87446","ieee_address":"0xe0798dfffeb87446","status":"started"},"type":"device_interview"}'
info  2023-08-26 09:13:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xe0798dfffeb87446","ieee_address":"0xe0798dfffeb87446"},"type":"device_announce"}'
info  2023-08-26 09:13:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xe0798dfffeb87446","ieee_address":"0xe0798dfffeb87446"},"type":"device_announce"}'
info  2023-08-26 09:13:46: MQTT publish: topic 'zigbee2mqtt/Kitchen Dishwasher EM Plug', payload '{"child_lock":"UNLOCK","current":0,"energy":156.4,"indicator_mode":"off/on","linkquality":0,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":249}'
info  2023-08-26 09:13:46: MQTT publish: topic 'zigbee2mqtt/Kitchen Dishwasher EM Plug', payload '{"child_lock":"UNLOCK","current":0,"energy":156.4,"indicator_mode":"off/on","linkquality":7,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":249}'
info  2023-08-26 09:13:47: MQTT publish: topic 'zigbee2mqtt/Kitchen Sink', payload '{"battery":0,"battery_low":null,"linkquality":29,"tamper":null,"temperature":22.14,"water_leak":null}'
info  2023-08-26 09:13:48: MQTT publish: topic 'zigbee2mqtt/Front Patio EM GPO', payload '{"current_left":0,"energy":0,"linkquality":58,"power_left":0,"state_left":"ON","state_right":"ON","voltage_left":249}'
info  2023-08-26 09:13:51: MQTT publish: topic 'zigbee2mqtt/Laundry Washing Machine EM Plug', payload '{"child_lock":"UNLOCK","current":0.11,"energy":57.71,"indicator_mode":"off/on","linkquality":21,"power":11,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":248}'
info  2023-08-26 09:13:51: MQTT publish: topic 'zigbee2mqtt/Office EM Plug Lauren', payload '{"child_lock":"UNLOCK","current":0,"energy":6.38,"indicator_mode":"off/on","linkquality":61,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":249}'
info  2023-08-26 09:13:53: MQTT publish: topic 'zigbee2mqtt/Front Patio EM GPO', payload '{"current_left":0,"energy":0,"linkquality":58,"power_left":0,"state_left":"ON","state_right":"ON","voltage_left":249}'
info  2023-08-26 09:13:55: MQTT publish: topic 'zigbee2mqtt/Laundry Washing Machine EM Plug', payload '{"child_lock":"UNLOCK","current":0.07,"energy":57.71,"indicator_mode":"off/on","linkquality":25,"power":11,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":248}'
info  2023-08-26 09:13:56: MQTT publish: topic 'zigbee2mqtt/Laundry Washing Machine EM Plug', payload '{"child_lock":"UNLOCK","current":0.07,"energy":57.71,"indicator_mode":"off/on","linkquality":25,"power":33,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":248}'
info  2023-08-26 09:13:56: MQTT publish: topic 'zigbee2mqtt/Kitchen Galley Motion', payload '{"battery":100,"illuminance":922,"keep_time":"10","linkquality":94,"occupancy":false,"sensitivity":null}'
info  2023-08-26 09:13:57: MQTT publish: topic 'zigbee2mqtt/Kitchen Galley Motion', payload '{"battery":100,"illuminance":922,"keep_time":"10","linkquality":91,"occupancy":true,"sensitivity":null}'
info  2023-08-26 09:13:57: MQTT publish: topic 'zigbee2mqtt/Garage Door', payload '{"garage_door_contact":true,"linkquality":32,"trigger":false}'
info  2023-08-26 09:13:58: MQTT publish: topic 'zigbee2mqtt/Front Patio EM GPO', payload '{"current_left":0,"energy":0,"linkquality":61,"power_left":0,"state_left":"ON","state_right":"ON","voltage_left":249}'
```
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
[... cut ...]
            // [11, 'ecvalue', tuya.valueConverter.raw],
            // [108, 'ecmaxvalue', tuya.valueConverter.raw],
            // [109, 'ecminvalue', tuya.valueConverter.raw],
            // [110, 'orpmaxvalue', tuya.valueConverter.raw],
            // [111, 'orpminvalue', tuya.valueConverter.raw],
            // [112, 'clmaxvalue', tuya.valueConverter.raw],
            // [113, 'clminvalue', tuya.valueConverter.raw],
            // [117, 'salt', tuya.valueConverter.raw],

        ],
    },
};

module.exports = definition;
```

mochitoju, 2023-09-01:
Is the devixe already available in z2m or hast anybody a solution?

cloudbr34k84, 2023-09-06:
> Here is a list of all [standard exposes](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a25c4b71d032a2a29dbb4e59dba52313f95f05b2/src/lib/exposes.ts#L603), but you can easily create new [...]

thanks but I don't really know what im looking for

tschiex, 2023-09-06:
I just got the very same device here, connected to my Home Assistant server, through ZHA.
I would be happy to help here. I can write code, but am a ZigBee/Z2MQTT/ZHA newbie...

@cloudbr34k84 Brad, how did you get the Data Points?

cloudbr34k84, 2023-09-06:
i follow the the instructions on Z2M
https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

cloudbr34k84, 2023-09-09:
@Koenkk any chance you can help create the converter pls?

mochitoju, 2023-09-09:
Yes, that would bei great!

Freev4lley, 2023-09-09:
That would be great for me too!!!

Koenkk, 2023-09-09:
With the following ext converter, what shows up in the z2m frontend -> device -> state page? 

https://gist.github.com/Koenkk/7262a8a36bb84327ff563e0b8256024b

tschiex, 2023-09-10:
Sorry Koenkk, I don't have Z2M installed on HA, but ZHA. Is it easy to switch back and forth?

BTW, I bought a Tuya Gateway and followed the instructions on Z2M
https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

and found more DP Ids:

1: TDS
2:  Temperature
101: ORP
102: Chlorine
7: Battery Percentage
103: pH Calibration
104: Backlight
105: Backlight Value
10: pH Value
11: EC Value
106: pH Max Value
107: pH Min Value
108: EC Max Value
109: EC Min Value
110: ORP Max Value
111: ORP Min Value
112: Chlorine Max Value
113: Chlorine Min Value
114: PH Calibration
115: EC Calibration
116: ORP Calibration
117: Salt

(Translation of Chinese by Google Translate camera from Tuya IoT platform website).

There are apparently 2 pH calibration DP Ids (and there are 2 pH calibration standards: Asia uses 6.86/4.00/9.18, EU uses 7/4/10). The two pH calibration DPs could be these 2 maybe?

Apparently, I was told the device works on deconz: https://forum.phoscon.de/t/tuya-tze200-v1jqz5cy-zigbee-chlorine-meter/3930 (see https://github.com/zigpy/zha-device-handlers/issues/2565).

and here: dresden-elektronik/deconz-rest-plugin#7215

cloudbr34k84, 2023-09-10:
> With the following ext converter, what shows up in the z2m frontend -> device -> state page?
> 
> https://gist.github.com/Koenkk/7262a8a36bb84327ff563e0b8256024b

Nothing shows except Linkquality
[image]

tschiex, 2023-09-11:
Brad, 

I'm not a Z2M specialist, but the fact that Z2M says:

> warn  2023-09-11 09:30:48: Device '0xdc8e95fffe832b2f' with Zigbee model 'TS0601' and manufacturer name '_TZE200_v1jqz5cy' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

may be a hint that Koenkk's handler above is not accessible to Zigbee2MQTT in your setup.

tschiex, 2023-09-11:
Or may be the handler is wrong.... Looking into it, I see:

```
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'BLE-YL01',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: 'TZE200_v1jqz5cy',
        },
    ],
```
I assume the modelId is important for recognizing the device? 

Because it reports 'TS0601' not 'BLE-YL01'.

Brad, I suppose you should edit Koenkk's script and replace the 

```
            modelID: 'BLE-YL01',
```

by 
```
            modelID: 'TS0601',
```

BabaIsYou, 2023-09-11:
> I was told the device apparently works on deconz: https://forum.phoscon.de/t/tuya-tze200-v1jqz5cy-zigbee-chlorine-meter/3930 (see [zigpy/zha-device-handlers#2565](https://github.com/zigpy/zha-device-handlers/issues/2565)).

It's a wok in progress ... ;-)

Koenkk, 2023-09-11:
Made a mistake, updated https://gist.github.com/Koenkk/7262a8a36bb84327ff563e0b8256024b

cloudbr34k84, 2023-09-11:
> Made a mistake, updated https://gist.github.com/Koenkk/7262a8a36bb84327ff563e0b8256024b

Hey @Koenkk Thanks, its now added correctly, but it has not exposed any entities
```
info  2023-09-12 07:44:57: Successfully interviewed '0xdc8e95fffe832b2f', device has successfully been paired
info  2023-09-12 07:44:57: Device '0xdc8e95fffe832b2f' is supported, identified as: yieryi Smart WiFi Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Swimming Pool Water Quality Analyzer USB Solar Powered (WiFi Chlorine Meter BLE-YL01)
info  2023-09-12 07:44:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart WiFi Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Swimming Pool Water Quality Analyzer USB Solar Powered","exposes":[{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"WiFi Chlorine Meter BLE-YL01","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"}],"supports_ota":false,"vendor":"yieryi"},"friendly_name":"0xdc8e95fffe832b2f","ieee_address":"0xdc8e95fffe832b2f","status":"successful","supported":true},"type":"device_interview"}'
info  2023-09-12 07:44:57: Configuring '0xdc8e95fffe832b2f'
info  2023-09-12 07:44:57: Successfully configured '0xdc8e95fffe832b2f'
```

alray31, 2023-09-12:
> Hey @Koenkk Thanks, its now added correctly, but it has not exposed any entities
>
> ```
> info  2023-09-12 07:44:57: Successfully interviewed '0xdc8e95fffe832b2f', device has successfully been paired [...]

I'm not an expert but if you look at the other tuya converters that are using datapoints in zigbee-herdsman-convert/src/devices/tuya.ts, you also need to add the corresponding entries in the expose section, i.e.:  exposes: [e.tds, e.temperature, etc...

[image]

Koenkk, 2023-09-13:
Please make a screenshot of the z2m frontend -> device page -> state page?

cloudbr34k84, 2023-09-13:
> Please make a screenshot of the z2m frontend -> device page -> state page?

@Koenkk 
[image]

Koenkk, 2023-09-14:
With the following converter: https://gist.github.com/Koenkk/7262a8a36bb84327ff563e0b8256024b , can you provide the debug log when pairing the device + 5 minutes after this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

cloudbr34k84, 2023-09-14:
> With the following converter: https://gist.github.com/Koenkk/7262a8a36bb84327ff563e0b8256024b , can you provide the debug log when pairing the device + 5 minutes after this?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Hey @Koenkk thanks for helping on this
When I copy the new code and enable debugging and restart I get an error from the converter

```
[06:33:10] INFO: Preparing to start...
[06:33:10] INFO: Socat not enabled
[06:33:10] INFO: Zigbee Herdsman debug logging enabled
[06:33:10] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:27
    onEvent: tuya.onEvent(), // Add this if you are getting no converter for 'commandMcuSyncTime'
                  ^
TypeError: tuya.onEvent is not a function
    at /app/data/extension/externally-loaded.js:27:19
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:152:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:159:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:169:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

Koenkk, 2023-09-16:
I forgot to mention that this converter only works with the latest-dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

cloudbr34k84, 2023-09-16:
> I forgot to mention that this converter only works with the latest-dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Ok so I have updated my Z2M Edge version
I have deleted and re-added the device
there are no entities in the State Tab except "linkquality": 87
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12640259/log.txt)

cloudbr34k84, 2023-09-16:
> > > Made a mistake, updated https://gist.github.com/Koenkk/7262a8a36bb84327ff563e0b8256024b
> > 
> > 
> > Hey @Koenkk Thanks, its now added correctly, but it has not exposed any entities
> > ```
> > info  2023-09-12 07:44:57: Successfully interviewed '0xdc8e95fffe832b2f', device has successfully been paired
> > info  2023-09-12 07:44:57: Device '0xdc8e95fffe832b2f' is supported, identified as: yieryi Smart WiFi Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Swimming Pool Water Quality Analyzer USB Solar Powered (WiFi Chlorine Meter BLE-YL01)
> > info  2023-09-12 07:44:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart WiFi Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Swimming Pool Water Quality Analyzer USB Solar Powered","exposes":[{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"WiFi Chlorine Meter BLE-YL01","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"}],"supports_ota":false,"vendor":"yieryi"},"friendly_name":"0xdc8e95fffe832b2f","ieee_address":"0xdc8e95fffe832b2f","status":"successful","supported":true},"type":"device_interview"}'
> > info  2023-09-12 07:44:57: Configuring '0xdc8e95fffe832b2f'
> > info  2023-09-12 07:44:57: Successfully configured '0xdc8e95fffe832b2f'
> > ```
> 
> I'm not an expert but if you look at the other tuya converters that are using datapoints in zigbee-herdsman-convert/src/devices/tuya.ts, you also need to add the corresponding entries in the expose section, i.e.: exposes: [e.tds, e.temperature, etc...
> 
> example: [image]
> 
> So maybe try this? :
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     // Since a lot of TuYa devices use the same modelID, but use different datapoints
>     // it's necessary to provide a fingerprint instead of a zigbeeModel
>     fingerprint: [
>         {
>             // The model ID from: Device with modelID 'TS0601' is not supported
>             // You may need to add \u0000 at the end of the name in some cases
>             modelID: 'TS0601',
>             // The manufacturer name from: Device with modelID 'TS0601' is not supported.
>             manufacturerName: '_TZE200_v1jqz5cy',
>         },
>     ],
>     model: 'WiFi Chlorine Meter BLE-YL01',
>     vendor: 'yieryi',
>     description: 'Smart WiFi Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Swimming Pool Water Quality Analyzer USB Solar Powered',
>     fromZigbee: [tuya.fz.datapoints, tuya.fz.gateway_connection_status],
>     toZigbee: [tuya.tz.datapoints],
>     onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
>     configure: tuya.configureMagicPacket,
>     exposes: [e.tds(), e.temperature(), e.orp(), e.cl(), e.batterypercentage(), e.backlightvalue(), e.pHvalue(), e.ecvalue(), e.ecmaxvalue(), e.orpmaxvalue(), e.orpminvalue(), e.clmaxvalue(), e.clminvalue(), e.salt()],
>     meta: {
>         // All datapoints go in here
>         tuyaDatapoints: [
>             [1, 'tds', tuya.valueConverter.raw],
>             [2, 'temperature', tuya.valueConverter.raw.divideBy10],
>             [101, 'orp', tuya.valueConverter.raw],
>             [102, 'cl', tuya.valueConverter.raw],
>             [7, 'batterypercentage', tuya.valueConverter.raw],
>             [105, 'backlightvalue', tuya.valueConverter.raw],
>             [10, 'pHvalue', tuya.valueConverter.raw],
>             [11, 'ecvalue', tuya.valueConverter.raw],
>             [108, 'ecmaxvalue', tuya.valueConverter.raw],
>             [109, 'ecminvalue', tuya.valueConverter.raw],
>             [110, 'orpmaxvalue', tuya.valueConverter.raw],
>             [111, 'orpminvalue', tuya.valueConverter.raw],
>             [112, 'clmaxvalue', tuya.valueConverter.raw],
>             [113, 'clminvalue', tuya.valueConverter.raw],
>             [117, 'salt', tuya.valueConverter.raw],
> 
>         ],
>     },
> };
[... 1 lines cut ...]
> module.exports = definition;
> ```

tried this, same issue as when I attempted. i don't understand how to create a new function

``
[08:30:22] INFO: Preparing to start...
[08:30:22] INFO: Socat not enabled
[08:30:23] INFO: Zigbee Herdsman debug logging enabled
[08:30:23] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:29
    exposes: [e.tds(), e.temperature(), e.orp(), e.cl(), e.batterypercentage(), e.backlightvalue(), e.pHvalue(), e.ecvalue(), e.ecmaxvalue(), e.orpmaxvalue(), e.orpminvalue(), e.clmaxvalue(), e.clminvalue(), e.salt()],
                ^
[... cut ...]
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:152:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:159:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:169:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
``

cloudbr34k84, 2023-09-16:
@Koenkk 
not sure if this is helping but I added the exposes entities
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_v1jqz5cy',
        },
    ],
    model: 'WiFi Chlorine Meter BLE-YL01',
    vendor: 'yieryi',
    description: 'Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Solar Powered',
    fromZigbee: [tuya.fz.datapoints, tuya.fz.gateway_connection_status],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent(), // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.numeric('tdsvalue', ea.STATE).withUnit('ppm').withDescription('Total Dissolved Solids Value'),
        e.temperature(),
        e.numeric('orpvalue', ea.STATE).withUnit('mV').withDescription('Oxidation Reduction Potential Value'),
        e.numeric('clvalue', ea.STATE).withUnit('ppm').withDescription('Free Chlorine Value'),
        e.numeric('batterypercentage', ea.STATE).withUnit('%').withDescription('Remaining battery in %, can take up to 24 hours before reported.').withValueMin(0).withValueMax(100),
        // e.backlightvalue(),
        e.numeric('pHvalue', ea.STATE).withUnit('pH').withDescription('pH Value'),
        e.numeric('ecvalue', ea.STATE).withUnit('µS/cm').withDescription('Electrical Conductivity Value'),
        e.numeric('ecmaxvalue', ea.STATE).withUnit('µS/cm').withDescription('Electrical Conductivity Max Value'),
        e.numeric('ecminvalue', ea.STATE).withUnit('µS/cm').withDescription('Electrical Conductivity Min Value'),
        e.numeric('orpmaxvalue', ea.STATE).withUnit('mV').withDescription('Oxidation Reduction Potential Max Value'),
        e.numeric('orpminvalue', ea.STATE).withUnit('mV').withDescription('Oxidation Reduction Potential Min Value'),
        e.numeric('CLmaxvalue', ea.STATE).withUnit('ppm').withDescription('Free Chlorine Max Value'),
        e.numeric('clmaxvalue', ea.STATE).withUnit('ppm').withDescription('Free Chlorine Min Value'),
        e.numeric('saltvalue', ea.STATE).withUnit('µS/cm').withDescription('Salt Value'),
        e.numeric('phmaxvalue', ea.STATE).withUnit('pH').withDescription('pH Max Value'),
        e.numeric('phminvalue', ea.STATE).withUnit('pH').withDescription('pH Min Value')
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'tdsvalue', tuya.valueConverter.raw],
            [2, 'temperature', tuya.valueConverter.raw.divideBy10],
            [7, 'batterypercentage', tuya.valueConverter.raw],
            [10, 'pHvalue', tuya.valueConverter.raw],
            [11, 'ecvalue', tuya.valueConverter.raw],
            [101, 'orpvalue', tuya.valueConverter.raw],
            [102, 'clvalue', tuya.valueConverter.raw],
            [105, 'backlightvalue', tuya.valueConverter.raw],
            [106, 'phmaxvalue', tuya.valueConverter.raw],
            [107, 'phminvalue', tuya.valueConverter.raw],
            [108, 'ecmaxvalue', tuya.valueConverter.raw],
            [109, 'ecminvalue', tuya.valueConverter.raw],
            [110, 'orpmaxvalue', tuya.valueConverter.raw],
            [111, 'orpminvalue', tuya.valueConverter.raw],
            [112, 'clmaxvalue', tuya.valueConverter.raw],
            [113, 'clminvalue', tuya.valueConverter.raw],
            [117, 'saltvalue', tuya.valueConverter.raw],

        ],
    },
};

module.exports = definition;

```

Koenkk, 2023-09-17:
> not sure if this is helping but I added the exposes entities

It would expect it still does not provide any data?

Can you provide me the debug log from starting z2m, pairing the device and waiting 5 minutes after that?

cloudbr34k84, 2023-09-17:
im not 100% sure how i get the logs tbh
I have it enabled and on the tabs inside the Z2M add on, I can see all the logs but I just don't understand how to get the a logfile, however, im not running a docker version of Z2M..

Koenkk, 2023-09-17:
If you go to the Z2M page logs, keep the page open and pair the device + wait for 5 mins, you can copy the logs from there.

cloudbr34k84, 2023-09-17:
just to confirm we are talking about the same page..
[image]

cloudbr34k84, 2023-09-17:
@Koenkk Or this page
[image]

cloudbr34k84, 2023-09-18:
can anyone else help me out here where to find these logs??

Koenkk, 2023-09-18:
yes thats the correct page

cloudbr34k84, 2023-09-18:
> yes thats the correct page

So what I just copy what's on that page???

cloudbr34k84, 2023-09-18:
@Koenkk i really dont understand how to get the logs. Here is a video of the page after I connected the device?? how does this help?
can you please help me understand how you get the logs especially when they disappear constantly??
[screencast-bpconcjcammlapcogcnnelfmaeghhagj-2023.09.19-08_21_38.webm](https://github.com/Koenkk/zigbee2mqtt/assets/58960644/1f54eed0-7a5e-48fd-958b-545578564a09)

mageali, 2023-09-19:
@cloudbr34k84 Not an expert either but very intrested in the device. First tip with logs is filter them for "TS0601" by adding that in "filter by text" field

Koenkk, 2023-09-19:
@cloudbr34k84 ah sorry, I wasn't aware that the logs immediately disappear. With [this](https://community.home-assistant.io/t/home-assistant-addon-filebrowser/282108) addon you should be able to browse the files. From the z2m directory -> data -> logs folder grab the latest logs.

cloudbr34k84, 2023-09-19:
How is this any different to grabbing the logs via VSCode?

cloudbr34k84, 2023-09-19:
> @cloudbr34k84 ah sorry, I wasn't aware that the logs immediately disappear. With [this](https://community.home-assistant.io/t/home-assistant-addon-filebrowser/282108) addon you should be able to browse the files. From the z2m directory -> data -> logs folder grab the latest logs.

@Koenkk is this what you are after??
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12666958/log.txt)

cloudbr34k84, 2023-09-19:
@Koenkk 

i did a quick copy and paste when I paired
```
2023-09-20 08:25:23MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xdc8e95fffe832b2f","ieee_address":"0xdc8e95fffe832b2f"},"type":"device_announce"}'
Info 2023-09-20 08:25:24Successfully interviewed '0xdc8e95fffe832b2f', device has successfully been paired
Info 2023-09-20 08:25:24Device '0xdc8e95fffe832b2f' is supported, identified as: yieryi Smart WiFi Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Swimming Pool Water Quality Analyzer USB Solar Powered (WiFi Chlorine Meter BLE-YL01)
Info 2023-09-20 08:25:25MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart WiFi Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Swimming Pool Water Quality Analyzer USB Solar Powered","exposes":[{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"WiFi Chlorine Meter BLE-YL01","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","label":"Temperature precision","name":"temperature_precision","property":"temperature_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the temperature value (absolute offset), takes into effect on next report of device.","label":"Temperature calibration","name":"temperature_calibration","property":"temperature_calibration","type":"numeric"}],"supports_ota":false,"vendor":"yieryi"},"friendly_name":"0xdc8e95fffe832b2f","ieee_address":"0xdc8e95fffe832b2f","status":"successful","supported":true},"type":"device_interview"}'
```

Koenkk, 2023-09-20:
@cloudbr34k84 yes that is the log I need, but make sure to set the log level to debug.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

cloudbr34k84, 2023-09-20:
> @cloudbr34k84 yes that is the log I need, but make sure to set the log level to debug.
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

@Koenkk I have had them enabled for a long time. See my Z2M config. 
What am I doing wrong here? 
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: true
mqtt:
  base_topic: zigbee2mqtt
  user: mqtt-user
  password: xxxxxxxxxxxxxxxxx
  server: mqtt://core-mosquitto:1883
serial:
  port: tcp://192.168.68.97:6638
zigbee_herdsman_debug: true
availability: true
advanced:
  log_directory: /config/data/log/%TIMESTAMP%
  timestamp_format: DD-MM-YYYY HH:mm:ss
  log_file: log.txt
  log_level: debug
  pan_id: GENERATE
  log_output:
    - console
    - file
  ext_pan_id:
    - 1
    - 2
    - 3
    - 4
    - 5
    - 6
    - 7
    - 8
  network_key: GENERATE
  channel: 24
homeassistant: true
discovery_topic: homeassistant
status_topic: hass/status
```

cloudbr34k84, 2023-09-20:
@Koenkk so for some reason in my add on config, all the logging was set, but when I check the config file in VS code, it doesn't have the same logging permissions.. sigh

Is this now what you need?
[Uploading log.txt…]()

Koenkk, 2023-09-21:
That exactly the logging I need, but the device doesn't report any datapoints. I guess the TuYa gateway does some magic setup in order to get the device start reporting. To debug this further, I need to have a sniff when pairing this device with the TuYa gateway: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

cloudbr34k84, 2023-09-21:
> Ok. I have a spare EZSP (pop) zigbee dongle somewhere, a tuya gateway and the tuya phmeter. I will try to sniff... this weekend hopefully.

Oh man thank you! I don't have the sniffer gear.

tschiex, 2023-09-22:
I used my POPP key and it correctly sniffs the Zigbee traffic in my main Zigbee network. Wireshark already had the trust center and default network key in it. I powered up the Tuya gateway, associated it with my wifi router, and then Zigbee-paired the pH meter to it. The problem I have is that I don't know the network key of the Tuya gateway Zigbee network. I successfully repaired the ph meter to it but... saw no "Device Announcement" packet coming through in Wireshark.

I retried several times, to no avail.

cloudbr34k84, 2023-09-23:
> I used my POPP key and it correctly sniffs the Zigbee traffic in my main Zigbee network. Wireshark already had the trust center and default network key in it. I powered up the Tuya gateway, [...]

Is this because you don't have the network key?

tschiex, 2023-09-23:
My bad really. I was using the wrong ZigBee channel (had to try all of them... the Tuya Gateway chose 25). I now have the Device announcement and the network key, I added it to the Wireshark Zigbee keys store and have all the frames following the pairing of the pH meter. The Tuya GW  apparently writes attributes on it, as it starts. I'm not a Wireshark expert. I don't know how to save frames to make them available to Koenkk. Will ask Google...

tschiex, 2023-09-23:
OK. I have a pH-meter.pcapng file that should be fine (attached after zipping). I don't think there is any confidential info in it: the network is not supposed to survive these tests and the only device on it is the pH meter. I only saved the (udp.port == 17754) packets. To save your time parsing wireshark output, the network key for decryption is B3:D6:C7:C3:2C:D8:A2:74:F5:6C:FF:3E:6D:6E7F:00.

It looks like the GW is writing attribute 0xffde with 0x13 when it starts (packet 120).  Then it starts using the Tuya cluster (packet 129) issuing command 0x03 on Endpoint 1.

[pH-meter.zip](https://github.com/Koenkk/zigbee2mqtt/files/12706031/pH-meter.zip)

Have fun !

Koenkk, 2023-09-23:
Thanks, please try with the following ext converter: https://gist.github.com/Koenkk/7262a8a36bb84327ff563e0b8256024b (only works on the latest-dev branch)

tschiex, 2023-09-23:
I will let cloudbr34k84 do the converter testing. I'm still running ZHA :-)

tschiex, 2023-09-23:
Same gist link/converter as before?

cloudbr34k84, 2023-09-23:
@Koenkk 
we are getting somewhere now
[image]

Koenkk, 2023-09-24:
Great, now it is a matter of completing the expose section, I've made a start but you can add the rest (I have no clue about what to fill in the descriptions and units): https://gist.github.com/Koenkk/7262a8a36bb84327ff563e0b8256024b

cloudbr34k84, 2023-09-24:
@Koenkk 
Thats ok, ill give it ago! and report back
How did you know to use "tuya.onEvent({queryOnDeviceAnnounce: true})" to get it working?

Koenkk, 2023-09-24:
@cloudbr34k84 In the sniff I saw the `dataQuery` from the device, then I remembered I had to do the same to get data reportings from another TuYa device I recently added.

cloudbr34k84, 2023-09-24:
@Koenkk , here is the update converted which now exposes the entities
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_v1jqz5cy',
        },
    ],
    model: 'WiFi Chlorine Meter BLE-YL01',
    vendor: 'yieryi',
    description: 'Smart WiFi Zigbee Chlorine Meter PH ORP EC TDS Salinity Temp CL Tester Swimming Pool Water Quality Analyzer USB Solar Powered',
    fromZigbee: [tuya.fz.datapoints, tuya.fz.gateway_connection_status],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({queryOnDeviceAnnounce: true}), // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.numeric('TDS Value', ea.STATE_S).withUnit('ppm').withDescription('Total Dissolved Solids Value'),
        e.temperature(),
        e.numeric('ORP Value', ea.STATE).withUnit('mV').withDescription('Oxidation Reduction Potential Value. If the ORP value is above 850mv, it means that the disinfectant has been added too much, and it is necessary to add water or change the water for neutralization. If the ORP value is below 487mv, it means that too little disinfectant has been added and the pool needs to be disinfected again'),
        e.numeric('ORP Max Value', ea.STATE_SET).withUnit('mV').withDescription('Oxidation Reduction Potential Max Value'),
        e.numeric('ORP Min value', ea.STATE_SET).withUnit('mV').withDescription('Oxidation Reduction Potential Min Value'),
        e.numeric('Chlorine Value', ea.STATE).withUnit('mg/L').withDescription('Free Chlorine Value. The water in the swimming pool should be between 6.5-8ph and ORP should be between 487-840mv, and the chlorine value will be displayed normally. Chlorine will not be displayed if either value is out of range'),
        e.numeric('Chlorine Max Value', ea.STATE_SET).withUnit('ppm').withDescription('Free Chlorine Max Value. '),
        e.numeric('Chlorine Min Value', ea.STATE_SET).withUnit('ppm').withDescription('Free Chlorine Min Value'),
        e.numeric('Battery Percentage', ea.STATE).withUnit('%').withDescription('Remaining battery in %, can take up to 24 hours before reported.').withValueMin(0).withValueMax(100),
        // e.backlightvalue(),
        e.numeric('PH Value', ea.STATE).withUnit('pH').withDescription('pH Value. If the PH value is lower than 6.5, it means that the water quality is too acidic and has impurities, and it is necessary to add disinfectant water for disinfection'),
        e.numeric('PH Max Value', ea.STATE_SET).withUnit('pH').withDescription('pH Max Value'),
        e.numeric('PH Min Value', ea.STATE_SET).withUnit('pH').withDescription('pH Min Value'),
        e.numeric('EC Value', ea.STATE).withUnit('µS/cm').withDescription('Electrical Conductivity Value'),
        e.numeric('EC Max Value', ea.STATE_SET).withUnit('µS/cm').withDescription('Electrical Conductivity Max Value'),
        e.numeric('EC Min Value', ea.STATE_SET).withUnit('µS/cm').withDescription('Electrical Conductivity Min Value'),
        e.numeric('Salinity Value', ea.STATE).withUnit('µS/cm').withDescription('Salt Value'),

    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'TDS Value', tuya.valueConverter.raw],
            [2, 'Temperature', tuya.valueConverter.raw.divideBy10],
            [7, 'Battery Percentage', tuya.valueConverter.raw],
            [10, 'PH Value', tuya.valueConverter.raw],
            [11, 'EC value', tuya.valueConverter.raw],
            [101, 'ORP Value', tuya.valueConverter.raw],
            [102, 'Chlorine Value', tuya.valueConverter.raw],
            [105, 'backlightvalue', tuya.valueConverter.raw],
            [106, 'PH Max Value', tuya.valueConverter.raw],
            [107, 'PH Min Value', tuya.valueConverter.raw],
            [108, 'EC Max Value', tuya.valueConverter.raw],
            [109, 'EC Min Value', tuya.valueConverter.raw],
            [110, 'ORP Max Value', tuya.valueConverter.raw],
            [111, 'ORP Min value', tuya.valueConverter.raw],
            [112, 'Chlorine Max Value', tuya.valueConverter.raw],
            [113, 'Chlorine Min value', tuya.valueConverter.raw],
            [117, 'Salinity Value', tuya.valueConverter.raw],

        ],
    },
};

module.exports = definition;

```

tschiex, 2023-09-24:
Given its behaviour when it's used on with Tuya, I would suspect that it is a wake-up mechanism, possibly for saving power. When the phMeter is not specifically used in the Tuya app, its side LED is off (I suspect this may be controlled through the "Backlight" DP). When you go to the phMeter device page, the LED on the side of the pH meter lights up and the measures come in and update quite quickly (every second or even faster. I observe this as I touch the temperature probe with my finger for example). If I leave the pH meter device page on the Tuya app, the LED on the side of the pH meter device turns off.

So, it is possible that the device will drain its batteries (possibly much) faster if it's not 'turned off' between measurement periods. Indeed, Temperature, EC, ORP and pH do not change very quickly in a swimming pool and it's far better to sleep for long period of times between measures.

Koenkk, could you give me the dataQuery packets numbers in my sniff capture, the one that "starts" the device?
I will sniff for a full session, where I start and then stop using the device from the Tuya app and see how it's turned off.

[120 more replies]