Koenkk/zigbee2mqtt #18705: TuYa ZS06 flooding messages

TEF2one, 2023-08-23:
### What happened?

I have a strange behaviour, when using the TuYa ZS06 IR blaster, If I send multiple commands from multiple blasters at once, one of the blasters starts to spam tons of messages causing everything to become unresponsive until I restart Z2M.

Below I have trended the number of MQTT messages per minute, and noticed the spike from normally around 70 to 1000 during the period of unresponsiveness that matched when I was using the blaster to turn off various devices in the house as I left...
[image]
After having logged all the messages during these times, I notice they mostly originated from a single IR blaster, one out of my 4 blasters but not always the same one...

When I send each IR command individually I do not replicate the issue, but when I send a group of them it systematically replicates the issue.

### What did you expect to happen?

I was not expecting a flood of MQTT messages from the ZS06 devices.

### How to reproduce it (minimal and precise)

I am reproducing this issue by sending multiple IR commands from distinct IR blaster at the same time.

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Smlight – Slzb-06

### Debug log

[Logs](https://docs.google.com/spreadsheets/d/e/2PACX-1vSp8i3llH0dmIO1BW5SXlk4UBErujqEuoFR_cMvVAr0w6UFs4VBwjsxfE4Ecvgyud35v-iSxqqt1cuK/pubhtml?gid=648319485&single=true)

shing6326, 2023-09-03:
I also facing the same issues

closed 2023-10-11 by github-actions[bot]

KiraPC, 2024-04-06:
I guess I have the same issue. I'm using the ZS06 with SmartIR, when I change channel with Alexa to a channel with 2 digits (eg 31) the bluster stop working until I restart it. Is there some news?

Is there a way to make the blaster dequeue a single message per time?

miloumeeloo, 2024-10-04:
Same issue. I created mqtt switchs and buttons to interact whith the blaster.
If i use them in more than 1 action on the same sequence, it always flood with this message:
 `{"last_seen":"2024-10-04T14:40:54.718Z","learn_ir_code":null,"learned_ir_code":"BTAjjhE7AkABA30GOwJAAUAHAn0GgiADQAcBOwJAAeAHB+ADAeADG8AL4AcB4AMXQAvgAwMLOwI7AlGgMCPACDsC","linkquality":240}`
 which is the last code i asked to learn.