Koenkk/zigbee2mqtt #18719: Legrand 067771 - OTA Issue - Update failed with reason: 'invalid image'

daangel27, 2023-08-24:
### What happened?

Hello,

Here is my model to be updated : 
[image]

In HA, the current version installed & last version available :
[image]

In Z2M, i see a firmware update available for my Legrand 067771.
[image]

Clicking the firmware update button, i've got the error 'Invalid image'.

Thanks for your help.
Regards

### How to reproduce it (minimal and precise)

In Z2M, clic on the firmware update button for Legrand 067771.

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

`
Debug 2023-08-24 13:54:12ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-000e-00474203-NLF-72.zigbee
Debug 2023-08-24 13:54:12OTA update checksum validation succeeded for '0x0004740000151d79'
Debug 2023-08-24 13:54:12getNewImage for '0x0004740000151d79', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4129,"imageType":14,"fileVersion":4669955,"zigbeeStackVersion":2,"otaHeaderString":" ","totalImageSize":254471}
Debug 2023-08-24 13:54:12Got new image for '0x0004740000151d79'
Debug 2023-08-24 13:54:12Starting upgrade
Debug 2023-08-24 13:54:16Received Zigbee message from 'Interrupteur lumières extérieures', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":4604927,"imageType":14,"manufacturerCode":4129}' from endpoint 1 with groupID 0
Info 2023-08-24 13:54:16MQTT publish: topic 'zigbee2mqtt/Interrupteur lumières extérieures', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":254,"ballast_minimum_level":254,"ballast_physical_maximum_level":254,"ballast_physical_minimum_level":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":0,"device_mode":"dimmer_off","last_seen":"2023-08-24T13:54:16+02:00","led_if_on":"ON","led_in_dark":"OFF","led_when_off":"OFF","linkquality":94,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4604927,"latest_version":4669955,"state":"available"},"update_available":true}'
Info 2023-08-24 13:54:24MQTT publish: topic 'zigbee2mqtt/Interrupteur lumières extérieures', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":254,"ballast_minimum_level":254,"ballast_physical_maximum_level":254,"ballast_physical_minimum_level":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":0,"device_mode":"dimmer_off","last_seen":"2023-08-24T13:54:24+02:00","led_if_on":"ON","led_in_dark":"OFF","led_when_off":"OFF","linkquality":94,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4604927,"latest_version":4669955,"state":"available"},"update_available":true}'
Debug 2023-08-24 13:54:24OTA update at 0.02%, remaining 57911.308020000004 seconds
Info 2023-08-24 13:54:24Update of 'Interrupteur lumières extérieures' at 0.02%, ≈ 965 minutes remaining
Info 2023-08-24 13:54:24MQTT publish: topic 'zigbee2mqtt/Interrupteur lumières extérieures', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":254,"ballast_minimum_level":254,"ballast_physical_maximum_level":254,"ballast_physical_minimum_level":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":0,"device_mode":"dimmer_off","last_seen":"2023-08-24T13:54:24+02:00","led_if_on":"ON","led_in_dark":"OFF","led_when_off":"OFF","linkquality":94,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4604927,"latest_version":4669955,"progress":0.02,"remaining":57911,"state":"updating"},"update_available":false}'
Info 2023-08-24 13:54:24MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Interrupteur lumières extérieures' at 0.02%, ≈ 965 minutes remaining","meta":{"device":"Interrupteur lumières extérieures","progress":0.02,"status":"update_progress"},"type":"ota_update"}'
Info 2023-08-24 13:54:24MQTT publish: topic 'zigbee2mqtt/Interrupteur lumières extérieures', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":254,"ballast_minimum_level":254,"ballast_physical_maximum_level":254,"ballast_physical_minimum_level":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":0,"device_mode":"dimmer_off","last_seen":"2023-08-24T13:54:24+02:00","led_if_on":"ON","led_in_dark":"OFF","led_when_off":"OFF","linkquality":98,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4604927,"latest_version":4669955,"progress":0.02,"remaining":57911,"state":"updating"},"update_available":false}'
Info 2023-08-24 13:54:24MQTT publish: topic 'zigbee2mqtt/Interrupteur lumières extérieures', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":254,"ballast_minimum_level":254,"ballast_physical_maximum_level":254,"ballast_physical_minimum_level":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":0,"device_mode":"dimmer_off","last_seen":"2023-08-24T13:54:24+02:00","led_if_on":"ON","led_in_dark":"OFF","led_when_off":"OFF","linkquality":94,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4604927,"latest_version":4669955,"progress":0.02,"remaining":57911,"state":"updating"},"update_available":false}'
Debug 2023-08-24 13:54:24Got upgrade end request for '0x0004740000151d79': {"status":150,"manufacturerCode":4129,"imageType":14,"fileVersion":4669955}
Debug 2023-08-24 13:54:24Update failed with reason: 'invalid image'
Debug 2023-08-24 13:54:24Update of 'Interrupteur lumières extérieures' failed (Error: Update failed with reason: 'invalid image')
Info 2023-08-24 13:54:24MQTT publish: topic 'zigbee2mqtt/Interrupteur lumières extérieures', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":254,"ballast_minimum_level":254,"ballast_physical_maximum_level":254,"ballast_physical_minimum_level":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":0,"device_mode":"dimmer_off","last_seen":"2023-08-24T13:54:24+02:00","led_if_on":"ON","led_in_dark":"OFF","led_when_off":"OFF","linkquality":94,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":4604927,"latest_version":4669955,"state":"available"},"update_available":true}'
Info 2023-08-24 13:54:24MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Interrupteur lumières extérieures' failed (Update failed with reason: 'invalid image')","meta":{"device":"Interrupteur lumières extérieures","status":"update_failed"},"type":"ota_update"}'
Info 2023-08-24 13:54:24MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Interrupteur lumières extérieures"},"error":"Update of 'Interrupteur lumières extérieures' failed (Update failed with reason: 'invalid image')","status":"error","transaction":"g43t8-2"}'
Error 2023-08-24 13:54:24Update of 'Interrupteur lumières extérieures' failed (Update failed with reason: 'invalid image')
Debug 2023-08-24 13:54:24Error: Update failed with reason: 'invalid image' at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:422:24`

lamerjack, 2023-08-25:
Same issue with a Bticino F20T60A that is a Legrand 412015
I tested also with SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P latest firmware. and with a different brand dongle.

Koenkk, 2023-08-25:
Let's continue in https://github.com/Koenkk/zigbee-OTA/issues/328

closed 2023-08-25 by Koenkk