Koenkk/zigbee2mqtt #18723: [New device support]:

patrick-foubert-56770, 2023-08-24:
### Link

https://www.aliexpress.com/item/1005005780858603.html?spm=a2g0o.order_detail.order_detail_item.6.6a2a7d56cKs1ZV

### Database entry

{"id":70,"type":"Router","ieeeAddr":"0xa4c138eccd4f5c72","nwkAddr":55104,"manufId":4417,"manufName":"_TZE200_hl0ss9oa","powerSource":"DC Source","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,57346,61184,60928,57344,1024],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS0225","manufacturerName":"_TZE200_hl0ss9oa","powerSource":4,"zclVersion":3,"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00257ccbe9","zoneState":0}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":20570}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":130,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1692902528160,"defaultSendRequestWhen":"immediate"}

### Comments

Need support for new device (proximity presense radar)

HA Z2M Version:  1.32.2-1

Koenkk, 2023-08-25:
Device is already supported in the dev branch and will be supported in the 1 September release. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-25 by Koenkk

patrick-foubert-56770, 2023-09-12:
Update to latest update broke the support for some devices such :
Zigbee Model
TS0001
Zigbee Manufacturer
_TZ3000_mantufyr
where "state" is not push and in HA i got the message 
Cette entité n'est plus fournie par l'intégration mqtt. Si l'entité n'est plus utilisée, supprimez-la depuis les paramètres.
This entity was not longer supported by mqtt...

Another solution is to built a converter for TS0225 and stay with my old version from backup