Koenkk/zigbee2mqtt #18724: Nodon SIN-4-2-20 and SIN-4-1-20 several issues

nanostra, 2023-08-24:
### What happened?

Hello everyone,
I have several Nodon modules, namely SIN-4-2-20 and SIN-4-1-20. 

The firmware version is displayed for some of them, but not all. 
An Firmware update is available for certain modules, but not for all of them. 
Additionally, some of the modules have extra characters in the ID model.
The NODON Connect application indicates that all of my modules are up to date.

How can I address these various issues?

I'm facing similar problems with Aqara modules as well. 
  They are all updated, but some don't display the firmware version correctly or showing an old version.

I’ve attempted to delete and reappair them, but it hasn't made a difference. 
I've also tried modifying the .json files under "devices," but the issue keeps reoccurring.

Note that the modules are functioning properly.

[image]

[image]

[image]

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (EFR32MG21)

nanostra, 2023-09-01:
> Same issue as already been reported here I guess? #18412

Hello, thank you, but, chese are totally different issues...

closed 2023-10-09 by github-actions[bot]

Pocky0013, 2023-11-02:
since August. it's exactly the same thing chosen for my 12 devices.
- Strange character in zigbee pattern
- Mention that an update is available, but nothing happens. With version 1.33.1-1 of Z2M, I have an error (no image block request). But since this morning with the new version 1.33.2-1, I no longer even have any errors and nothing happens.
- HA continues to mention that there are updates available.

[image]
[image]

AlexisPolegato, 2023-11-08:
Hello @Pocky0013 ,
You had an update in progress for "CMD Store Salon" in your screenshot ; is your update over now ?
What is the current version of your device ?

For other devices, were you able to finally launch the OTA ?

Pocky0013, 2023-11-08:
@AlexisPolegato 
Thank you for your answer.

I have no updates in progress in Z2M, yet HA continues to run as if an update in progress were present.
But this is new, before HA version 2023.11, I simply had the mention that an update was available.

[image]

But in Z2M I don't have any updates in progress and it doesn't work. I can't start an update i get an error : "device not respond to OKTA update"

[image]

I was able to update the device: SIN-4-RS-20
But all my devices: SIN-4-1-20 And SIN-4-2-20 it is not possible to update

Pocky0013, 2023-11-08:
Version For SIN-4-2-20  is :

AlexisPolegato, 2023-11-08:
What kind of Zigbee dongle are you using @Pocky0013  ?

Pocky0013, 2023-11-08:
@AlexisPolegato for my side is a : Sonoff_Zigbee_3.0_USB_Dongle_Plus

nanostra, 2023-11-09:
Hello, 

with the new realease [1.33.2] commit: [9996c93], I can update some of my Sin-4-1-20 and SIN-4-2-20.

I managed to update everyone who had a firmware version entered in the column. For those who do not have a firmware version, it is possible to launch the update but nothing happens.

The never-ending story !

[image]

Pocky0013, 2023-11-14:
@nanostra I wrote to NodOn and received this good answer : 
" Nous travaillons depuis la semaine passée sur l'intégration de nos firmwares sur le dépot Zigbee2MQtt pour que cela puisse se faire directement sous Home Assistant. Nous réalisons actuellement les premiers tests en local avant de valider la mise à dispo officielle."

hoping that the coming weeks unlock the solution

AlexisPolegato, 2023-11-14:
> I managed to update everyone who had a firmware version entered in the column. For those who do not have a firmware version, it is possible to launch the update but nothing happens.

The Firmware Version is displayed on latest versions of the Firmware. So if you don't see the Firmware version, it simply means that the device has not been update yet.
There is an issue in the Firmware Update mechanism in Z2M, so it would work randomly. If you retry several times it could work eventually - This is why you have been able to update some devices and not the others.

I will try to work on it on the following days.

nanostra, 2023-11-14:
> The Firmware Version is displayed on latest versions of the Firmware. So if you don't see the Firmware version, it simply means that the device has not been update yet. There is an issue in the [...]

Thank You... I would like to point out that it works the first time when a firmware version is displayed before starting the update.

nanostra, 2024-10-11:
ONE year later...and a lot of crappy or incomplete firmware updates!!! Buy Nodon?

[image]

[image]

yoch, 2024-12-03:
I'm encountering the same issue with NULL characters in NodOn SIN-4-2-20 `model_id`, that are causing JSON validation failures and preventing the successful storage of devices information in our database (PostgreSQL). Maybe the JSON validity can be enforced ?

DaCarei, 2024-12-11:
Hello, I had the same issue and I contacted Nodon. 
They found it surprising but it is a really common problem once you look at different forum. 
There is a solution to update manually with android  :

1. Go in Bluetooth and search for new devices.
2. Nodon devices start with SINxxxx
3. Look at the IEEE address in Z2M (ex. 0xf4b3b1fffee4239d), the last 4 caracters (239d)
4. With your phone, pair to the SIN239d in this example
5. Then with the application NODON Connect search the device and you will be able to download the latest firmware.

source :
https://forum.hacf.fr/t/modules-fil-pilote-nodon-sin-4-fp-21-probleme-comptage-energy/49237/46.

I don't know yet if this operation solves everything but it might help.