Koenkk/zigbee2mqtt #18726: IKEA E1524/E1810 - TRADFRI remote control do not work anymore...

zibous, 2023-08-25:
### What happened?

No more working on Dashboard.

[image]

Devices are only displayed under Devices and not on the dashboard. The control settings are missing.

### What did you expect to happen?

Since the last update, the IKEA E1524/E1810 - TRADFRI remote control are no longer displayed correctly on the dashboard and I can't manage them either. MQTT messages work perfectly.

I can't revert to the previous version because Homeassistant Discovery doesn't work on this one !

### How to reproduce it (minimal and precise)

Devices deleted and reconnected, no change.

### Zigbee2MQTT version

1.32.2 commit: 1ec1e57

### Adapter firmware version

0x26660700

### Adapter

ConBee2/RaspBee2

### Debug log

{
    "battery": 100,
    "last_seen": "2023-08-24T21:10:46+02:00",
    "linkquality": 111,
    "action": "",
    "device": {
        "applicationVersion": 33,
        "dateCode": "20190401",
        "friendlyName": "remotecontrol-ikea-01",
        "hardwareVersion": 1,
        "ieeeAddr": "0xec1bbdfffe59df34",
        "manufacturerID": 4476,
        "manufacturerName": "IKEA of Sweden",
        "model": "E1524/E1810",
        "networkAddress": 23726,
        "powerSource": "Battery",
        "softwareBuildID": "2.3.014",
        "stackVersion": 98,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "elapsed": 2508,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}

Version: 1.32.2 commit: 1ec1e57
Coordinator type: ConBee2/RaspBee2
Coordinator revision: 0x26660700
Coordinator IEEE Address: 0x00212effff060927
Frontend version: 0.6.133

2023-08-25 09:25:23MQTT publish: topic 'conbee2mqtt/remotecontrol-ikea-02', 
{
  "action": null,
  "battery": 87,
  "device": {
    "applicationVersion": 33,
    "dateCode": "20190401",
    "friendlyName": "remotecontrol-ikea-02",
    "hardwareVersion": 1,
    "ieeeAddr": "0x680ae2fffe81f382",
    "manufacturerID": 4476,
    "manufacturerName": "IKEA of Sweden",
    "model": "E1524/E1810",
    "networkAddress": 14756,
    "powerSource": "Battery",
    "softwareBuildID": "2.3.014",
    "stackVersion": 98,
    "type": "EndDevice",
    "zclVersion": 3
  },
  "elapsed": 3689705,
  "last_seen": "2023-08-25T09:25:23+02:00",
  "linkquality": 143,
  "update": {
    "installed_version": -1,
    "latest_version": -1,
    "state": null
  },
  "update_available": null
}

c4software, 2023-09-02:
Hi,

Same for me since the last update. I have to long press buttons to make it registered in HA. I try to change the battery same issue.

zibous, 2023-09-04:
**Nothing has changed with the version 1.33.0.**

Zigbee2MQTT version: [1.33.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.0) commit: [7ee207f](https://github.com/Koenkk/zigbee2mqtt/commit/7ee207f)
Coordinator type: ConBee2/RaspBee2
Coordinator revision: 0x26660700
Coordinator IEEE Address: 0x00212effff060927
Frontend version:  0.6.135

It's just that one device has received the new firmware, the OTA update doesn't work on the second. But nothing changes with firmware 24.4.5 either.

MQTT is working.

[image]

[image]

MattWestb, 2023-09-04:
Servus @zibous 
The difference is the first have one good route to the coordinator and back and the second is having one not not stable rout then the both OSRAM plugs is corrupting the frames and is making. Put the OSRAM plugs (and some lights) in the box of bad Zigbee devices and use good routing devices all shall working much better in you network.

zibous, 2023-09-05:
@MattWestb 

Thanks for the information, removing OSRAM plugs and link TRADFRI remote controls direct to the coordinaten. 
Now the firmware update for the TRADFRI remote control IKEA E1524/E1810 is now working, but the The IKEA E1524/E1810 - TRADFRI remote control are no longer displayed correctly on the dashboard and I can't manage them either.

- And the TRADFRI signal repeaters cannot be updated ?
[image]

MattWestb, 2023-09-05:
The signal repeaters can having problem getting updated from some version is my experience (downloading it OK but not flashing it and is looping the update infinity) and i have doing one with J-Link flashing.

The Rasp/CorBee is very "dominate" and like have all devices connected to its self = bad for one mesh network then its being one star network and is not working if the coordinator is offline. You can trying repairing the device to one good router and see if its staying there or if its jumping to the coordinator.
(In ZHA and EZSP i blocking the coordinator having end devices as direct children and they working much better with one router that is 24/7 online then the coordinator)

I recommending putt in some more good (or better not bad) routers so the network can finding other routes then one rout is getting problems and its making your network more stable and robust for interference and device problems.

The latest OTA is making one function upgrade and is changing the device properties and for getting the new one try deleting it and adding it new so the system can getting all information of it and reconicing it OK.

MattWestb, 2023-09-05:
You have some Aqara / Xiaomi sensors that is connected to the coordinator and they is very likely leaving the network if they is trying communicating and you is restarting the Z2M or the firmware is having problems. Try repairing them to one router if its possible (Aqara / Xiaomi is never jumping they is leaving silent the network).

My old standard is IKEA Outlets with one OnOff remote and Aqara Weather in every room and then i have adding more devices but its the "backbone" (They is good routers and is also stable if not having very bad devices that is killing them).

The Sonoff Zigbee 3.0 Dongle Plus is having fixed Zigbee 3 end device handling but is still having some problems so better waiting until the firmware is stable for testing with over 40 devices.

I dont using Z2M so i cant telling if its somthing wrong in the system but all IKEA controllers is working OK in ZHA but some have not getting the battery fixed for the last firmware but its one very small thing for my.
I wold deleting the remote and then restarting Z2M and adding it new then the system is stable and see if its being added OK but i dont knowing if Z2M is deleting it  100% or reusing old data that  is not correct for the device.

closed 2023-10-13 by github-actions[bot]