Koenkk/zigbee2mqtt #18731: After upgrading my RPI, my zigbee2mqtt docker not started

rafalolb, 2023-08-25:
### What happened?

I have installed zigbee2mqtt docker on my RPI.

1 years works fine, but 1 months ago I upgrade my RPI and I get the message:
Logs:
`Zigbee2MQTT:info  2023-08-25 12:25:11: Logging to console and directory: '/app/data/log/2023-08-25.12-25-11' filename: log.txt
Zigbee2MQTT:info  2023-08-25 12:25:11: Starting Zigbee2MQTT version 1.32.2 (commit #1ec1e57)
Zigbee2MQTT:info  2023-08-25 12:25:11: Starting zigbee-herdsman (0.17.2)
Zigbee2MQTT:error 2023-08-25 12:25:32: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-08-25 12:25:32: Failed to start zigbee
Zigbee2MQTT:error 2023-08-25 12:25:32: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-08-25 12:25:32: Exiting...
Zigbee2MQTT:error 2023-08-25 12:25:33: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:59:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)`

Then I back to my previous RPI backup and zigbee2mqtt works.

Today I ones again upgrade my RPI, and I have the same problem.

My device is available:
`rafal@raspberrypi:~ $ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 08-25 11:42 /dev/ttyACM0
rafal@raspberrypi:~ $ ls -l /dev/serial/by-id
ls: nie ma dostępu do '/dev/serial/by-id': Nie ma takiego pliku ani katalogu
`

Configuration is good.
What can I do?

### Zigbee2MQTT version

1.32.2 (commit #1ec1e57)

### Adapter firmware version

-

### Adapter

 Sonoff ZBDongle-E

### Debug log

Zigbee2MQTT:info  2023-08-25 12:25:11: Logging to console and directory: '/app/data/log/2023-08-25.12-25-11' filename: log.txt
Zigbee2MQTT:info  2023-08-25 12:25:11: Starting Zigbee2MQTT version 1.32.2 (commit #1ec1e57)
Zigbee2MQTT:info  2023-08-25 12:25:11: Starting zigbee-herdsman (0.17.2)
Zigbee2MQTT:error 2023-08-25 12:25:32: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-08-25 12:25:32: Failed to start zigbee
Zigbee2MQTT:error 2023-08-25 12:25:32: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-08-25 12:25:32: Exiting...
Zigbee2MQTT:error 2023-08-25 12:25:33: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:59:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

william-sy, 2023-08-25:
`Configuration is good.`

Please provide the config :)

y8s, 2023-08-25:
This also happened to me after a recent upgrade. Using ZBDongle-E in docker.

I can view the USB device in the docker container via 
```
pi@iot:~/docker $ docker exec -it zigbee2mqtt /bin/sh
/app # lsusb
Bus 001 Device 007: ID 1915:cafe
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 004: ID 1a86:55d4 (<--- that's the zigbee dongle)
Bus 001 Device 003: ID 1a40:0101
Bus 001 Device 002: ID 2109:3431
Bus 002 Device 002: ID 174c:55aa
Bus 002 Device 001: ID 1d6b:0003
Bus 001 Device 005: ID 04d9:0007
Bus 001 Device 006: ID 051d:0002
/app #

```
Here's my compose file:
I have a udev rule to create a device called /dev/zbdongle-e, but that doesn't work either. 
I tried a few other ways of passing the USB through but they don't work either. Those are commented. The original method is uncommented.

```
     zigbee2mqtt:
       container_name: zigbee2mqtt
       restart: unless-stopped
   #    image: koenkk/zigbee2mqtt:latest-dev
       image: koenkk/zigbee2mqtt
       volumes:
         - ./zigbee2mqtt/data:/app/data
         - /run/udev:/run/udev:ro
       ports:
         - 55555:55555
       environment:
         - TZ=America/New_York
   #      - DEBUG=zigbee-herdsman*
       devices:
   #      - /dev/zbdongle-e:/dev/ttyUSB0
         - /dev/ttyACM1:/dev/ttyUSB0
   #      - /dev/bus/usb:/dev/bus/usb
   #      - /dev/bus/usb/001/004:/dev/ttyUSB0
```
z2m config relevant parts (minus devices and key):

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://10.10.10.40
frontend:
  port: 55555
  host: 0.0.0.0
serial:
  adapter: ezsp
  port: /dev/ttyUSB0
advanced:
  log_level: warn
```

with herdsman debug on I get this:

```
zigbee2mqtt  | 2023-08-25T14:14:24.186Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/zbdongle-e","baudRate":115200,"rtscts":false,"autoOpen":false}
zigbee2mqtt  | 2023-08-25T14:14:24.268Z zigbee-herdsman:adapter:ezsp:erro Connection attempt 1 error: Error: Error while opening serialport 'Error: Error: Not a tty setting custom baud rate of 115200'
zigbee2mqtt  |     at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:94:28)
zigbee2mqtt  |     at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:75:22)
zigbee2mqtt  |     at /app/node_modules/@serialport/stream/dist/index.js:111:18
zigbee2mqtt  | 2023-08-25T14:14:29.277Z zigbee-herdsman:adapter:ezsp:ezsp Next attempt 2
```

rafalolb, 2023-08-25:
My configuration:

```
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.10.17
  user: zigbee2mqtt_client
  password: zigbee2mqtt_client_olb
serial:
  port: /dev/ttyACM0
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: info
device_options:
  legacy: false
devices:
  '0xa4c1384138434fc0':
    friendly_name: CzujnikTemperatury1
    availability: true
    temperature_calibration: ''
frontend:
  port: 8080
  host: 0.0.0.0
  url: https://zigbee2mqtt.myhouse.org
```

docker-compose:
```
version: '3'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    volumes:
      - /home/rafal/zigbee2mqtt/data:/app/data
      - /run/udev:/run/udev:ro
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    restart: unless-stopped
    ports:
      - 8080:8080
    environment:
      - TZ=Europe/Warsaw
```

rafalolb, 2023-08-25:
device from docker:
```
rafal@raspberrypi:~ $ docker exec -it zigbee2mqtt /bin/sh
/app # ls -l /dev/ttyACM0
crw-rw----    1 root     dialout   166,   0 Aug 25 19:43 /dev/ttyACM0
/app # exit
rafal@raspberrypi:~ $ 
```

rafalolb, 2023-08-25:
> `Configuration is good.`
> 
> Please provide the config :)

Configs added in previous message.

william-sy, 2023-08-26:
Thanks all, I have done some digging. From what I can gather is that you are in need of drivers :( 

https://community.home-assistant.io/t/synology-docker-usb-zigbee-cc2652rb-no-ttyacom-or-ttyusb/340415/6
https://github.com/Koenkk/zigbee2mqtt/discussions/8840
https://github.com/robertklep/dsm7-usb-serial-drivers

Not sure why the new kernel you are using has dropped support for it, or other reasons why its not seen as a TTY device, for that that would need proper debugging, I have a spare PI and USB dongle here but for that I would need more info such as which pi, What OS and what kernel.

y8s, 2023-08-26:
I added all those drivers to my `/etc/modules` but it didn't fix it. 

I reinspected my USB devices and it turned out that my zigbee dongle moved from `/dev/ttyACM1` to `/dev/ttyACM0`

(verified with `udevadm info /dev/ttyA* | less`)

docker device now looks like:
```
devices:
      - /dev/ttyACM0:/dev/ttyUSB0
#      - /dev/ttyACM1:/dev/ttyUSB0
```

william-sy, 2023-08-26:
> I added all those drivers to my `/etc/modules` but it didn't fix it.
>
> I reinspected my USB devices and it turned out that my zigbee dongle moved from `/dev/ttyACM1` to `/dev/ttyACM0`
>
> (verified with [...]

For that I use a udev rule so that it wont happen again: 

``` 
ACTION=="add", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60", ATTRS{serial}=="<snip>", SYMLINK+="sonoff_zigbee_usb_1" 
```

y8s, 2023-08-26:
I have a udev rule but it wasn't working in docker/z2m

what does `ATTRS{serial}=="<snip>` do?

william-sy, 2023-08-26:
> `ATTRS{serial}=="<snip>` 

Yes I removed my dongle serial number, but your udev rule seems overly complex.. 

Note that: 
```
/dev/zbdongle-e -> bus/usb/001/004
```

For me is just: 

```
/dev/sonoff_zigbee_usb_1 -> ttyUSB0
```
Not that it should really matter I guess..

y8s, 2023-08-26:
> > `ATTRS{serial}=="<snip>`
> 
> Yes I removed my dongle serial number, but your udev rule seems overly complex..
> 
> Note that:
> 
> ```
> /dev/zbdongle-e -> bus/usb/001/004
> ```
> 
> For me is just:
> 
> ```
> /dev/sonoff_zigbee_usb_1 -> ttyUSB0
> ```
> 
> Not that it should really matter I guess..

Which subsystem do you use for the dongle? is it USB or TTY?

william-sy, 2023-08-26:
Here I use TTY, Works perfect on Rocky and also on Debian

y8s, 2023-08-26:
that was my issue then. the rule was USB subsystem. I changed it to TTY and:

```
pi@iot:~ $ ll /dev/zbdongle-e
lrwxrwxrwx 1 root root 7 Aug 26 14:33 /dev/zbdongle-e -> ttyACM0
pi@iot:~ $
```
and now 
```
devices:
      - /dev/zbdongle-e:/dev/ttyUSB0
``` 
works fine with z2m.

william-sy, 2023-08-27:
Awesome glad you fixed it, please close if fixed :)

rafalolb, 2023-08-27:
>I would need more info such as which pi, What OS and what kernel.

```
rafal@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

```
rafal@raspberrypi:~ $ uname -mrs
Linux 6.1.21-v8+ aarch64
rafal@raspberrypi:~ $ cat /etc/debian_version
11.7
```

```
rafal@raspberrypi:~ $ lscpu
Architecture:                    aarch64
Byte Order:                      Little Endian
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
Vendor ID:                       ARM
Model:                           3
Model name:                      Cortex-A72
Stepping:                        r0p3
CPU max MHz:                     1500,0000
CPU min MHz:                     600,0000
BogoMIPS:                        108.00
L1d cache:                       128 KiB
L1i cache:                       192 KiB
L2 cache:                        1 MiB
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Mmio stale data:   Not affected
Vulnerability Retbleed:          Not affected
Vulnerability Spec store bypass: Vulnerable
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fp asimd evtstrm crc32 cpuid
```

What driver I need install?

william-sy, 2023-08-27:
May be silly, And I did over look it too but you posted this : 

```
rafal@raspberrypi:~ $ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 08-25 11:42 /dev/ttyACM0
```

Then your config was like :

```
       devices:
   #      - /dev/zbdongle-e:/dev/ttyUSB0
         - /dev/ttyACM1:/dev/ttyUSB0
```

Which obviously wont work it should be :

```
       devices:
         - /dev/ttyACM0:/dev/ttyUSB0
```

And then it should work

y8s, 2023-08-27:
> May be silly, And I did over look it too but you posted this :
...
> And then it should work

I have to apologize. I piped up with a similar issue in the OP's thread and may have confused the issue.

You've got quotes from @rafalolb and myself there mixed together.

@rafalolb try looking at your specific usb device and making sure it's actually the right one.

`sudo udevadm info /dev/ttyACM0` should show a bunch of info including a line like:
```
E: ID_USB_MODEL=SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
```
if it doesn't, try the other `/dev/ttyACM*` devices.

william-sy, 2023-08-27:
Ah, ok. mybad

william-sy, 2023-08-28:
Can you add these to your docker compose :) ? 
```
    cap_add:
      - NET_ADMIN
      - SYS_MODULE
```

rafalolb, 2023-08-28:
> Can you add these to your docker compose :) ?

Yes, no problem. I added 3 lines of the end of my docker compose file, I recreated docker container (with pulled image) and I get the same error:
`Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)`

william-sy, 2023-08-29:
Hmmm I am starting to run out of ideas... 

Nothing else hogging the USB stick? no secret ZHA process running on HA itself that already has a connection to the stick?

You can also sort of check this with `lsof | grep /dev/ttyACM0`
If you stop the z2m container this list should return sort of empty

william-sy, 2023-08-29:
Did you also look at this page?: 

https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port

rafalolb, 2023-08-29:
> You can also sort of check this with `lsof | grep /dev/ttyACM0`

If my docker is stopped I get info:
```
rafal@raspberrypi:~/Desktop/Konfiguracja/HomeAssistant/zigbee2mqtt pliki $ lsof | grep /dev/ttyACM0
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/110/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/bba5f0922e0a97e967984c475c59ab451799bef6bdd7e9df1797528caf616a8e/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/fe26867318933fff37f999b4466e93675cc85ff4576386a1110042fd12e49ff1/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/3d74806b560e8819a9aa888b42ddb8ab678bcf16c6b024741e5fa9131c13d476/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/56c5bf3a1b20f847008fa1bb0afe733c7c20c58102731742a129ebc515278815/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/a44fb1290294b0d3f03d9b226e848387bd8bf1b1938e6c821657e8bb90a0e4cb/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/a910eadfc4a3c661accbab46b4d3ea801b2d0a305617f372920e4f3f69d3bc9f/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/e9d43fee38fbdf541a0c67127a0651b96906b8f2322a7f6c8633517a286c468d/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/f119c5bc1a5298a7078cd7ce935cc50f83c29c40a778edaceee4a0c37124c7a4/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/facecb139e33f89f48c2739d63bf8025ff5bf685ad980d0e4ba2a0d6d7ff64ed/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/6b733946fe0b4f824860facb47aa3b3c81f7d412393582a1651c81d044fefdeb/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/78526b3c4faee4788068547f84028af87a30978582980b417bcacd0700408eb9/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/5d6de593bab965fda2610bc595343982b56dfba4b27ce50c4a7fc1e3ba8f6dfb/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() overlay file system /var/lib/docker/overlay2/771e6eaeb701208c4ed74688126c41a07f7344ec66848440b8768c47797ec12b/merged
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/default
      Output information may be incomplete.
lsof: WARNING: can't stat() tmpfs file system /var/lib/docker/containers/f38d1601b4173b9109f5283f69631b34e2b9eec88436d1f1a8bca63fcf5496c8/mounts/shm
      Output information may be incomplete.
lsof: WARNING: can't stat() tmpfs file system /var/lib/docker/containers/01f96f0c80e8137a52fb3a76d268f7c7982644e186d01a97bc1cccfc0699ba9e/mounts/shm
      Output information may be incomplete.
lsof: WARNING: can't stat() tmpfs file system /var/lib/docker/containers/d91c7459285495711a1f18dda0a6162f75384db38137392e0480a930f92f2e48/mounts/shm
      Output information may be incomplete.
lsof: WARNING: can't stat() tmpfs file system /var/lib/docker/containers/b8ccad4dab2824c9ba853c5b33174b98fc7d6f65d9e8f05b12527649b69644b4/mounts/shm
      Output information may be incomplete.
lsof: WARNING: can't stat() tmpfs file system /var/lib/docker/containers/7404b866122efa432ff5e58ab5afd24f58d464ec794604fd53ddefa776cec603/mounts/shm
      Output information may be incomplete.
lsof: WARNING: can't stat() tmpfs file system /var/lib/docker/containers/47e8d06ebe0f46535a36885d4ed0c8121cf367847ed6924585081096834fa624/mounts/shm
      Output information may be incomplete.
lsof: WARNING: can't stat() tmpfs file system /var/lib/docker/containers/b35c104d37d85dc1facf20faa223f40f50e7af481482b46e648d49f0af50541b/mounts/shm
      Output information may be incomplete.
lsof: WARNING: can't stat() tmpfs file system /var/lib/docker/containers/fc1fca4ed40b1c011d72a9adda913c32e8295b3c5b0a5dbec766ef2fa036538c/mounts/shm
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/030eb29a2088
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/87fca61daea6
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/e167bd657d0d
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/3844eb20814b
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/c7c8d70f5c03
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/95db6942fa83
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/06d61ddcc961
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/c75df06c62d7
      Output information may be incomplete.
lsof: WARNING: can't stat() nsfs file system /run/docker/netns/9894e0117c39
[... cut ...]
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
lsof: no pwd entry for UID 911
rafal@raspberrypi:~/Desktop/Konfiguracja/HomeAssistant/zigbee2mqtt pliki $ 
```

rafalolb, 2023-08-29:
> Did you also look at this page?:
> 
> https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port

User runs Zigbee2MQTT has write access to the port:
```
rafal@raspberrypi:~/Desktop/Konfiguracja/HomeAssistant/zigbee2mqtt pliki $ docker exec -it zigbee2mqtt /bin/sh
/app # test -w /dev/ttyACM0 && echo success || echo failure
success
/app # exit
rafal@raspberrypi:~/Desktop/Konfiguracja/HomeAssistant/zigbee2mqtt pliki $ 

```

william-sy, 2023-08-29:
> User runs Zigbee2MQTT has write access to the port:
>
> ```
> rafal@raspberrypi:~/Desktop/Konfiguracja/HomeAssistant/zigbee2mqtt pliki $ docker exec -it zigbee2mqtt /bin/sh
> /app # test -w /dev/ttyACM0 && [...]

Alright that is a win :D so then something must be hogging the device somewhere..

rafalolb, 2023-08-29:
> You might want to run that LSOF with `sudo` That way the output will be better :)

I have a lot of results. Can I paging results?

william-sy, 2023-08-30:
> > You might want to run that LSOF with `sudo` That way the output will be better :)
> 
> I have a lot of results. Can I paging results?

`sudo lsof | grep /dev/ttyACM0` should not result in a huge list, but you can always put it in a file or something..

Pipe to file: `sudo lsof | grep /dev/ttyACM0 > ./output` 
Pages: `sudo lsof | grep /dev/ttyACM0 | less`

rafalolb, 2023-08-30:
I trying use those command yesterday, less command don't stopping scrolling results. Saving to file command created empty file :(

rafalolb, 2023-08-30:
When I studed documentation https://www.zigbee2mqtt.io/guide/adapters/#experimental and I found:

```
If Zigbee2MQTT fails to start, try adding the following to your configuration.yaml

serial:
  adapter: ezsp
```

And my dongle started :)

william-sy, 2023-08-31:
Glad its fixed :) please mark as closed