Koenkk/zigbee2mqtt #18737: zigbee2mqtt not working as docker image

chgaertner, 2023-08-25:
### What happened?

I have been using zigbee2mqtt (with Slaesh's USB stick) successfully for several years on docker containers (zigbee2mqtt, HomeAssistant and an mqtt server). Lately I switched to Raspberry Pi OS 64 bit, upgraded all containers and now I cannot start zigbee2mqtt anymore.

The error log is:
_Using '/app/data' as data directory
Zigbee2MQTT:info  2023-08-25 21:54:17: Logging to console and directory: 'data/log/2023-08-25.21-54-17' filename: log.txt
Zigbee2MQTT:info  2023-08-25 21:54:17: Starting Zigbee2MQTT version 1.32.2 (commit #1ec1e57)
Zigbee2MQTT:info  2023-08-25 21:54:17: Starting zigbee-herdsman (0.17.2)
Zigbee2MQTT:error 2023-08-25 21:54:38: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-08-25 21:54:38: Failed to start zigbee
Zigbee2MQTT:error 2023-08-25 21:54:38: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-08-25 21:54:38: Exiting...
Zigbee2MQTT:error 2023-08-25 21:54:38: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:59:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)_

I have three Zigbee adapters (2 x Slaesh's USB stick, 1 x SONOFF Zigbee 3.0 USB Dongle P version). Starting zigbee2mqtt fails with _all_ adapters. I have flashed the newest coordinator firmware to one of Slaesh's sticks and the SONOFF stick, the second Slaesh stick is brand new and was delivered today.

This is my docker-compose.yaml (includes device mappings for all sticks, only one is active; mac address redacted):
_version: '3'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt-cg
    image: koenkk/zigbee2mqtt:latest
    volumes:
      - ./data:/app/data
      - /run/udev:/run/udev:ro
    devices:
#      - /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_**_**_**_**-if00-port0:/dev/ttyUSB0
#      - /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_**_**_**_**-if00-port0:/dev/ttyUSB0
      - /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0:/dev/ttyUSB0
    ports:
      - 192.168.1.21:8088:8088
    restart: unless-stopped
    privileged: true
    environment:
      - TZ=Europe/Vienna_

I also checked from within the zigbee2mqtt docker container:
test -w /dev/ttyUSB0 && echo success || echo failure
This command results in "success". So the zigbee stick is definitely visible within the docker container.

Furthermore, the sticks do work with ZHA in HomeAssistant. HomeAssistant is not running when I try to start zigbee2mqtt to avoid any conflicts between zigbee2mqtt and ZHA.

### What did you expect to happen?

I expected zigbee2mqtt to start successfully.

### How to reproduce it (minimal and precise)

Start the docker container and see the live logging in Portainer.

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20230507

### Adapter

2 x Slaesh, 1 x Sonoff (see above)

### Debug log

Using '/app/data' as data directory
Zigbee2MQTT:info  2023-08-25 21:54:17: Logging to console and directory: 'data/log/2023-08-25.21-54-17' filename: log.txt
Zigbee2MQTT:info  2023-08-25 21:54:17: Starting Zigbee2MQTT version 1.32.2 (commit #1ec1e57)
Zigbee2MQTT:info  2023-08-25 21:54:17: Starting zigbee-herdsman (0.17.2)
Zigbee2MQTT:error 2023-08-25 21:54:38: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-08-25 21:54:38: Failed to start zigbee
Zigbee2MQTT:error 2023-08-25 21:54:38: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-08-25 21:54:38: Exiting...
Zigbee2MQTT:error 2023-08-25 21:54:38: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:59:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

william-sy, 2023-08-26:
Can you post your `configuration.yml` ? 
do you have: 
```
serial:
  port: /dev/ttyUSB99
```

chgaertner, 2023-08-27:
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.21:1883
  user: ******
  password: **********
  reject_unauthorized: true
  include_device_information: true
  keepalive: 60
  version: 4
serial:
  port: /dev/ttyUSB0
  disable_led: false
frontend:
  port: 8088
advanced:
  pan_id: ****
  log_level: info
  log_directory: data/log/%TIMESTAMP%
  log_file: log.txt
devices:
  '0x086bd7ffff4192bd':
    friendly_name: steckdose_wohnzimmer
  # (and so on)
```

william-sy, 2023-08-27:
2 things that come to mind:

1: the actual `/dev/serial/by-id/*` is wrong, doubtful but worth double checking ? 
2: you run privileged mode try removing that and run this instead:

```
    cap_add:
      - NET_ADMIN
      - SYS_MODULE
```

chgaertner, 2023-08-27:
Dear William,

thank you for your suggestions.

ad 1) I have double- and tripleckeched the "/serial/by-id" line, it is definitely correct. I copied it from the HomeAssistant docker-compose.yaml, where all the zigbee sticks are working.

ad 2) I tried and ... it worked!!!! And then I tried to _only remove the "privileged: true" line_ without adding "cap_add: NET_ADMIN, SYS_MODULE" and it still works. I have no error message during startup, I can see all the mqtt messages appearing live in the log output and the zigbee2mqtt webfrontend is accessible!

ad 3) The dongle is connected as /dev/ttyUSB0.

Thank you very much for this solution for my problem! I am still curious why _removing_ the "privileged: true" did make the container run, but this is a question I won't be able to answer. And unfortunately I do not know why I have added this line to my docker-compose-file some day in the past. It worked for several years with this option set. As this seems to have changed some time within the last years, it might be worth adding this to https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for those who updated their image.

Greetings and thank you again,
Christian

william-sy, 2023-08-27:
Cool, Dont forget to marked closed :)

Here is some more info in case you are curious: 
https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities

no problemo happy to help!
Cheers!

chgaertner, 2023-08-27:
Thank you for the link. After reading the article, I can imagine that the problem might be that a provileged docker container can access all devices on the host and therefore might lock the device by accessing it directly on the host. When the container wants to access the mapped device, it is not accessible. But zigbee2mqtt tries to access the mapped device (according to its configuration file) and cannot do so.

Greetings,
Christian

closed 2023-08-27 by chgaertner