Koenkk/zigbee2mqtt #18743: Unsupported attribute aurora downlights

Fezziemk4, 2023-08-26:
### What happened?

Fitted some of these bulbs https://www.zigbee2mqtt.io/devices/AU-A1ZBMPRO1ZX.html
The options for power on behaviour show up but when I try to change them I get 'unsupported attribute' error.

### What did you expect to happen?

To be able to change the power on behaviour as it shows in z2mqtt

### How to reproduce it (minimal and precise)

Try to change power on behaviour of the units

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8 SONOFF Zigbee 3.0 USB Dongle (E)

Koenkk, 2023-08-27:
Device doesn't support it, disabled it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-27 by Koenkk

Pel1can111, 2024-08-30:
I have installed some of these light and I'm getting the same error with the "Color temp startup" setting. @Fezziemk4, can you confirm if this has worked in the past or if this setting is working for you?