Koenkk/zigbee2mqtt #18744: Pairing Moes MS-105B no longer possible

marcus0704, 2023-08-26:
### What happened?

After updating to version 1.30.4 it's no longer possible to pair new devices (Moes MS-105B).
Allowing new devices to join does not detect new devices.

### What did you expect to happen?

New devices should be detected.

### How to reproduce it (minimal and precise)

Start Zigbee2MQTT
Click on "detect devices"
Bring Moes MS-105B into pairing mode
--> Zigbee2MQTT did not detect the device

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

20210708

### Adapter

Texas Instruments CC1352P-2

closed 2023-10-24 by marcus0704

gaetan-steijaert, 2023-11-17:
@marcus0704 were you able to fix this issue?

marcus0704, 2023-11-28:
@gaetan-steijaert: to be honest I think my issue was caused by too much distance between the coordinator and the device that should be paired. 
I brought device and coordinator in closer distance and the pairing worked fine.