Koenkk/zigbee2mqtt #18745: [New device support]: LIVARNO Blinds (LIDL / TS030F)

milkpirate, 2023-08-26:
### Link

https://www.lidl.de/p/livarno-home-automatik-verdunkelungsrollo-zigbee-smart-home-120-x-195-cm/p100359579

### Database entry

{   "id": 43,   "type": "EndDevice",   "ieeeAddr": "0x70ac08fffe3e9bbd",   "nwkAddr": 26011,   "manufId": 4098,   "manufName": "_TZB000_42ha4rsc",   "powerSource": "Battery",   "modelId": "TS030F",   "epList": [     1   ],   "endpoints": {     "1": {       "profId": 260,       "epId": 1,       "devId": 514,       "inClusterList": [         0,         4,         5,         3,         258,         1,         57345,         61184       ],       "outClusterList": [         25,         10       ],       "clusters": {         "genBasic": {           "attributes": {             "modelId": "TS030F",             "manufacturerName": "_TZB000_42ha4rsc",             "stackVersion": 0,             "dateCode": ""           }         }       },       "binds": [],       "configuredReportings": [],       "meta": {}     }   },   "appVersion": 70,   "stackVersion": 0,   "hwVersion": 1,   "dateCode": "",   "zclVersion": 3,   "interviewCompleted": true,   "meta": {},   "lastSeen": 1693047158607,   "defaultSendRequestWhen": "immediate" }

### Comments

I am trying to pair the Zigebee LIDL blinds, they get recognized as https://www.zigbee2mqtt.io/devices/TS130F.html but are still shown as  "unsupported":

[image]

Tried with 1.30.1, 1.32.2 and latest-dev. Shouldnt it be supported? What am I doing wrong?

Koenkk, 2023-08-29:
@JuMi2006 what's not working with this converter? code looks good

JuMi2006, 2023-08-29:
There happens nothing. I get nor response from the roller but also no error message about unknown converters.
Can I log something that would help? From the tuya development website I can control the device if it is connected to the tuya gateway/app.
Last week I tried some other converters (with success) just changing the fingerprint, but I have not written down what this was...my fault. There just the borders were not working - so principally the connection seems to be ok.

JuMi2006, 2023-08-29:
I figured out that this would work for basic functions:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	fingerprint: [
        {
            modelID: 'TS030F',
            manufacturerName: '_TZB000_42ha4rsc',
        },
    ],
    model: 'TS030F',
    vendor: '_TZB000_42ha4rsc',
    description: 'Smart blind controller',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_position_tilt, tz.cover_state],
    //onEvent: tuya.onEventSetTime,
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.currentPositionLiftPercentage(endpoint);
        },    
    exposes: [
        e.battery(), e.cover_position(),
        //e.enum('border', ea.STATE_SET, ['up', 'down', 'up_delete', 'down_delete', 'remove_top_bottom']),
        e.binary('motor_fault', ea.STATE, true, false),
    ],
	
	//meta: {
    //    // All datapoints go in here
    //    tuyaDatapoints: [
    //        [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
    //        [2, 'position', tuya.valueConverter.coverPosition],
    //        [3, 'position', tuya.valueConverter.raw],
    //        [12, 'motor_fault', tuya.valueConverter.trueFalse1],
    //        [13, 'battery', tuya.valueConverter.raw],
    //        [16, 'border', tuya.valueConverterBasic.lookup({'up': tuya.enum(0), 'down': tuya.enum(1), 'up_delete': tuya.enum(2), 'down_delete': tuya.enum(3)})],
    //    ],
    //},	
};

module.exports = definition;
```

Koenkk, 2023-08-30:
- Can you try changing the configure to `configure: tuya.configureMagicPacket,` and see if that helps? After this re-pair the device
- Is it possible to control the device physically? If yes, can you try changing the position and provide the debug log of this

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

sauer-m, 2023-09-01:
Thanks for your effort and quick integration. Im really curious :)
Can anyone tell me how long it will take till 15.66.0 (or later) is integrated in the iobroker zigbee-adapter? 😬
I know - a bit OT, but I couldnt find a proper way to ask this here)

Koenkk, 2023-09-01:
- Regarding the border, use the info below `Command: Write Attributes` to fill in a converter, [example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/3aad8a810b1ebc6435d0c7d16bc564e5dcbc708c/src/devices/tuya.ts#L28)
- Regarding the control, doesn't this work already?

JuMi2006, 2023-09-02:
Implemented the borders into the converter.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
  border: {
	  key: ['border'],
	  convertSet: async (entity, key, value, meta) => {
          switch (value) {
          case 'set_up':
              await entity.write(0xe001, {0xe001: {value: 0, type: 0x30}} /* set up */);
              break;
          case 'set_down':
              await entity.write(0xe001, {0xe001: {value: 1, type: 0x30}} /* set down */);
              break;
          case 'delete_up':
              await entity.write(0xe001, {0xe001: {value: 2, type: 0x30}} /* delete up */);
              break;
          case 'delete_down':
              await entity.write(0xe001, {0xe001: {value: 3, type: 0x30}} /* delete down*/);
              break;
		}
	  },
  },
};

const definition = {
	fingerprint: [{   
		modelID: 'TS030F',
        manufacturerName: '_TZB000_42ha4rsc',
        },],
    model: 'TS030F',
    vendor: '_TZB000_42ha4rsc', // LIDL LIVARNO blinds
    description: 'Smart blind controller',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_position_tilt, tz.cover_state, tzLocal.border],
    //onEvent: tuya.onEventSetTime,
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.currentPositionLiftPercentage(endpoint);
        },    
    exposes: [
        e.battery(), e.cover_position(),
        e.binary('motor_fault', ea.STATE, true, false).withDescription('motor fault'),
		e.enum('border', ea.STATE_SET, ['set_up', 'set_down', 'delete_up', 'delete_down', 'delete_init']).withDescription('set/delete borders'),
    ],
};

module.exports = definition;
```

Koenkk, 2023-09-02:
Instead of `up`/`down`, doesn't `open`/`close` make more sense?

milkpirate, 2023-09-02:
> Instead of `up`/`down`, doesn't `open`/`close` make more sense?

What are the other blindes using? We should use the same to stay consistent.

Great you guys were figuring that device out so fast 🥰

JuMi2006, 2023-09-02:
I figured out that the "Invert cover" function in the specific settings not works - so I builded a enum in the exposes.
It is not documentated and not supported by the native app but cluster 'closuresWindowCovering' and "tuyaMotorReversal" works from the dev console.
Also it is possible to read "moesCalibrationTime" (as factor of 100 in seconds) from the same cluster.

Maybe someone can clean the code? But - it works. ;)

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
  border: {
	  key: ['border'],
	  convertSet: async (entity, key, value, meta) => {
          switch (value) {
          case 'set_up':
              await entity.write(0xe001, {0xe001: {value: 0, type: 0x30}} /* set up */);
              break;
          case 'set_down':
              await entity.write(0xe001, {0xe001: {value: 1, type: 0x30}} /* set down */);
              break;
          case 'delete_up':
              await entity.write(0xe001, {0xe001: {value: 2, type: 0x30}} /* delete up */);
              break;
          case 'delete_down':
              await entity.write(0xe001, {0xe001: {value: 3, type: 0x30}} /* delete down*/);
              break;
		}
	  },
  },
  
  calibration_time: {
		key: ['calibration_time'],
		convertGet: async (entity, key, meta) => {
                await entity.read('closuresWindowCovering', ['moesCalibrationTime']);
            },
	},
	
  invert_cover: {
	  key: ['invert_cover'],
	  convertSet: async (entity, key, value, meta) => {
          switch (value) {
          case 'normal':
              await entity.write('closuresWindowCovering', {'tuyaMotorReversal': {value: 0}} /* set normal */);
              break;
          case 'invert':
              await entity.write('closuresWindowCovering', {'tuyaMotorReversal': {value: 1}} /* set invert */);
              break;
		}
	  },
  }
};

const fzLocal = {
	calibration_time: {
    cluster: 'closuresWindowCovering',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};
        if (msg.data.hasOwnProperty('moesCalibrationTime')) {
            const value = parseFloat(msg.data['moesCalibrationTime']) / 100.0;
            result.calibration_time = value;
        }
        return result;
    },
	},
	
	invert_cover: {
    cluster: 'closuresWindowCovering',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};
        if (msg.data.hasOwnProperty('tuyaMotorReversal')) {
            const value = parseFloat(msg.data['tuyaMotorReversal']);
            result.invert_cover = value;
        }
        return result;
    },
	}
	
}

const definition = {
	fingerprint: [{   
		modelID: 'TS030F',
        manufacturerName: '_TZB000_42ha4rsc',
        },],
    model: 'TS030F',
    vendor: 'Lidl' ,// '_TZB000_42ha4rsc', // LIDL LIVARNO blinds
    description: 'LIDL LIVARNO blinds',
    fromZigbee: [fz.cover_position_tilt, fzLocal.calibration_time, fzLocal.invert_cover],
    toZigbee: [tz.cover_position_tilt, tz.cover_state, tzLocal.border, tzLocal.calibration_time, tzLocal.invert_cover],
    //onEvent: tuya.onEventSetTime,
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.currentPositionLiftPercentage(endpoint);			
            await reporting.tuyaMotorReversal(endpoint);
        },    
    exposes: [
        e.battery(), e.cover_position(),
        e.binary('motor_fault', ea.STATE, true, false).withDescription('motor fault'),
		e.enum('border', ea.STATE_SET, ['set_up', 'set_down', 'delete_up', 'delete_down']).withDescription('set/delete borders'),
		e.enum('invert_cover', ea.STATE_SET, ['normal', 'invert']).withDescription('Invert cover'),
		e.numeric('calibration_time', ea.STATE_GET).withUnit('s').withDescription('Time of full cylce'),
    ],
};

module.exports = definition;
```

Koenkk, 2023-09-03:
@JuMi2006 thanks, integrated it!

Is there a model number written on the device/manual?

JuMi2006, 2023-09-03:
I was not sure writing the converter which model/manufacturer to choose.

LIVARNO is the brand - LIDL is distributor. I think LIDL as manufacturer could work - I think in future we will get more whitelabels.
Here are two links to the manuals (nearly identic):
short: https://stesbintegrationprod.blob.core.windows.net/public/articlemanual/d8b486fe-6b6e-4885-a8d4-6d95b5b877b2.pdf
long: https://stesbintegrationprod.blob.core.windows.net/public/articlemanual/52d824f6-280f-4bb1-9fb2-f156a25ef072.pdf

I also found a csa-document:
https://csa-iot.org/csa_product/automatic-blackout-roller-blind/
There it is named HG09648 - also found on the manuals.

Inside it has a ZS3L-Chipset (Tuya). 
The motor element is named TQL25-KT938-V2.02(1.2)

FYI:
I disassembled the roller in cause of the ugly design and find more information to get it reverse engeneered.
My brother will make me new brackets with a 3D-printer so that it can look very slim.

sauer-m, 2023-09-04:
> @JuMi2006 thanks, integrated it!
> 
> Is there a model number written on the device/manual?

The manual sais, art-nr. is IAN 415185_2204 (or 415186_2204 - I guess one is the 1m version the other the 1.2m version)

JuMi2006, 2023-09-04:
> BTW: Is there a way to program the start- and endpositions without the app? Maybe via payload? Its not possible to set them via remot, afaik :/

Yes. It's implemented via UI in z2m now by the latest commit

sauer-m, 2023-09-04:
> > BTW: Is there a way to program the start- and endpositions without the app? Maybe via payload? Its not possible to set them via remot, afaik :/
> 
> Yes. It's implemented via UI in z2m now by the latest commit

Im using the zigbee adapter by iobroker.
Can you give me an example payload or a source? 

[image]

JuMi2006, 2023-09-04:
I wrote all down in this thread - also for to remind. ;)
https://github.com/Koenkk/zigbee2mqtt/issues/18745#issuecomment-1703097724

closed 2023-10-12 by github-actions[bot]

Aziona, 2023-12-17:
Hi, my Livarno roller blinds roll the wrong way. The problem is that the curain is now about 3 cm from the wall. Is there a way to correct this using Zigbee2MQTT?

BenjiMcKenzie, 2024-02-23:
Hello all,

can anybody tell me how to pair this blinds? i bought two of the 1m blinds. I´m using Zigbee2MQTT with Home Assistant with the Skyconnect Adapter. If i´m looking for a Documentation how to pair, i get this Thread for pairing information.

br Benji

wormiedk, 2024-07-12:
> My brother will make me new brackets with a 3D-printer so that it can look very slim.
Did your brother manage to print brackets? If he did could you share the stl files?

zunepl, 2024-11-24:
Has anyone 
who is not a programming genius has success with this product :D 
I can tinker a bit, but nowhere near Cand. something something IT related.. 

Is there a easy how to guide?