Koenkk/zigbee2mqtt #18750: [New device support]: 3-Phase Multifunction Energy Meter PJ-DAC7361C

PetrozPL, 2023-08-26:
### Link

https://allegro.pl/oferta/licznik-2-kierunkowy-3f-zigbee-tuya-przekaznik-80a-13496396813

### Database entry

 {"id":48,"type":"Router","ieeeAddr":"0xa4c13812c266c281","nwkAddr":7873,"manufId":4417,"manufName":"_TZE200_x8fp01wi","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"y\u0002},iy\u0002},iz\u0002},iz\u0002},i{\u0002},i{\u0002},i|\u0002},i|\u0002},i","65506":54,"65508":1,"65534":0,"stackVersion":0,"dateCode":"","appVersion":70,"manufacturerName":"_TZE200_x8fp01wi","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1693069732032,"defaultSendRequestWhen":"immediate"}

### Comments

I've bought a 3-Phase Multifunction Energy Meter PJ-DAC7361C that isn't supported in zigbee2mqtt yet. So, I prepared it's support, thanks to other two similar models:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4938
https://github.com/Koenkk/zigbee2mqtt/issues/13097

It properly shows meter ID, current/voltage/power values for all 3 phases, total power, total energy consumed and produced, total monthly/daily power consumption, overcurrent treshold value and alarm status.

My solution isn't complete. I would like to fix it to use tuya.valueConverter presets like divideBy100, because when I've tried to use them it wasn't working. I've also would like to add support for re-program treshold value (if applicable), add support of "Frozen set", which I don't know what is, but is exposed at datapoint 11, and add the possibility to clear alarm, when triggerd. Finally, it would be great if complete support will be merged to zigbee2mqtt next release.

I would appriciate any help to make this support complete. :)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const exposes_presets_loc = {
	daily_energy: () => exposes.numeric('daily_energy', ea.STATE).withUnit('kWh').withDescription('Daily sum of consumed energy'),
	monthly_energy: () => exposes.numeric('monthly_energy', ea.STATE).withUnit('kWh').withDescription('Monthly sum of consumed energy'),
	produced_energy: () => exposes.numeric('produced_energy', ea.STATE).withUnit('kWh').withDescription('Sum of produced energy'),
	meterid: () => exposes.text('meterid', ea.STATE).withDescription('Meter S/N'),
	current_phase_a: () => exposes.numeric('current_phase_a', ea.STATE).withUnit('A').withDescription('Instantaneous measured electrical current on phase A'),
	voltage_phase_a: () => exposes.numeric('voltage_phase_a', ea.STATE).withUnit('V').withDescription('Measured electrical potential value on phase A'),
	power_phase_a: () => exposes.numeric('power_phase_a', ea.STATE).withUnit('W').withDescription('Instantaneous measured power on phase A'),
	power_phase_b: () => exposes.numeric('power_phase_b', ea.STATE).withUnit('W').withDescription('Instantaneous measured power on phase B'),
	power_phase_c: () => exposes.numeric('power_phase_c', ea.STATE).withUnit('W').withDescription('Instantaneous measured power on phase C'),
};
const le = exposes_presets_loc;

const valueConverter = {
    phaseA: {
        from: (v) => {
            return {current_phase_a: ((v[4] | v[3] << 8) / 1000), voltage_phase_a: ((v[1] | v[0] << 8) / 10), power_phase_a: ((v[7] | v[6] << 8))}
        },
    },
    phaseB: {
        from: (v) => {
            return {current_phase_b: ((v[4] | v[3] << 8) / 1000), voltage_phase_b: ((v[1] | v[0] << 8) / 10), power_phase_b: ((v[7] | v[6] << 8))}
        },
    },
    phaseC: {
        from: (v) => {
            return {current_phase_c: ((v[4] | v[3] << 8) / 1000), voltage_phase_c: ((v[1] | v[0] << 8) / 10), power_phase_c: ((v[7] | v[6] << 8))}
        },
    },
    threshold: {
        from: (v) => {
            const lookup = {0: 'not_set', 1: 'over_current', 3: 'over_voltage'};
            
            return {
                threshold_1: lookup[v[0]],
                threshold_1_value : (v[3] | v[2] << 8),
            };
        },
    },
	energy: {
		from: (v) =>  {
			return {energy: (v/100) }
		},
	},
	produced_energy: {
		from: (v) =>  {
			return {produced_energy: (v/100) }
		},
	},
	daily_energy: {
		from: (v) =>  {
			return {daily_energy: ((v[7] | v[6] << 8 | v[5] << 8 | v[4] << 8) / 100) }
		},
	},
	monthly_energy: {
		from: (v) =>  {
			return {monthly_energy: ((v[7] | v[6] << 8 | v[5] << 8 | v[4] << 8) / 100) }
		},
	},
	power: {
		from: (v) =>  {
			return {power: (v) }
		},
	},
	meterid: {
		from: (v) =>  {
			return {meterid: v.trim()}
		},
	},
    alarm: tuya.valueConverterBasic.lookup({'clear': 0, 'over_current': 1, 'over_power': 2, 'over_voltage': 4, 'wrong_frequency': 8}),
};

const tuyaExposes = {
    threshold: (l) => exposes.enum(`threshold_${l}`, ea.STATE, ['not_set', 'over_current', 'over_voltage'])
        .withDescription(`Treshold ${l} state`),
    tresholdValue: (l) => exposes.numeric(`threshold_${l}_value`, ea.STATE)
        .withDescription(`Treshold ${l} value`),
    alarm: () => exposes.enum(`alarm`, ea.STATE, ['clear', 'over_current', 'over_voltage', 'over_power', 'wrong_frequency'])
        .withDescription(`Alarm state`),  
};

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_x8fp01wi']),
    model: 'PJ-DAC7361C',
    vendor: 'Dongguan Pinjia Technology Co., Ltd',
    description: '3-Phase Multifunction Energy Meter',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    configure: tuya.configureMagicPacket,
    exposes: [
		tuya.exposes.switch(), 
		e.energy(),
		e.produced_energy(),
		le.daily_energy(),
		le.monthly_energy(),
		e.power(),
		le.power_phase_a(), le.voltage_phase_a(), le.current_phase_a(),
		le.power_phase_b(),  e.voltage_phase_b(),  e.current_phase_b(),
		le.power_phase_c(),  e.voltage_phase_c(),  e.current_phase_c(),
        tuyaExposes.threshold('1'),
		tuyaExposes.tresholdValue('1'),
        tuyaExposes.alarm(),
		le.meterid()
	],
    meta: {
        tuyaDatapoints: [
            [1, null, valueConverter.energy],
            [2, null, valueConverter.produced_energy],
			[3, null, valueConverter.monthly_energy],
			[4, null, valueConverter.daily_energy],
            [6, null, valueConverter.phaseA], // voltage, power and current - phase A
            [7, null, valueConverter.phaseB], // voltage, power and current - phase B
            [8, null, valueConverter.phaseC], // voltage, power and current - phase C
			[9, null, valueConverter.power], // Total power
            [10, 'alarm', valueConverter.alarm],
			//[11, null,valueConverter.unknown],//?
            [16, 'state', tuya.valueConverter.onOff],
			[17, null,valueConverter.threshold],//tresholds
			[18, null,valueConverter.meterid]
        ],
    },
};
```

PetrozPL, 2023-08-26:
Frozen set could be the day of the month/hour of the day when daily/monthly values are reset:

[image]

Cartol, 2023-09-01:
Hi,

i bought exact the same device via aliexpress some days ago. The manufacturer in my case is EARU as printed.
Mine is bidirictional by the way.

https://de.aliexpress.com/item/1005005679887461.html?spm=a2g0o.order_list.order_list_main.11.21ef5c5fQ1gjzL&gatewayAdapt=glo2deu

Nevertheless its a tuya device! So you have to go the "add new tuya device" way.
I have managed to get the external converter to work yesterday following the instructions this way:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_x8fp01wi',
        },
    ],
    model: 'TS0601_din_3-Phase',
    vendor: 'TuYa',
    description: 'Earu Zigbee 3-Phase bidirictional Meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
            // Here you should put all functionality that your device exposes
            tuya.exposes.voltageWithPhase('L1'), 
            tuya.exposes.voltageWithPhase('L2'), 
            tuya.exposes.voltageWithPhase('L3'),
            tuya.exposes.currentWithPhase('L1'), 
            tuya.exposes.currentWithPhase('L2'), 
            tuya.exposes.currentWithPhase('L3'),
            tuya.exposes.powerWithPhase('L1'), 
            tuya.exposes.powerWithPhase('L2'), 
            tuya.exposes.powerWithPhase('L3'),
            //tuya.exposes.powerFactorWithPhase('L1'),
            //tuya.exposes.powerFactorWithPhase('L2'),
            //tuya.exposes.powerFactorWithPhase('L3'),
            //tuya.exposes.energyFlowWithPhase('L1'), 
            //tuya.exposes.energyFlowWithPhase('L2'),
            //tuya.exposes.energyFlowWithPhase('L3'),
            //tuya.exposes.energyWithPhase('L1'), 
            //tuya.exposes.energyWithPhase('L2'),
            //tuya.exposes.energyWithPhase('L3'),
            //tuya.exposes.energyProducedWithPhase('L1'), 
            //tuya.exposes.energyProducedWithPhase('L2'),
            //tuya.exposes.energyProducedWithPhase('L3'),
            
            // Change the description according to the specifications of the device
            e.energy().withDescription('Total forward active energy'),
            e.produced_energy().withDescription('Total reverse active energy'),
            //e.enum('energy_flow', ea.STATE, ['consuming', 'producing']).withDescription('Direction of energy'),
            //e.numeric('update_frequency', ea.STATE).withUnit('s').withDescription('Update frequency'),
            //e.ac_frequency(),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
                
                [1, 'energy', tuya.valueConverter.divideBy100],
                [2, 'produced_energy', tuya.valueConverter.divideBy100],
                [6, null, tuya.valueConverter.phaseVariant2WithPhase('L1')],
                [7, null, tuya.valueConverter.phaseVariant2WithPhase('L2')],
                [8, null, tuya.valueConverter.phaseVariant2WithPhase('L3')],
        ],
    },
};

module.exports = definition;
```

Took the code from devices reporting same values in Koenkk's tuya.ts file. Searched for _TZE204 devices and copied parts of it.

The comment parts in exposes do not work i tried line after line. Zigbee2MQTT addon does not start after enabling any of this. 

Greetings

PetrozPL, 2023-09-02:
Hi Cartlol,

Thanks for your input. I wasn't aware that in tuya there's an expose part. Thanks to your suggestions I've modified my converter, so it's now more inline with other tuya devices from library. It would be great if we could match and replace as much tuya functions as possible. I don't have my device anymore, because it was initially bought for my brother and I was just checking it for him, but maybe you could make some more test on your device? Firstly, If you could check my modified version of converter that is using tuya functions and exposes, and see if all of devices functions are still working. 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const le = {
    daily_energy: () => exposes.numeric('daily_energy', ea.STATE).withUnit('kWh').withDescription('Daily sum of consumed energy'),
    monthly_energy: () => exposes.numeric('monthly_energy', ea.STATE).withUnit('kWh').withDescription('Monthly sum of consumed energy'),
    produced_energy: () => exposes.numeric('produced_energy', ea.STATE).withUnit('kWh').withDescription('Sum of produced energy'),
    meterid: () => exposes.text('meterid', ea.STATE).withDescription('Meter S/N'),
    current_phase_a: () => exposes.numeric('current_phase_a', ea.STATE).withUnit('A').withDescription('Instantaneous measured electrical current on phase A'),
    voltage_phase_a: () => exposes.numeric('voltage_phase_a', ea.STATE).withUnit('V').withDescription('Measured electrical potential value on phase A'),
    power_phase_a: () => exposes.numeric('power_phase_a', ea.STATE).withUnit('W').withDescription('Instantaneous measured power on phase A'),
    power_phase_b: () => exposes.numeric('power_phase_b', ea.STATE).withUnit('W').withDescription('Instantaneous measured power on phase B'),
    power_phase_c: () => exposes.numeric('power_phase_c', ea.STATE).withUnit('W').withDescription('Instantaneous measured power on phase C'),
    clear_fault: () => exposes.binary('clear_fault', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn ON to clear last the fault'),
    fault: () => exposes.enum(`fault`, ea.STATE, ['clear', 'over_current_threshold', 'over_voltage_threshold']).withDescription(`Fault state: status of the device (clear = nothing)`),
    threshold: (l) => exposes.enum(`threshold_${l}`, ea.STATE, ['not_set', 'over_current_threshold', 'over_voltage_threshold']).withDescription(`Threshold ${l} type: type of treshold ${l}`),
    tresholdProtection: (l) => exposes.binary(`threshold_${l}_protection`, ea.STATE_SET, 'ON', 'OFF').withDescription(`Treshold ${l} protection: OFF - alarm only, ON - relay will be off when threshold reached`),
    tresholdValue: (l) => exposes.enum(`threshold_${l}_value`, ea.STATE_SET, ['80A','63A', '50A', '40A', '32A', '25A', '20A', '16A', '10A' ]).withDescription(`Treshold ${l} value: Can be in Volt or Ampere depending on threshold setting. Setup the value on the device`),   
    monthly_energy_rst_day: () => exposes.numeric('monthly_energy_rst_day', ea.STATE_SET).withValueMin(1).withValueMax(31).withDescription('Day of the month for monthly consumed energy meter reset'),
    daily_energy_rst_hour: () => exposes.numeric('daily_energy_rst_hour', ea.STATE_SET).withValueMin(0).withValueMax(24).withDescription('Hour of the day for daily consumed energy meter reset'),
};

const valueConverter = {
    phaseA: {
        from: (v) => {
            return {current_phase_a: ((v[4] | v[3] << 8) / 1000), voltage_phase_a: ((v[1] | v[0] << 8) / 10), power_phase_a: ((v[7] | v[6] << 8))}
        },
    },
    phaseB: {
        from: (v) => {
            return {current_phase_b: ((v[4] | v[3] << 8) / 1000), voltage_phase_b: ((v[1] | v[0] << 8) / 10), power_phase_b: ((v[7] | v[6] << 8))}
        },
    },
    phaseC: {
        from: (v) => {
            return {current_phase_c: ((v[4] | v[3] << 8) / 1000), voltage_phase_c: ((v[1] | v[0] << 8) / 10), power_phase_c: ((v[7] | v[6] << 8))}
        },
    },
    energy_rst_hour_day: {
        from: (v) => {
            return {
                monthly_energy_rst_day: v[0],
                daily_energy_rst_hour: v[1],
            };
        },
    },
    threshold: {
        from: (v) => {
            const stateLookup = {0: 'not_set', 1: 'over_current_threshold', 3: 'over_voltage_threshold'};
            const protectionLookup = {0: 'OFF', 1: 'ON'};
            const valueLookup = {
                80 : '80A', 
                63 : '63A', 
                50 : '50A', 
                40 : '40A', 
                32 : '32A', 
                25 : '25A', 
                20 : '20A', 
[... 179 lines cut ...]
        tuyaDatapoints: [
            #[1, null, valueConverter.energy], // Consumed energy
            #[2, null, valueConverter.produced_energy], // Produced energy
			[1, 'energy', tuya.valueConverter.divideBy100], // Consumed energy
			[2, 'produced_energy', tuya.valueConverter.divideBy100], // Produced energy
[... cut ...]
            [9, null, valueConverter.power], // Total power
            [10, 'fault', valueConverter.fault],
            [11, null, valueConverter.energy_rst_hour_day], // Frozen - strange function, in native app - nothing is clear
            [16, 'state', tuya.valueConverter.onOff],
            [17, null, valueConverter.threshold], // It's settable, but can't write converter
            [18, null,valueConverter.meterid],
            [20, 'clear_fault', tuya.valueConverter.onOff] // Clear fault
        ],
    },
};

module.exports = definition;`

PetrozPL, 2023-09-23:
I've bought one of these for myself from aliexpress : 

https://pl.aliexpress.com/item/1005005679887461.html?spm=a2g0o.order_list.order_list_main.105.21ef1c24lwBKMz&gatewayAdapt=glo2pol

This device seems to be fully compatible identical, but had different fingerprint : '_TZE204_x8fp01wi'. After updating converter with this fingerprint everything worked just fine. I've also checked changes proposed by Cartol and it also worked. Now I'll try to find more matching exports and functions in tuya library.

closed 2023-10-31 by github-actions[bot]

derkorte, 2023-11-04:
Hey, love you work. 
Are there any new cluster translated from the exports?

teras, 2024-01-10:
Hello people

@Cartol 
> i bought exact the same device via aliexpress some days ago. The manufacturer in my case is EARU as printed. Mine is bidirictional by the way with exact same hardware values and wiring as yours.

I am thinking of buying the same device.  I am confused though. Is it working with Z2M? Is this the integration? https://www.zigbee2mqtt.io/devices/TS0601_3_phase_clamp_meter.html
If not, any plans on how to expose this information?

Thank you for your efforts!

IgnasGi, 2024-03-12:
Hi,
Works fine with Z2M. Nothing to code or tune. Just connect and use.

> Hello people
> I am thinking of buying the same device. I am confused though. Is it working with Z2M? Is this the integration? https://www.zigbee2mqtt.io/devices/TS0601_3_phase_clamp_meter.html If not, any plans on how to expose this information?
> Thank you for your efforts!

[image]

coo-private, 2025-03-12:
> ### Comments
> I've bought a 3-Phase Multifunction Energy Meter PJ-DAC7361C that isn't supported in zigbee2mqtt yet.

> 
> My solution isn't complete. I would like to fix it to use tuya.valueConverter presets like divideBy100, because when I've tried to use them it wasn't working. I've also couldn't use "to:" mechanism inside local VectorConverter (commented out) and I've had to add my own tzLocal. It would be great to know how to use this "to:" correctly. Still missing support of "Frozen set", which I don't know what is, but is exposed at datapoint 11.
> 
> Finally, it would be great if complete support will be merged to zigbee2mqtt next release.
> 
> I would appriciate any help to make this support complete. :)
> 

As I see it, after two years of development (@Koenkk ) unfortunately, support for meter functionality has not improved.
I purchased a similar model and it is still not supported.
I have created a new issue to support the new model https://github.com/Koenkk/zigbee2mqtt/issues/26642 
But as I see from similar devices - there is no support for full functionality. Only telemetry transmission. ((

@PetrozPL how did you manage to add this additional data?
I tried applying your converter, but unfortunately it didn't work.
It only worked with the converter from @Cartol  , but it is correspondingly limited (

[image]

PetrozPL, 2025-03-13:
@coo-private 

I've just observed the data that was reported by the device and tried to sort it out based on other converters for similar devices.

I would love to see it supported in zigbee2mqtt, but I don't have a spare hardware platform to run z2mqtt directly on it - I'm using docker for and so far was unsuccessful in building my own docker image from git. Back then, If I was able to do so I would probably be working on adding permanent support. But due to lack of time and help I just let it go.

coo-private, 2025-03-13:
@PetrozPL 

I want to buy thuja gateway and get the data as per these instructions 

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_2-signin-up-for-tuya-iot-platform

Tell me, if I give you the data, can you help with the converter?

PetrozPL, 2025-03-29:
@coo-private 

It's just not a matter of data - It's a matter of testing/debugging environment - without docker image that I could possibly rebuild on demand I'm not able to easily develop and net check the converter and it's integration into zigbee2mqtt. Without it I can only prepare converter that is suited to current zigbee2mqtt architecture - next big change and it will be possibly outdated again. It's not worth the time it takes to get it working again.

And regarding the data - you don't need tuya gateway to get the data from the device. Just remove any converters for this device (to show it as unsupported) and check the debug log - you should see there all the data that device is publishing. Next all that needs to be done is interpretation of the frames and data in it and cross reference them with the values that are shown on display.

If you really want to give it a try, then for this particular device I believe all the data that could be parsed and interpreted were already supported in my converter, but as you already noticed - it is a little bit outdated. But still, it can be used as reference to implement the required functionality in new format. If you dig into the files format and are eager to search for already supported functions in zigbee2mqtt git repository, then you should be able to add the support of all the parameters by step-by-step modifications of Cartol`s converter.