Koenkk/zigbee2mqtt #18752: SyntaxError: Unexpected token '(' in node_modules/rimraf/node_modules/minimatch/dist/cjs/ast.js:86

jpara, 2023-08-26:
### What happened?

Aug 26 19:23:32 SmartHomeNG npm[1337]: /opt/zigbee2mqtt/node_modules/rimraf/node_modules/minimatch/dist/cjs/ast.js:86
Aug 26 19:23:32 SmartHomeNG npm[1337]:     #fillNegs() {
Aug 26 19:23:32 SmartHomeNG npm[1337]:              ^
Aug 26 19:23:32 SmartHomeNG npm[1337]: SyntaxError: Unexpected token '('

### What did you expect to happen?

to start also after the latest update

### How to reproduce it (minimal and precise)

git pull and restart

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

n.a. -> dead!

### Adapter

CC2531

Koenkk, 2023-08-27:
You are using an unsupported Node.js error, updating it will solve your issue (use e.g. Node.js 18)

closed 2023-08-27 by Koenkk

jpara, 2023-08-27:
thanks a lot