Koenkk/zigbee2mqtt #18772: Aqara Smart Wall Switch (with neutral, triple Rocker) - SKU AK095GLW01 and S/N: A00385/LS1CQA00311

Jens-Wymeersch, 2023-08-28:
### Is your feature request related to a problem? Please describe

I'm moving all my LELLKI switches to AQARA. This implies changing 1-rocker, 2-rocker and 3-rocker with neutral switches to aqara H1 switches. 

The single and double rockers are supported. The Triple rocker without neutral (QBKG29LM) is also supported. Only the Triple rocker with neutral isn't. I assume as nobody in Europe has it yet, and it is extremely hard to find... 

### Describe the solution you'd like

I would like to get this switch to be supported like the switch without neutral.

### Describe alternatives you've considered

The only alternative is using the aqara app. However the integration with Home Assistant isn't what it should be.

### Additional context

My Z2M coordinator connects quite quickly to the device (but indicates that it is unsupported).
As stated I have an aqara hub - so tests in all direction can be made.

Jens-Wymeersch, 2023-08-30:
@Koenkk Can you please help me ? It shouldn't be a big project as the non-neutral switch is already supported. Thank you so much

ssnaveen10, 2023-08-31:
+1.. I'm based in Singapore and I have a new Aqura SW H1  3-rocker (7 Qty).. as of now I'm using Zigbee-herdsman-converters as a workaround .
[WSDCGQ01LM.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/12485854/WSDCGQ01LM.js.txt)

Koenkk, 2023-09-01:
thanks @ssnaveen10, added!

Device will be supported in todays release.

closed 2023-09-01 by Koenkk

ssnaveen10, 2023-09-02:
@Koenkk - Thank you so much.. Now works with a new update! Have a great weekend!
[image]

Jens-Wymeersch, 2023-09-02:
@Koenkk Thank you so much. After update, it recognized the device instantly !