Koenkk/zigbee2mqtt #18775: Gledopto Model GL-S-006P requests OTA every 30 seconds

Luligu, 2023-08-28:
### What happened?

The device request OTA each 30 seconds. So the log are just full of these requests.

### What did you expect to happen?

How can I disable completely the OTA support for this device or maybe limit the request in the log to once in a day?

### How to reproduce it (minimal and precise)

Just pair it.

### Zigbee2MQTT version

1.32.2 commit: 1ec1e57

### Adapter firmware version

20221226

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

### Debug log

Zigbee2MQTT:debug 2023-08-28 11:09:57: Device '0xa4c138071003495f' requested OTA
Zigbee2MQTT:debug 2023-08-28 11:09:57: Responded to OTA request of '0xa4c138071003495f' with 'NO_IMAGE_AVAILABLE'
Zigbee2MQTT:debug 2023-08-28 11:10:27: Device '0xa4c138071003495f' requested OTA
Zigbee2MQTT:debug 2023-08-28 11:10:27: Responded to OTA request of '0xa4c138071003495f' with 'NO_IMAGE_AVAILABLE'
Zigbee2MQTT:debug 2023-08-28 11:10:57: Device '0xa4c138071003495f' requested OTA
Zigbee2MQTT:debug 2023-08-28 11:10:57: Responded to OTA request of '0xa4c138071003495f' with 'NO_IMAGE_AVAILABLE'

oat1, 2023-09-04:
The same problem with https://www.zigbee2mqtt.io/devices/GL-B-002P.html

Koenkk, 2023-09-04:
> How can I disable completely the OTA support for this device or maybe limit the request in the log to once in a day?

You cannot, this is incorrect behaviour from the device itself. Z2M responds with a `NO_IMAGE_AVAILABLE` but the device keeps requesting.

Luligu, 2023-09-05:
Ok thank you for clarifying.

closed 2023-09-05 by Luligu

mrmaximas, 2024-02-06:
Debug 2024-02-06 11:40:54Received Zigbee message from 'Living Room Main Light 1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:40:54Device 'Living Room Main Light 1' requested OTA
Debug 2024-02-06 11:40:54Responded to OTA request of 'Living Room Main Light 1' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:40:59Received Zigbee message from 'Living Room Main Light 6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:40:59Device 'Living Room Main Light 6' requested OTA
Debug 2024-02-06 11:40:59Responded to OTA request of 'Living Room Main Light 6' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:02Received Zigbee message from 'Living Room Main Light 3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:02Device 'Living Room Main Light 3' requested OTA
Debug 2024-02-06 11:41:02Responded to OTA request of 'Living Room Main Light 3' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:02Received Zigbee message from 'Living Room Main Light 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:02Device 'Living Room Main Light 2' requested OTA
Debug 2024-02-06 11:41:02Responded to OTA request of 'Living Room Main Light 2' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:02Received Zigbee message from 'Living Room Main Light 5', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:02Device 'Living Room Main Light 5' requested OTA
Debug 2024-02-06 11:41:02Responded to OTA request of 'Living Room Main Light 5' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:03Received Zigbee message from 'Living Room Main Light 4', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:03Device 'Living Room Main Light 4' requested OTA
Debug 2024-02-06 11:41:03Responded to OTA request of 'Living Room Main Light 4' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:24Received Zigbee message from 'Living Room Main Light 1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:24Device 'Living Room Main Light 1' requested OTA
Debug 2024-02-06 11:41:24Responded to OTA request of 'Living Room Main Light 1' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:29Received Zigbee message from 'Living Room Main Light 6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:29Device 'Living Room Main Light 6' requested OTA
Debug 2024-02-06 11:41:29Responded to OTA request of 'Living Room Main Light 6' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:32Received Zigbee message from 'Living Room Main Light 3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:32Device 'Living Room Main Light 3' requested OTA
Debug 2024-02-06 11:41:32Responded to OTA request of 'Living Room Main Light 3' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:32Received Zigbee message from 'Living Room Main Light 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:32Device 'Living Room Main Light 2' requested OTA
Debug 2024-02-06 11:41:32Responded to OTA request of 'Living Room Main Light 2' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:32Received Zigbee message from 'Living Room Main Light 5', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:32Device 'Living Room Main Light 5' requested OTA
Debug 2024-02-06 11:41:32Responded to OTA request of 'Living Room Main Light 5' with 'NO_IMAGE_AVAILABLE'
Debug 2024-02-06 11:41:33Received Zigbee message from 'Living Room Main Light 4', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":587280385,"imageType":5142,"manufacturerCode":4687}' from endpoint 11 with groupID 0
Debug 2024-02-06 11:41:33Device 'Living Room Main Light 4' requested OTA
Debug 2024-02-06 11:41:33Responded to OTA request of 'Living Room Main Light 4' with 'NO_IMAGE_AVAILABLE'