Koenkk/zigbee2mqtt #18776: IKEA LED2004G8 report wrong state

legalgig, 2023-08-28:
### What happened?

IKEA LED2004G8 report wrong state after about 15mins from turning them off.

### How to reproduce it (minimal and precise)

1. Pair the bulb
2. turn it on 
3. turn it off
4. the bulb reports correct state to the zigbee2mqtt
5. go to the dev console inside of the device 
6. select endpoint 1, cluster onOff, attribute onOff and then read
7. it should report `Read result of 'genOnOff': {"onOff":0}`
8. after around 15-60mins read out the state again
9. now it reports `Read result of 'genOnOff': {"onOff":1}` even though it isn't on

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

SLZB-06 Ethernet

LookedPath, 2023-08-31:
I think this might be similar or related to #16388

legalgig, 2023-10-01:
Closing, duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/16388

closed 2023-10-01 by legalgig