Koenkk/zigbee2mqtt #18777: Devices suddenly disappearing, unable to re-pair

Stooovie, 2023-08-28:
### What happened?

Multiple devices disappearing within 24 hours. Slae.sh stick 2562 stick.

Issue 1 - [Tradfri E27 bulb](https://www.zigbee2mqtt.io/devices/LED1732G11.html) left the network, unable to re-pair again. Now it says it's an unsupported device. 

Issue 2 - [Xiaomi motion sensor ](https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html#xiaomi-rtcgq11lm) left network, was able to re-pair.

Multiple other devices including a [Lidl plug](https://www.zigbee2mqtt.io/devices/HG06337.html#lidl-hg06337) reporting issues now.

[Log including herdsman-debug](https://transfer.sh/z18Js4HXYG/log.txt). All this has been working for years. Please advise. Switching to Z2M Edge did not fix the issue, nor did complete reboot of the HA VM. HAOS VM running under Proxmox 8.0.4 with Z2M passthrough. There has been no update to Proxmox in this time.

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

20220219

### Adapter

Slae'sh 2652RB

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12453189/log.txt)

deanpribetic, 2023-08-29:
I'm having a similar issue…lights are suddenly offline (mostly Hue bulbs)

`2023-08-29 16:30:48Failed to ping 'Entrance Light' (attempt 1/1, Read 0x00178801040f685c/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (193)))`

Stooovie, 2023-08-29:
I'm getting MAC channel and ACK errors - am I right this is all caused by some sort of interference that wasn't physically there before?

deanpribetic, 2023-08-29:
That's what is happening with mine too. I asked in the Home Assistant Discord and they told me it would be interference. Not very helpful as nothing in my set up has changed.

closed 2023-10-06 by github-actions[bot]