Koenkk/zigbee2mqtt #18788: Invalid option for select.cover01_set_bottom_limit: 'True' (valid options: ['SET']) | Moes AM43-0.45/40-ES-EB

szatan81, 2023-08-29:
### What happened?

After updating HA (2023.8.4) & z2m (1.32.2) to latest versions my HA got flooded with that error for every Moes AM43-0.45/40-ES-EB, example:

Invalid option for select.roleta_salon_01_set_bottom_limit: 'True' (valid options: ['SET'])
Invalid option for select.roleta_salon_02_set_upper_limit: 'True' (valid options: ['SET'])
Invalid option for select.roleta_salon_02_set_bottom_limit: 'True' (valid options: ['SET'])
Invalid option for select.roleta_salon_03_set_upper_limit: 'True' (valid options: ['SET'])
Invalid option for select.roleta_salon_03_set_bottom_limit: 'True' (valid options: ['SET'])

### What did you expect to happen?

Get rid of an error

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

Invalid option for select.roleta_salon_01_set_bottom_limit: 'True' (valid options: ['SET'])
Invalid option for select.roleta_salon_02_set_upper_limit: 'True' (valid options: ['SET'])
Invalid option for select.roleta_salon_02_set_bottom_limit: 'True' (valid options: ['SET'])
Invalid option for select.roleta_salon_03_set_upper_limit: 'True' (valid options: ['SET'])
Invalid option for select.roleta_salon_03_set_bottom_limit: 'True' (valid options: ['SET'])

szatan81, 2023-09-19:
Maybe someone could taka a look on that. It spams my HA log pretty bad:

First occurred: August 29, 2023 at 3:56:47 PM (5834 occurrences) 

It is probably small fix in zigbee-herdsman-converters

Thanks

closed 2023-10-30 by github-actions[bot]

WRLPDZ, 2023-12-22:
Same here.
Any info?

szatan81, 2023-12-24:
What i did:

1.Entered integrations in HA -> MQTT
2. Found cover entity
3. I disabled "set_upper_limit" and "set_bottom_limit" entities in "Controls" card.

It's a workaround but i don't have errors anymore.