Koenkk/zigbee2mqtt #18791: _TZE200_5toc8efa problem with current_heating_setpoint and local_temperature value

VoltickVL, 2023-08-30:
### What happened?

When I change the current_heating_setpoint setting to, say, a temperature of 28 degrees, a huge numeric value is automatically set. It is impossible to set the temperature in the range from 5 to 35 degrees Celsius. Also, the value of local_temperature is displayed incorrectly. Both parameters should be 10 times lower. In addition, the device supports a temperature change step equal to 0.5, and in zigbee2mqtt, the default step is 1, can this be fixed?

[image]
[image]

### What did you expect to happen?

I have the same device, but with a slightly different version - _TZE200_aoclfnxz and it displays the correct current_heating_setpoint and local_temperature parameters and works correctly. But he also needs to correct the temperature change step from 1 to 0.5.
[image]
[image]

### How to reproduce it (minimal and precise)

1) These parameters need to be reduced by 10 times:
- current_heating_setpoint
- local_temperature

2) It is necessary to reduce the step value when changing the current_heating_setpoint parameter from 1 to 0.5

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

20230507

### Adapter

zStack3x0

VoltickVL, 2023-09-03:
Plz fix it :(

VoltickVL, 2023-09-03:
Need to fix zigbee-herdsman-converters/src/devices/moes.ts

VoltickVL, 2023-09-17:
Any body can help me???

drbartek, 2023-09-22:
I confirm this problem

Koenkk, 2023-09-23:
@VoltickVL I'm a bit confused here, you made a PR to multiply the current_heating_setpoint by 10 here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/5372

drbartek, 2023-09-23:
The original delivered 10kOhm thermistor is connected to my thermostat (_TZE200_5toc8efa). The problem may be that 'someone' connected the device to another thermistor (different resistance value). 
In my opinion, there should be a linear calibration option where the 'default' value is '1' (original thermistor in this case 10kOhm). Maybe the solution is not perfect, but it will allow to use different thermistors.
I have a really big problem with eliminating this issue ie 'x10'. I'm getting a little crazy with the 'Better Thermostat' ;-)

You should probably go back to the previous settings. Thank you

VoltickVL, 2023-09-24:
> @VoltickVL I'm a bit confused here, you made a PR to multiply the current_heating_setpoint by 10 here: [Koenkk/zigbee-herdsman-converters#5372](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5372) . Should I revert your fix?

Apparently you've made changes now. The Current heating setpoint is now displayed correctly. BUT the error with the display Local temperature = 0.1 ... It is always equal to this value and does not change
[image]

VoltickVL, 2023-09-24:
> The original delivered 10kOhm thermistor is connected to my thermostat (_TZE200_5toc8efa). The problem may be that 'someone' connected the device to another thermistor (different resistance value). [...]

I use a thermistor from the kit, everything works fine on other versions of underfloor heating, so the problem is not in this, but in the technical part of the converter. 

Now the problem is almost solved - it remains only to fix the display of the current temperature.

drbartek, 2023-09-24:
I updated 'zigbee2mqtt' docker to the 'dev' version:

[image]

and see no difference

[image]

@VoltickVL Could you please check the 'resistance' of your thermistor. Thanks.

VoltickVL, 2023-09-24:
There is no multimeter at home to check, but the problem is definitely not in the resistor. The cable is standard and the resistor too, also the device itself displays the correct information and if the problem was in the resistor (resistance), the device would also show incorrect temperature data. Here the problem is in the formula in the converter.

drbartek, 2023-09-24:
> There is no multimeter at home to check, but the problem is definitely not in the resistor. The cable is standard and the resistor too, also the device itself displays the correct information and if [...]

Correct, I also have the right temperature information on the device display - clearly something is wrong with the conversion formula inside 'zigbee2mqtt'

VoltickVL, 2023-09-25:
@Koenkk plz help us

Koenkk, 2023-09-25:
@VoltickVL what should I do, revert https://github.com/Koenkk/zigbee-herdsman-converters/pull/5372 ? (then `current_heating_setpoint` divided by 10 will be send to the device)

drbartek, 2023-09-25:
@Koenkk thanks!
From my side it looks like a good approach - let's just see. If this information is helpful, I have a second thermostat (_TZE200_aoclfnxz) that shows the correct values of tempertures in 'zigbee2mqtt'
@VoltickVL you are the author of this request confirm please return to previous settings.

drbartek, 2023-09-27:
Hi @VoltickVL  could you please accept the suggestions above, also looking forward to fixing this bug. Thanks

drbartek, 2023-09-30:
@Koenkk Could I ask to return to the previous settings within this request or should I create a new one?

Koenkk, 2023-10-03:
@drbartek reverted https://github.com/Koenkk/zigbee-herdsman-converters/pull/5372 (will be included in todays release)

drbartek, 2023-10-03:
Thanks for the update.
The issue with 'Current heating setpoint' has been fixed but there is still a 'x10' problem with 'Local temperature'.

[image]

[image]

I suspect that 'Local temperature calibration' is also going crazy for this reason.
[image]

I have updated docker to today's dev-version. I did not remove and add the device again to 'zigbee2mqtt'.

Koenkk, 2023-10-03:
@drbartek pushed a fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

drbartek, 2023-10-03:
Almost perfect. Thanks. Both problems have been fixed.  @Koenkk  please check "Local temperature calibration" again, because when you move the slider, a very high value appears.
Which changes the 'Local temperature' value reading.
[image]
after
[image]
back to '0 ' with 'Local temperature calibration' value and (click 'off' and 'heat') and it is fine
[image]

drbartek, 2023-10-03:
You know @Koenkk the same problem (with the 'Local temperature calibration' ) also appears on another thermostat ie.
_TZE200_aoclfnxz wich I have:
[image]

[image]

the same 'play' click click and you see result below:
[image]

maybe it's a feature and not a bug ;-)

kastmgnru, 2023-10-03:
Hi @Koenkk !
Same problem for _TZE204_5toc8efa. 
Values for: "Deadzone temperature", "Max temperature limit", "Min temperature limit" and "Current heating setpoint" are multiplied by 10.
[image]

Koenkk, 2023-10-04:
can you provide the debug log when setting the local temp calibration? 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

kastmgnru, 2023-10-04:
For _TZE204_5toc8efa, the local temp calibration setting works correctly if I put integers. If I use decimal numbers, the setting breaks. But also my thermostat does not support decimal numbers for calibration.

drbartek, 2023-10-04:
Hope this helps you (log from _TZE200_5toc8efa)
.....

Zigbee2MQTT:info  2023-10-04 21:08:37: MQTT publish: topic 'homeassistant/sensor/0x003c84fffed332f5/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Floor_Heating_Thermostat_01/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"configuration_url":"https://XXXX.XX/#/device/0x003c84fffed332f5/info","identifiers":["zigbee2mqtt_0x003c84fffed332f5"],"manufacturer":"Moes","model":"Moes BHT series Thermostat (BHT-002-GCLZB)","name":"Floor_Heating_Thermostat_01"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/Floor_Heating_Thermostat_01","name":"Linkquality","object_id":"floor_heating_thermostat_01_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.33.1-dev","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Floor_Heating_Thermostat_01","unique_id":"0x003c84fffed332f5_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

Zigbee2MQTT:info  2023-10-04 21:08:37: MQTT publish: topic 'homeassistant/sensor/0x003c84fffed332f5/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Floor_Heating_Thermostat_01/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"configuration_url":"https://XXXX.XX/#/device/0x003c84fffed332f5/info","identifiers":["zigbee2mqtt_0x003c84fffed332f5"],"manufacturer":"Moes","model":"Moes BHT series Thermostat (BHT-002-GCLZB)","name":"Floor_Heating_Thermostat_01"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","json_attributes_topic":"zigbee2mqtt/Floor_Heating_Thermostat_01","name":"Last seen","object_id":"floor_heating_thermostat_01_last_seen","origin":{"name":"Zigbee2MQTT","sw":"1.33.1-dev","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Floor_Heating_Thermostat_01","unique_id":"0x003c84fffed332f5_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'

Zigbee2MQTT:info  2023-10-04 21:08:37: MQTT publish: topic 'zigbee2mqtt/Floor_Heating_Thermostat_01', payload '{"child_lock":"UNLOCK","current_heating_setpoint":30,"deadzone_temperature":0,"heat":"OFF","last_seen":"2023-10-04T21:08:16+02:00","linkquality":109,"local_temperature":22,"local_temperature_calibration":0,"max_temperature":30,"max_temperature_limit":35,"min_temperature_limit":0,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":0,"saturday_p1_minute":0,"saturday_p1_temperature":0,"saturday_p2_hour":0,"saturday_p2_minute":0,"saturday_p2_temperature":0,"saturday_p3_hour":0,"saturday_p3_minute":0,"saturday_p3_temperature":0,"saturday_p4_hour":0,"saturday_p4_minute":0,"saturday_p4_temperature":0,"sunday_p1_hour":0,"sunday_p1_minute":0,"sunday_p1_temperature":0,"sunday_p2_hour":0,"sunday_p2_minute":0,"sunday_p2_temperature":0,"sunday_p3_hour":0,"sunday_p3_minute":0,"sunday_p3_temperature":0,"sunday_p4_hour":0,"sunday_p4_minute":0,"sunday_p4_temperature":0,"weekdays_p1_hour":0,"weekdays_p1_minute":0,"weekdays_p1_temperature":0,"weekdays_p2_hour":0,"weekdays_p2_minute":0,"weekdays_p2_temperature":0,"weekdays_p3_hour":0,"weekdays_p3_minute":0,"weekdays_p3_temperature":0,"weekdays_p4_hour":0,"weekdays_p4_minute":0,"weekdays_p4_temperature":0},"running_state":"idle","sensor":"OU","system_mode":"off"}'

Zigbee2MQTT:info  2023-10-04 21:08:37: MQTT publish: topic 'zigbee2mqtt/Floor_Heating_Thermostat_01', payload '{"child_lock":"UNLOCK","current_heating_setpoint":30,"deadzone_temperature":0,"heat":"OFF","last_seen":"2023-10-04T21:08:37+02:00","linkquality":94,"local_temperature":22,"local_temperature_calibration":0,"max_temperature":30,"max_temperature_limit":35,"min_temperature_limit":0,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":0,"saturday_p1_minute":0,"saturday_p1_temperature":0,"saturday_p2_hour":0,"saturday_p2_minute":0,"saturday_p2_temperature":0,"saturday_p3_hour":0,"saturday_p3_minute":0,"saturday_p3_temperature":0,"saturday_p4_hour":0,"saturday_p4_minute":0,"saturday_p4_temperature":0,"sunday_p1_hour":0,"sunday_p1_minute":0,"sunday_p1_temperature":0,"sunday_p2_hour":0,"sunday_p2_minute":0,"sunday_p2_temperature":0,"sunday_p3_hour":0,"sunday_p3_minute":0,"sunday_p3_temperature":0,"sunday_p4_hour":0,"sunday_p4_minute":0,"sunday_p4_temperature":0,"weekdays_p1_hour":0,"weekdays_p1_minute":0,"weekdays_p1_temperature":0,"weekdays_p2_hour":0,"weekdays_p2_minute":0,"weekdays_p2_temperature":0,"weekdays_p3_hour":0,"weekdays_p3_minute":0,"weekdays_p3_temperature":0,"weekdays_p4_hour":0,"weekdays_p4_minute":0,"weekdays_p4_temperature":0},"running_state":"idle","sensor":"OU","system_mode":"off"}'

Zigbee2MQTT:info  2023-10-04 21:09:00: MQTT publish: topic 'zigbee2mqtt/Floor_Heating_Thermostat_01', payload '{"child_lock":"UNLOCK","current_heating_setpoint":30,"deadzone_temperature":0,"heat":"OFF","last_seen":"2023-10-04T21:09:00+02:00","linkquality":109,"local_temperature":22,"local_temperature_calibration":268470682,"max_temperature":30,"max_temperature_limit":35,"min_temperature_limit":0,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":0,"saturday_p1_minute":0,"saturday_p1_temperature":0,"saturday_p2_hour":0,"saturday_p2 [... cut ...] temperature":0,"sunday_p3_hour":0,"sunday_p3_minute":0,"sunday_p3_temperature":0,"sunday_p4_hour":0,"sunday_p4_minute":0,"sunday_p4_temperature":0,"weekdays_p1_hour":0,"weekdays_p1_minute":0,"weekdays_p1_temperature":0,"weekdays_p2_hour":0,"weekdays_p2_minute":0,"weekdays_p2_temperature":0,"weekdays_p3_hour":0,"weekdays_p3_minute":0,"weekdays_p3_temperature":0,"weekdays_p4_hour":0,"weekdays_p4_minute":0,"weekdays_p4_temperature":0},"running_state":"idle","sensor":"OU","system_mode":"off"}'
.....

VoltickVL, 2023-10-05:
After update to ZigBee version: 1.33.1-1 again i have bug with Local temperature, Max temperature limit, and Current heating setpoint :(((((

 
[image]
[image]

VoltickVL, 2023-10-05:
And now when I set the temperature to 35 degrees, 3.5 degrees is set on the device.

VoltickVL, 2023-10-05:
I think need change code at .../zigbee-herdsman-converters/tree/master/src/lib//legacy.ts

stroke 3853
`                if (!['_TZE200_ztvwu4nk', '_TZE200_ye5jkfsb'].includes(meta.device.manufacturerName)) {
`
to

`                if (!['_TZE200_ztvwu4nk', '_TZE200_ye5jkfsb', '_TZE200_5toc8efa'].includes(meta.device.manufacturerName)) {`

and at strokes 6705-6706 need to add new data only for '_TZE200_5toc8efa'

```
            const temperature = Math.round(value*10);
            await tuya.sendDataPointValue(entity, tuya.dataPoints.moesHeatingSetpoint, temperature);
```

VoltickVL, 2023-10-05:
But I don't know how change code

`    moes_thermostat_current_heating_setpoint: {
        key: ['current_heating_setpoint'],
        convertSet: async (entity, key, value, meta) => {
            await sendDataPointValue(entity, dataPoints.moesHeatingSetpoint, value);
        },`

We need this code only for '_TZE200_5toc8efa':

``    moes_thermostat_current_heating_setpoint: {
        key: ['current_heating_setpoint'],
        convertSet: async (entity, key, value, meta) => {
          const temp = Math.round(value*10);
          await sendDataPointValue(entity, dataPoints.moesHeatingSetpoint, temp);
        },``

drbartek, 2023-10-05:
@VoltickVL  I don't know what is wrong with this model but it now displays everything correctly.
Of course calibration is a separate topic.
[image]

drbartek, 2023-10-05:
Ok I checked on the device and there is really a problem because it now shows 100 times less i.e. I set '30' (Current heating setpoint) it shows '0.30' on the device

drbartek, 2023-10-05:
Maybe something is wrong with the sensor after all (1kOhm?).  @Koenkk  does this make sense? Maybe @VoltickVL  should check the resistances of his sensor? This is probably the only variable that can affect everything.

VoltickVL, 2023-10-05:
> Maybe something is wrong with the sensor after all (1kOhm?). @Koenkk does this make sense? Maybe @VoltickVL should check the resistances of his sensor? This is probably the only variable that can affect everything.

Once again, I repeat. The device shows the correct temperature values, so there can be no problems with the sensor and there is no point in changing its resistance.

drbartek, 2023-10-05:
> And now when I set the temperature to 35 degrees, 3.5 degrees is set on the device.

The same thing happens with me, but the "Local Temperature" is now showed correct and with you the "Local Temperature" is now displayed correctly, while with you the "Local Temperature" is 10 higher and it depends on the sensor, yes?
That's why I think you should check it out.
I think we want to solve this problem together. 

@Koenkk does it make sense to measure sensor resistance?

VoltickVL, 2023-10-05:
> > And now when I set the temperature to 35 degrees, 3.5 degrees is set on the device.
> 
> The same thing happens with me, but the "Local Temperature" is now displayed correctly, while with you the "Local Temperature" is 10 higher and it depends on the sensor, yes? That's why I think you should check it out - we want to solve this problem together, probably.
> 
> @Koenkk does it make sense to measure sensor resistance?

[image]
[image]

VoltickVL, 2023-10-05:
If i set Sensor to "IN", i have same 250*C local temperature :(

drbartek, 2023-10-05:
My short comparison @drbartek / @VoltickVL:
                        
Sensor:  10kOhm / XkOhm
'Local temperature':  25 / 250
Temperature on the device display 'Current heating setpoint' (permanent): 25 / 25
Temperature on the device display (temporary)* - if you are changing by slider: 3,5 / X

*Watch the devices as you change the 'Current heating setpoint' in zigbee2mqtt 

> If i set Sensor to "IN", i have same 250*C local temperature :(

The same thing happens with me - no change if I chaned IN/AL/OUT

Koenkk, 2023-10-05:
@drbartek seems something is wrong with on of your devices indeed. I cannot fix the problem for you both.

VoltickVL, 2023-10-06:
@Koenkk plz check this code:

>     moes_thermostat_current_heating_setpoint: {
>     key: ['current_heating_setpoint'],
>     convertSet: async (entity, key, value, meta) => {
>         if (value >= 1 && value <= 5) 
>         value = Math.round(value * 10);
>       await sendDataPointValue(entity, dataPoints.moesHeatingSetpoint, value);
>     },

If the device tries to set the temperature from 1 degree Celsius to 5 degrees Celsius, then we use the formula to increase the value by 10 times: (value * 10).
It's okey? Sorry but I'm not programmer :-)

VoltickVL, 2023-10-06:
@drbartek 

> My short comparison @drbartek / @VoltickVL:
> 
> Sensor: 10kOhm / XkOhm
> 'Local temperature': 25 / 250
> Temperature on the device display 'Current heating setpoint' (permanent): 25 / 25
> Temperature on the device display (temporary)* - if you are changing by slider: 3,5 / X

My short comparison @drbartek / @VoltickVL:

Sensor: 10kOhm / 10kOhm
'Local temperature': 25 / 250
Temperature on the device display 'Current heating setpoint' (permanent): 25 / 25 | X / 250
Temperature on the device display (temporary)* - if you are changing by slider: 3,5 / 2,5 | X /25

kamilnuriev, 2023-10-07:
I'll add

**ToZigbee**
Set 'Current heating setpoint': 300
If I set 'Max temperature limit': 35, then the device does not turn on the heating.
If I set 'Max temperature limit': 350, then the device turns on the heating.
**_Need to fix (value * 10)_**

VoltickVL, 2023-10-07:
> I'll add
> 
> **ToZigbee** Set 'Current heating setpoint': 300 If I set 'Max temperature limit': 35, then the device does not turn on the heating. If I set 'Max temperature limit': 350, then the device turns on the heating. **_Need to fix (value * 10)_**

It's no need to fix when add my patch. I create pr

VladimirTuzovGitHub, 2023-10-08:
Hi all. Has anyone had any positive results? I don’t have any changes, all settings in z2m are multiplied by 10, and in version z2m edge 15.90.0 they are also multiplied by 10.

drbartek, 2023-10-11:
@Koenkk, @VoltickVL  thanks for the support, I exchanged the thermostat for a different version.

Koenkk, 2023-10-11:
@VoltickVL is this issue fixed with the latest dev branch?

VoltickVL, 2023-10-11:
> @VoltickVL is this issue fixed with the latest dev branch?

I installed the Zigbee2MQTT module, installed Zigbee2MQTT Edge and don't see any changes for some reason... The temperature of the device is displayed 250 * C, when you try to change the temperature and set for example 30 * C, a command is sent to the device - set the temperature 3 * C... I don't understand why the changes didn't apply?(

VladimirTuzovGitHub, 2023-10-12:
> I have no changes, all settings in z2m are multiplied by 10, and in version z2m Edge 15.90.0 they are also multiplied by 10.

VladimirTuzovGitHub, 2023-10-12:
I have no changes, all settings in z2m are multiplied by 10, and in version z2m Edge 15.90.0 they are also multiplied by 10.

VoltickVL, 2023-10-12:
```
moes_thermostat_current_heating_setpoint: {
key: ['current_heating_setpoint'],
convertSet: async (entity, key, value, meta) => {
if (value >= 1 && value <= 5)
value = Math.round(value * 10);
await sendDataPointValue(entity, dataPoints.moesHeatingSetpoint, value);
},
```
I'm sorry, but this code doesn't work. Now if you set the temperature to 5 *C, 50 *C is displayed in the line, if you set the temperature to 2*C, for example, 20 *C will be displayed in the Current heating setpoint line, but the values are still sent to the device 10 times less than the desired one. And Local temperature doesn't work either, the values are displayed 10 times more:(

VoltickVL, 2023-10-12:
Line 3851:
>             case dataPoints.moesLocalTemp:
>                 temperature = value & 1<<15 ? value - (1<<16) + 1 : value;
>                 if (!['_TZE200_ztvwu4nk', '_TZE200_ye5jkfsb'].includes(meta.device.manufacturerName)) {
>                     // https://github.com/Koenkk/zigbee2mqtt/issues/11980
>                     temperature = temperature / 10;
>                 if (!['_TZE200_5toc8efa'].includes(meta.device.manufacturerName)) {
>                     temperature = value / 10;
>                 }
> 
>                 return {local_temperature: parseFloat(temperature.toFixed(1))};

Maybe this is a working version of the code?

VoltickVL, 2023-10-12:
I create a new PR - https://github.com/Koenkk/zigbee-herdsman-converters/pull/6276 @Koenkk plz check it

Koenkk, 2023-10-12:
@VoltickVL how are you running z2m?

VoltickVL, 2023-10-14:
> @VoltickVL how are you running z2m? (addon, docker, bare metal?)

Addon

Koenkk, 2023-10-14:
@VoltickVL see [this guide](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#testing-changes-locally) how to make changes locally

VoltickVL, 2023-10-14:
@Koenkk thank you, I will try and look for the right solution :) I used to do this with an external converter, and after updating z2m and adding Legacy, the external converter stopped working =(

VoltickVL, 2023-10-14:
I have HaOS 11, setup on Virtual Machine at synology ... Gow i can make changes locally? Code from guide don't work (

VladimirTuzovGitHub, 2023-10-14:
@VoltickVL Greetings. If you can fix it, please post it here. And write what actions you took.

VoltickVL, 2023-10-14:
Good news... I fix current_temperature... Wait and it's be ok)

VoltickVL, 2023-10-14:
Good news I managed to fix the problem with local_temperature . I continue to work on the fix !
[image]

VoltickVL, 2023-10-15:
For current_heating_setpoint code:
[image]

This code works fine with _TZE200_aoclfnxz . With the device version _TZE200_5toc8efa, the code also works, but in zigbee2mqtt it displays values 10 times larger. That is, when you set 25 degrees, it displays 250, but sends 25 to the device. It remains to change this code a little so that the correct value is displayed in z2m as well

[31 more replies]

closed 2024-05-28 by github-actions[bot]