Koenkk/zigbee2mqtt #18794: Docker image does not distiguish data and config folders

nettnikl, 2023-08-30:
### What happened?

Splitting data and config from each other is widely considered a best practice.  However, for this /app/data/ has to be used not for config files.

### What did you expect to happen?

Mounting possible with data as rw and config as ro.
```
    volumes:
      - ./zigbee2mqtt-config/configuration.yaml:/app/data/configuration.yaml:ro
      - ./zigbee2mqtt-config/custom-adapter.js:/app/data/custom-adapter.js:ro
      - ./zigbee2mqtt-data:/app/config:rw
```
But it is required to ensure, the start script does not copy stuff around (especially not only in the initial case), so if the config is changed afterwards, the change is reflected.

### Zigbee2MQTT version

latest

### Adapter firmware version

-

### Adapter

-

Hypfer, 2023-09-27:
To my understanding, the config file needs to be writable so that configuration via the frontend works.

> is widely considered a best practice

Any source on that? And why does that matter? Sounds like just an opinion to me

nettnikl, 2023-09-27:
A config file the program has to write to sounds like an unusual design choice to me. Not inherently a bad one, but for me an non cloud friendly one. For example, I like to reset my systems by deleting the data, while keeping the settings - similar to what MS started to offer for soft resetting Windows and what docker containers generally do on recreation (unless you specified to keep the data at a specific location via the volumes).

closed 2023-11-12 by github-actions[bot]

nettnikl, 2023-11-12:
This is very disappointing.

nettnikl, 2023-11-12:
I'd be thankful if i could get any reason for why this was closed. If it is not planned, i'm sure you have a good reason not to, i'd be interested to hear about it.